wmbusmeters/wmbusmeters #344: AMB8665-M stick not beeing detected

LarsBorris, 2021-09-24:
Hello!

What I want to do:
I have a Kamstrup Multical 602 heatmeter.
I want to use wmbusmeters to read the wm-bus telegram and feed it to my home assistant using mqtt. This is my goal.

I have the Amber 8665-M dongle.
Flashed the latest firmware on it yesterday from windows.
All ok!

I have installed (build from source) wmbusmeters on my laptop running Ubuntu.
The dongle is located in /dev/ttyUSB0:

mesg | grep tty
[    0.140155] printk: console [tty0] enabled
[  660.963692] usb 1-4: FTDI USB Serial Device converter now attached to ttyUSB0

(I have read that the AMB8465 and AMB8665 are the same but the AMB8665 has external antenna, is that correct?)

CONF:
loglevel=normal
device=/dev/ttyUSB0:amb8465:c1,t1
logtelegrams=false
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log

Output when running wmbusmeters:
wmbusmeters --debug auto
(wmbusmeters) version: _
(config) number of meters: 0
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) EVENT thread interrupted
(amb8465) could not open tty for detection
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyUSB0, it does not respond as any of the supported wmbus devices.
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(shell) return code 1
(shell) 14223 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(serial) EVENT thread interrupted
(shell) output: >>>14219
<<<
(shell) return code 0
(shell) 14224 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted
^C(serial) stopping manager
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!

What am I missing? I would greatly appreciate a little push in the right direction.
Thanks!
Have a great day.

weetmuts, 2021-09-24:
Hi! The log says: (amb8465) could not open tty for detection
do you have a wmbusmeters daemon running, or some other software that hogs the ttyUSB0?

Do:
lsof /dev/ttyUSB0
to see which process is using the device.

weetmuts, 2021-09-24:
Or, you have not run install, so that your user that wants to use the /dev/ttyUSB0 is member of the dialout group.

LarsBorris, 2021-09-24:
Hello!

Thank you for taking the time to help me.
I downloaded the source files, then ran the ./configure, then the make install command.
If I understand it right (i might not) then when I run the wmbusmeters application, this will conflict when running as a daemon right? Therefore I have not done the step with the daemon, nor have I configured any meters. 

When I run the lsof /dev/ttyUSB0 nothing is returned.

LarsBorris, 2021-09-24:
Your comment about install made me think.
I installed as daemon and put it into running.
Now when I check the wmbusmeters.log file I have communication!

Logfile:

(wmbusmeters) logging started 2021-09-24 12:51:08 using _
[2021-09-24_12:51:09] Started config amb8465[00000000] on /dev/ttyUSB0 listenin>
(wmbusmeters) waiting for telegrams
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 78152801
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Heat meter (0x04)
                   ver: 0x1c
                device: amb8465[00000000]
                  rssi: -31 dBm
                driver: unknown!

It says driver unknown, is the something that I can ignore? 
I have the KEM file for the device - its the encryption key from Kamstrup.

weetmuts, 2021-09-25:
The multical 602 heat meter is not supported yet. You can help me add this meter! 

Extract your meter key from your KEM file, see the README for how to do it.

Stop the daemon: sudo systemctl stop wmbusmeters.service

Run with another heat meter driver.
wmbusmeters --logtelegrams dev/ttyUSB0:amb8465:c1,t1 MyHeat multical603 78152801 YOUR_METER_KEY

Post the telegram=|.....| output here.

You can also collect some telegrams using: wmbusmeters --logtelegrams dev/ttyUSB0:amb8465:c1,t1 
and email me some telegrams, the KEM file and your password and I will do it for you.

LarsBorris, 2021-09-25:
Hello!

Here are the output of the command with my key:

Started config amb8465[00000000] on /dev/ttyUSB0 listening on c1,t1
(meter) MyHeat: meter detection did not match the selected driver multical603! correct driver is: unknown!
(meter) Not printing this warning agin for id: 78152801 mfct: (KAM) Kamstrup Energi (0x2c2d) type: Heat meter (0x04) ver: 0x1c
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) newly created meter (MyHeat 78152801 multical603) did not handle telegram!
telegram=|7A442D2C012815781C048D206450E76322344678|02F9FF1511130406690B010004EEFF07C1BC020004EEFF0890D401000414A925040084401400000000848040140000000002FD170000026CB929426CBF284406100A01004414D81A0400C4401400000000C480401400000000043B3900000002592A17025D2912|+24
MyHeat	78152801	68457 kWh	2717.85 m3	0.057 m3/h	59.3 °C	46.49 °C		2021-09-25 21:51.27
telegram=|4F442D2C012815781C048D206551E76322BE767900843005|1113690B0100C1BC020090D40100A925040000000000000000000000B929BF28100A0100D81A04000000000000000000390000002A172912|+40
MyHeat	78152801	68457 kWh	2717.85 m3	0.057 m3/h	59.3 °C	46.49 °C		2021-09-25 21:51.43
telegram=|4F442D2C012815781C048D206651E76322BE767900843005|1113690B0100C1BC020090D40100A925040000000000000000000000B929BF28100A0100D81A04000000000000000000390000002A172912|+56
MyHeat	78152801	68457 kWh	2717.85 m3	0.057 m3/h	59.3 °C	46.49 °C		2021-09-25 21:51.59
telegram=|4F442D2C012815781C048D206751E76322BE767900843005|1113690B0100C1BC020090D40100A925040000000000000000000000B929BF28100A0100D81A04000000000000000000390000002A172912|+71
MyHeat	78152801	68457 kWh	2717.85 m3	0.057 m3/h	59.3 °C	46.49 °C		2021-09-25 21:52.14
telegram=|4F442D2C012815781C048D206951E76322BE767900843005|1113690B0100C1BC020090D40100A925040000000000000000000000B929BF28100A0100D81A04000000000000000000390000002A172912|+103
MyHeat	78152801	68457 kWh	2717.85 m3	0.057 m3/h	59.3 °C	46.49 °C		2021-09-25 21:52.46
telegram=|4F442D2C012815781C048D206A51E76322BE767900843005|1113690B0100C1BC020090D40100A925040000000000000000000000B929BF28100A0100D81A04000000000000000000390000002A172912|+119
MyHeat	78152801	68457 kWh	2717.85 m3	0.057 m3/h	59.3 °C	46.49 °C		2021-09-25 21:53.02
telegram=|4F442D2C012815781C048D206B51E76322BE767900843005|1113690B0100C1BC020090D40100A925040000000000000000000000B929BF28100A0100D81A04000000000000000000390000002A172912|+134
MyHeat	78152801	68457 kWh	2717.85 m3	0.057 m3/h	59.3 °C	46.49 °C		2021-09-25 21:53.17
telegram=|7A442D2C012815781C048D206C70E76322344678|02F9FF1511130406690B010004EEFF07C1BC020004EEFF0890D401000414A925040084401400000000848040140000000002FD170000026CB929426CBF284406100A01004414D81A0400C4401400000000C480401400000000043B3900000002592A17025D2912|+150
MyHeat	78152801	68457 kWh	2717.85 m3	0.057 m3/h	59.3 °C	46.49 °C		2021-09-25 21:53.33
telegram=|4F442D2C012815781C048D206D71E76322BE767900843005|1113690B0100C1BC020090D40100A925040000000000000000000000B929BF28100A0100D81A04000000000000000000390000002A172912|+166
MyHeat	78152801	68457 kWh	2717.85 m3	0.057 m3/h	59.3 °C	46.49 °C		2021-09-25 21:53.49
telegram=|4F442D2C012815781C048D206E71E76322BE767900843005|1113690B0100C1BC020090D40100A925040000000000000000000000B929BF28100A0100D81A04000000000000000000390000002A172912|+181
MyHeat	78152801	68457 kWh	2717.85 m3	0.057 m3/h	59.3 °C	46.49 °C		2021-09-25 21:54.04
telegram=|4F442D2C012815781C048D206F71E76322BE767900843005|1113690B0100C1BC020090D40100A925040000000000000000000000B929BF28100A0100D81A04000000000000000000390000002A172912|+197
MyHeat	78152801	68457 kWh	2717.85 m3	0.057 m3/h	59.3 °C	46.49 °C		2021-09-25 21:54.20
telegram=|4F442D2C012815781C048D207071E76322BE767900843005|1113690B0100C1BC020090D40100A925040000000000000000000000B929BF28100A0100D81A04000000000000000000390000002A172912|+212
MyHeat	78152801	68457 kWh	2717.85 m3	0.057 m3/h	59.3 °C	46.49 °C		2021-09-25 21:54.35
telegram=|4F442D2C012815781C048D207171E76322BE767900843005|1113690B0100C1BC020090D40100A925040000000000000000000000B929BF28100A0100D81A04000000000000000000390000002A172912|+228
MyHeat	78152801	68457 kWh	2717.85 m3	0.057 m3/h	59.3 °C	46.49 °C		2021-09-25 21:54.51
telegram=|4F442D2C012815781C048D207271E76322BE767900843005|1113690B0100C1BC020090D40100A925040000000000000000000000B929BF28100A0100D81A04000000000000000000390000002A172912|+244
MyHeat	78152801	68457 kWh	2717.85 m3	0.057 m3/h	59.3 °C	46.49 °C		2021-09-25 21:55.07
telegram=|4F442D2C012815781C048D207371E76322BE767900843005|1113690B0100C1BC020090D40100A925040000000000000000000000B929BF28100A0100D81A04000000000000000000390000002A172912|+259
MyHeat	78152801	68457 kWh	2717.85 m3	0.057 m3/h	59.3 °C	46.49 °C		2021-09-25 21:55.22
telegram=|7A442D2C012815781C048D207490E763227C7478|02F9FF1511130406690B010004EEFF07C1BC020004EEFF0890D401000414AA25040084401400000000848040140000000002FD170000026CB929426CBF284406100A01004414D81A0400C4401400000000C480401400000000043B3900000002593517025D0112|+275
MyHeat	78152801	68457 kWh	2717.86 m3	0.057 m3/h	59.41 °C	46.09 °C		2021-09-25 21:55.38

Thank you so much for taking the time to help me.
It is greatly appreciated!

Have a great evening!

weetmuts, 2021-11-06:
Thank you very much for your telegrams! I have pushed support for the multical602 meter.

uberboffin, 2021-11-12:
This appears to be the same as the Multical 402 telegrams. Issue raised. Many thanks.

closed 2022-01-13 by weetmuts