wmbusmeters/wmbusmeters #346: Support for Itron/EquaScan

BubbleRep, 2021-09-27:


weetmuts, 2021-09-28:
Sure can, but these telegrams are encrypted, do you have a key to your meter?

BubbleRep, 2021-09-30:
huh thought || was indicator for unencrypted telegrams, tried to get the key today, sadly they wouldn't give it to me :(

BubbleRep, 2021-09-30:
Received telegram from: 13010647
          manufacturer: (ITW) ITRON Water, France (0x2697)
                  type: Water meter (0x07)
                   ver: 0x02
      Concerning meter: 20529976
          manufacturer: (ITW) Unknown (0xa697)
                  type: Reserved for system devices (0x30)
                   ver: 0x00
                device: rtl433[00000001]
                  rssi: 999 dBm
                driver: unknown!
telegram=||4E449726470601130207727699522097A60030330030A5E704DFE72E18E4893BB5197252E4322EFC590269C9979B060CFE2DBA87EE138CF08D448F4E350372BB874F539DFCF04B0E79755447020000|+79

Key seems to be B9A3E6E7559A8FC776E25324E47C13A8

weetmuts, 2021-10-01:
Cool, I can decrypt this telegram but not decode it. It looks like this is a bidrectional protocol. I.e. you send a telegram to the meter, then it responds, ie C2 mode. So I am not sure if this is the response from the meter or the selection for readout request from some system that queries the meter.
 
This will be the first meter for which we implement C2. Wmbusmeters has all the infrastructure in place to implement polling of such a meter, but you need a dongle like im871a/amb8465 to actually transmit the request. But we need to implement the protocol properly.

I need to see both the request telegram (that triggers the meter to transmit) and the response telegram from the meter.
Can you collect some more telegrams?

BubbleRep, 2021-10-01:
i sure can log more telegrams, only from that meter or all telegrams near by?

if only this meter, is there a way to log telegrams only for that meter? adding the meter id seemed to not log the telegrams anymore.

Anas92230, 2021-11-02:
> Received telegram from: 13010647
>           manufacturer: (ITW) ITRON Water, France (0x2697)
>                   type: Water meter (0x07)
>                    ver: 0x02
>       Concerning meter: [...]

Hello
I have an Itron Cyble5 and I catch nothing :-( can you give me the command? And other question : whats is the repetition time?

Anas92230, 2021-11-03:
> Hi, possible to add support for the Water Meter Allmess/Itron EquaScan wMIURF
> 
> https://www.itron.com/-/media/feature/products/documents/brochure/equascan-wmiuhe00490en0518kl.pdf
> 
> Received telegram from: 13010657
>           manufacturer: (ITW) ITRON Water, France (0x2697)
>                   type: Water meter (0x07)
>                    ver: 0x02
>       Concerning meter: 20529976
>           manufacturer: (ITW) Unknown (0xa697)
>                   type: Reserved for system devices (0x30)
>                    ver: 0x00
>                 device: rtl433[00000001]
>                   rssi: 999 dBm
>                 driver: unknown!
> telegram=||4E449726570601130207727699522097A60030970030A53CB944BAC4824BFAF7299F51B1995D802384C02FCE212784A1FEF2B081C4003BEB2E1B5FFA5239F118799C564D1EAE110E79684236020000|+64
> Received telegram from: 13029621
>           manufacturer: (ITW) ITRON Water, France (0x2697)
>                   type: Water meter (0x07)
>                    ver: 0x02
>                 device: rtl433[00000001]
>                   rssi: 999 dBm
>                 driver: unknown!
> telegram=||3A4497262196021302077A8F0020A567C33972B57BD1A23A8944290FE6558D667F070F95D3D315AD269D0988BBA997027F8E2A0E79604236020000|+67
> Received telegram from: 12021057
>           manufacturer: (ITW) ITRON Water, France (0x2697)
>                   type: Heat Cost Allocator (0x08)
>                    ver: 0x02
>                 device: rtl433[00000001]
>                   rssi: 999 dBm
>                 driver: unknown!
> telegram=||2E4497265710021202087A6F0020A52A1FF19896EC7DE594A868CDD54F0631317EEBE4E2058B2B2347783FCC2BF49F|+70
> Received telegram from: 12021048
>           manufacturer: (ITW) ITRON Water, France (0x2697)
>                   type: Heat Cost Allocator (0x08)
>                    ver: 0x02
>                 device: rtl433[00000001]
>                   rssi: 999 dBm
>                 driver: unknown!
> telegram=||2E4497264810021202087A7D0020A599221EC980DD589B449A30BC82F2C3CB6156546D5FC638D64F694555E4F0F80C|+70
> Received telegram from: 12020071
>           manufacturer: (ITW) ITRON Water, France (0x2697)
>                   type: Heat Cost Allocator (0x08)
>                    ver: 0x02
>                 device: rtl433[00000001]
>                   rssi: 999 dBm
>                 driver: unknown!
> telegram=||2E4497267100021202087A5A0020A5C79C11EB25C6ACDFDE9F8159ADA4A2BDA6495856A435CAC38D037412621BF7BF|+73
> Received telegram from: 12020550
>           manufacturer: (ITW) ITRON Water, France (0x2697)
>                   type: Heat Cost Allocator (0x08)
>                    ver: 0x02
>                 device: rtl433[00000001]
>                   rssi: 999 dBm
>                 driver: unknown!
> telegram=||2E4497265005021202087A690020A5EF342F524F578BAE8CA3C850C6F90C7D7C8FB292F49002960E77D591E4F694E9|+74
> Received telegram from: 13029623
>           manufacturer: (ITW) ITRON Water, France (0x2697)
>                   type: Water meter (0x07)
>                    ver: 0x02
>                 device: rtl433[00000001]
>                   rssi: 999 dBm
>                 driver: unknown!
> telegram=||3A4497262396021302077AAD0020A529EBE6F5FA3641EAED537AACE03AB3728582DC2021776E18D9DEFE8B23F89A2E027F8E2A0E79474236020000|+78
> Received telegram from: 12019862
>           manufacturer: (ITW) ITRON Water, France (0x2697)
>                   type: Heat Cost Allocator (0x08)
>                    ver: 0x02
>                 device: rtl433[00000001]
>                   rssi: 999 dBm
>                 driver: unknown!
> telegram=||2E4497266298011202087A070020A506CC5F24AE08DC0AD73AE050FFE99F7613DBF05D460E5E96149A6AD09C32580E|+79
> Received telegram from: 13010647
>           manufacturer: (ITW) ITRON 
@BubbleRep can you give me some detail of your command? I have an ITRON Cyble HRF 868M. Thanks for your help

xaelsouth, 2021-11-03:
> Cool! Buy an im871a dongle and we can start talking c2 to it!
> 
> In the meantime: run rtlwmbus tuned to 869.525 MHz for a week to capture the transmissions going to the meter! and run cul at the same time collecting all 868.95 Mhz telegrams coming from the meter. (Cul needs to be pretty close to the meter when listening to c1 telegrams.)
> 
> @xaelsouth Is this the best way to capture C2 tx/rx telegrams? Or can rtl_wmbus be tweaked to capture both at the same time?

rtl_wmbus does not support C2 mode out of the box. As I know the bitrate utilized in the C2 mode is just a half of that used in the the C1 mode (50kbps). Sure, rtl_wmbus could be tweaked for capture C2 datagrams too, but I have never seen a C2 device and would need some raw datagram samples to play with.

weetmuts, 2022-06-19:
@BubbleRep @Anas92230 Have you had any luck in getting keys to your meters? All the telegrams above are encrypted.

BIBOLV, 2023-01-01:
Closing due to inactivity. Please reopen issue if you was able to get decryption key and post unencrypted telegrams.

closed 2023-01-01 by BIBOLV

jonathanD91, 2023-03-18:
Hi, it’s possible to say the configuration for this water? I’ve the same but i don’t say wath i write in my wmbusmeters.conf