wmbusmeters/wmbusmeters #349: [ALARM DeviceFailure] too many protocol errors(20)

LarsBorris, 2021-10-01:
Hello,

Just wanted to know if this was normal behaviour
I have attached the wmbusmeters.log and the latest metering log files.
I can see that the last reset was at 07:52, but the last metering entry was at 05:59
Has my dongle stopped working?

[TEST-Device_2021-10-01.log](https://github.com/weetmuts/wmbusmeters/files/7264569/TEST-Device_2021-10-01.log)
[wmbusmeters.log](https://github.com/weetmuts/wmbusmeters/files/7264570/wmbusmeters.log)

Thanks!

weetmuts, 2021-10-02:
Thanks!
What version of wmbusmeters are you running?
Are you running on a raspberry pi?
Does this occur after the reboot of the pi? If you stop the daemon and start it again, does it work then?
If you stop the daemon and start wmbusmeters manually does it work then?
Can you enable loglevel=debug and capture the problem again?

Thanks!

LarsBorris, 2021-10-05:
Hello,

It looks like it is still running.
I let it run for a few days and I had data for every day.
I am running it on a Rpi.
So I do think that the problem was not a problem at all.

However I have noticed this:
When running the wmbusmeters with no meters configured, I get a crash.
I use this method to see how many meters I can detect, then afterwards I can add the meters and decrypt the telegram.

I start the program using this command (with no meters configured so I can see how far it reaches):
sudo wmbusmeters /dev/ttyUSB0:amb8465:c1,t1
Started config amb8465[00000000] on /dev/ttyUSB0 listening on c1,t1
No meters configured. Printing id:s of all telegrams heard!

It will output telegrams and work fine for 10 minutes then show this:

[ALARM DeviceFailure] too many protocol errors(20) resetting /dev/ttyUSB0 amb8465
(wmbus) resetting /dev/ttyUSB0
Warning! Did not get confirmation on set link mode for amb8465
(wmbus) successfully reset wmbus device

It will recover and start outputting telegrams again
After 10 minutes or so it will crash:

[ALARM DeviceFailure] too many protocol errors(20) resetting /dev/ttyUSB0 amb8465
(wmbus) resetting /dev/ttyUSB0
wmbusmeters: src/wmbus.cc:3852: bool WMBusCommonImplementation::waitForResponse(int): Assertion `waiting_for_response_id_ == 0 && id != 0' failed.
Aborted

The full log is attached.
Thanks!
[wmbusmeters-scan.txt](https://github.com/weetmuts/wmbusmeters/files/7284987/wmbusmeters-scan.txt)

BIBOLV, 2023-01-01:
@weetmuts - were you able to reproduce issue?

closed 2023-06-02 by github-actions[bot]