wmbusmeters/wmbusmeters #350: HA addon additional parameters for mosquitto_pub [not a issue]

reaper7, 2021-10-03:
Is it possible to add parameters for mosquitto_pub in HA addon?
Exactly the additional RETAIN flag as an mqtt option, when configuring this add-on.

This would solve the case of unknown values when restarting the HA, while the meters are not sending information (over the weekend)

reaper7, 2021-10-10:
@BIBOLV - it would be better if it could be written as option, hardcoded may not suit everyone.

BIBOLV, 2021-10-10:
Yes, it makes more sense, canceled PR. Leaving for @to-masz to think about implementstion approach.

weetmuts, 2022-09-03:
Do we need to follow up on this one, or can it be closed? :-)

SaSa1983, 2022-11-02:
I'd like to have support for the retain parameter.
Thus upon restart of HA (e.g. after an update) the sensor value willnot fallback to "unavailable" until a packetis received, but rather stay on the previous value

SaSa1983, 2022-11-08:
@weetmuts Is there any plan about the retain option?

BIBOLV, 2023-03-12:
Retain flag is present in mqtt script by default - https://github.com/wmbusmeters/wmbusmeters-ha-addon/blob/main/wmbusmeters-ha-addon-edge/run.sh#L89 (-r = retain)
and it can be confirmed by looking at mqtt message itself:
[image]

BIBOLV, 2024-02-14:
Is this question still actual or we can close this ticket?

SaSa1983, 2024-02-15:
As far as I know there's still no option in the current release version of the addon

BIBOLV, 2024-02-15:
As I explained - retain flag is on by default, then you want possibility to remove it, correct?