wmbusmeters/wmbusmeters #351: HydroClima RFM suported?

djmalin, 2021-10-05:
Hello, this is a heat allocator Wireless MBUS (PN-EN 13757) B Meters [(](https://bmeters.pl/pl/hydroclima_rfm/48/)
is it possible to read data from it?

weetmuts, 2021-10-05:
Probably! Please collect a telegram and post it here and we can take a look!

djmalin, 2021-10-05:
@weetmuts 
Received telegram from: 74393732
          manufacturer: (BMP) BMETERS Polska, Poland (0x9b0)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x33
                device: rtlwmbus[00000001]
                  rssi: 63 dBm
                driver: unknown!
telegram=||2D44B009323739743308780F0513000255FBFF999CB0D1BBABB48EBEABB4ABB4ABB4ABB4ABB4ABB4ABB4CAB4ABBC|+10

weetmuts, 2021-10-16:
Thanks! When I decode this telegram it looks like most of the content is proprietary:
Can you collect some more telegrams? Some HCA:s alternate between OMS-compliant telegrams and proprietary telegrams. The HydroClima OMS 2.9 should be easy to support, since it is OMS compliant. But this meter is not?
 
telegram=|2D44B00932373974330878|0F0513000255FBFF999CB0D1BBABB48EBEABB4ABB4ABB4ABB4ABB4ABB4ABB4CAB4ABBC|+0
(auto) 00: 2d length (45 bytes)
(auto) 01: 44 dll-c (from meter SND_NR)
(auto) 02: b009 dll-mfct (BMP)
(auto) 04: 32373974 dll-id (74393732)
(auto) 08: 33 dll-version
(auto) 09: 08 dll-type (Heat Cost Allocator)
(auto) 0a: 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
(auto) 0b: 0F manufacturer specific data 0513000255FBFF999CB0D1BBABB48EBEABB4ABB4ABB4ABB4ABB4ABB4ABB4CAB4ABBC

djmalin, 2021-11-10:
Hello @weetmuts any news from my itergration?

djmalin, 2021-12-07:
@weetmuts  ?

weetmuts, 2022-05-05:
Ok, I just added a hydroclima driver for the Hydroclima RFM HCA (https://github.com/weetmuts/wmbusmeters/issues/534)

unfortunately I cannot decode your telegrams. Your telegrams are from version 0x33 but the ones
that I know how to decode are version 0x53.

weetmuts, 2022-05-05:
Closing this one. Lets continue the discussion in https://github.com/weetmuts/wmbusmeters/issues/434
:-)

closed 2022-05-05 by weetmuts