wmbusmeters/wmbusmeters #352: Qundis Q module 5.5 water Modularis is supported by lse_07_17

scy, 2021-10-05:
Hi there!

In my apartment, there are water meters supplied with [Q module 5.5 add-on modules](https://qundis.com/products/modules/) that make them C-mode enabled.

These send a manufacturer of `(QDS) Qundis, Germany (0x4493)`, type `Radio converter (meter side) (0x37)` and `ver: 0x33`. I’ve tried setting the driver to `lse_07_17` (the QWater 5.5 introduced in #277), and it worked instantly 😁 🎉 

The cold water meter correctly reports `media: water` and the correct `total_m3` value; the warm water meter (same mfct/media/version numbers) reports `media: warm water` and the correct `total_m3`, too.

This is awesome, I got my RTL-SDR today and just wanted to track my heat cost allocator values, but I got my water meters at no additional cost, too 😉  Thanks for this project!

weetmuts, 2021-10-16:
Great to hear that! I have added the new mapping to the LSE_07_17 driver from QDS 0x37 0x33.
Please test if you have the time!

weetmuts, 2021-11-02:
If you have a telegram, please post it here!

We need it for the regression tests! Thanks!

scy, 2021-11-02:
I’ll test it and supply you with a telegram, but it will take a few days until I find the time, sorry :)

weetmuts, 2021-11-02:
No worries! Thanks!

scy, 2021-12-29:
Took me some time, but here are your telegrams. :) Note that I did _not_ update wmbusmeters to a version with your change from Oct 16, so the meters are still detected as “unknown”.

First, two random meters from somewhere inside the house.

```console
$ rtl_sdr -f 868.95M -s 1.6M - | rtl_wmbus -p s -a | wmbusmeters --logtelegrams stdin:rtlwmbus
[…]
Received telegram from: 35207125
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x33
      Concerning meter: 35207125
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Warm Water (30°C-90°C) meter (0x06)
                   ver: 0x33
                device: rtlwmbus[]
                  rssi: 16 dBm
                driver: unknown!
telegram=||3C449344257120353337722571203593443306420000200C13870200004C1364000000426C9F2CCC081387020000C2086CBE2B326CFFFF046D0B00BE2C|+15
[…]
Received telegram from: 35207063
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x33
      Concerning meter: 35207063
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Water meter (0x07)
                   ver: 0x33
                device: rtlwmbus[]
                  rssi: 77 dBm
                driver: unknown!
telegram=||3C449344637020353337726370203593443307850000200C13217800004C1304020000426C9F2CCC081373520000C2086CBE2B326CFFFF046D0F00BE2C|+105
```

Next, the meters in my apartment.

```console
$ rtl_sdr -f 868.95M -s 1.6M - | rtl_wmbus -p s -a | wmbusmeters --logtelegrams stdin:rtlwmbus cold lse_07_17 35207059 NOKEY warm lse_07_17 35207060 NOKEY
[…]
(meter) cold: meter detection did not match the selected driver lse_07_17! correct driver is: unknown!
(meter) Not printing this warning agin for id: 35207059 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x33
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|3C44934459702035333772597020359344330764000020|0C13339501004C1331180000426C9F2CCC081398780100C2086CBE2B326CFFFF046D1300BE2C|+25
cold    35207059        19.533 m3       1.831 m3        2020-12-31      OK      2127-15-31      2021-12-30 00:19        2021-12-29 23:15.27
(meter) warm: meter detection did not match the selected driver lse_07_17! correct driver is: unknown!
(meter) Not printing this warning agin for id: 35207060 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x33
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|3C44934460702035333772607020359344330671000020|0C13770601004C1353040000426C9F2CCC081305960000C2086CBE2B326CFFFF046D1600BE2C|+338
warm    35207060        10.677 m3       0.453 m3        2020-12-31      OK      2127-15-31      2021-12-30 00:22        2021-12-29 23:20.40
```

Is there anything else you need?

patrislav1, 2022-09-13:
I see similar stuff with my Q module 5.5 water:

```
(wmbusmeters) logging started 2022-09-13 09:30:27 using 1.9.0-13-gc15f959
[2022-09-13_09:30:27] Started config rtlwmbus listening on none using CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)
(wmbusmeters) waiting for telegrams
[2022-09-13_09:30:29] (memory) rss 4390912 peak 4.18 MiB
[...]
[2022-09-13_09:36:26] (meter) water: meter detection could not find driver for id: 26966667 mfct: (QDS) Qundis, Germany (0x4493) type: Water meter (0x07) ver: 0x35
[2022-09-13_09:36:26] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2022-09-13_09:36:26] (meter) to add support for this unknown mfct,media,version combination
(wmbusmeters) started meter 6 (water 26966667 auto)
[2022-09-13_09:36:26] (meter) water: meter detection did not match the selected driver unknown! correct driver is: unknown!
(meter) Not printing this warning again for id: 36682268 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x35
[2022-09-13_09:36:26] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2022-09-13_09:36:26] (meter) to add support for this unknown mfct,media,version combination
telegram=|3C449344682268363537726766962693443507AF000020_0C13959711004C1361100300426CBF2CCC081344501100C2086CDF28326CFFFF046D240ACD29|+359
```

closed 2023-04-06 by github-actions[bot]