wmbusmeters/wmbusmeters #354: telegram contains a register (71) with unknown size.

KrzysiekLegnica, 2021-10-06:
Debian 11 and RTL2832 R820T2 
Home Assistant (core-2021.9.7, supervisor-2021.09.6), 
Wmbusmeters (W-MBus to MQTT) Current version: 0.2.19

Log:
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Syncing wmbusmeters configuration ...
Registering meters ...
Adding meter-0001 ...
Generating MQTT configuration ... 
Broker mqtt://core-mosquitto will be used.
Running wmbusmeters ...

Listening telegram: 
Received telegram from: 03098143
          manufacturer: (APA) Apator, Poland (0x601)
                  type: Water meter (0x07)
                   ver: 0x05
                device: rtlwmbus[00000001]
                  rssi: 107 dBm
                driver: apator162
                
meters:
  - |-
    name=MainWater
    driver=apator162
    id=03098143
    key=00000000000000000000000000000000          
    
After adding the meter, I get the following data

 Started auto rtlwmbus[00000001] listening on c1
(apator162) telegram contains a register (71) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 3E4401064381090305077AB10030852F2F0F74467A959C0000435601875EA2B6D88AB706B90F001003380100710126320100880E0100A0D3E3EF03FFFF5881
Unable to connect (Lookup error.).
(shell) /bin/sh exited with non-zero return code: 1
(apator162) telegram contains a register (71) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 3E4401064381090305077AB20030852F2F0F75467A959C000043560182B753B2C458B5FF3D38011003380100710126320100880E0100A0D3E3EF03FFFF54F6
Unable to connect (Lookup error.).
(shell) /bin/sh exited with non-zero return code: 1

weetmuts, 2021-10-06:
Thanks! That prefix seems to indicate a length of 9 bytes. Fix pushed. Please test.

KrzysiekLegnica, 2021-10-06:
There are still some bugs

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Syncing wmbusmeters configuration ...
Registering meters ...
Adding meter-0001 ...
Generating MQTT configuration ... 
Broker mqtt://core-mosquitto will be used.
Running wmbusmeters ...
Started auto rtlwmbus[00000001] listening on c1
Unable to connect (Lookup error.).
(shell) /bin/sh exited with non-zero return code: 1
Unable to connect (Lookup error.).
(shell) /bin/sh exited with non-zero return code: 1
Unable to connect (Lookup error.).
(shell) /bin/sh exited with non-zero return code: 1
Unable to connect (Lookup error.).
(shell) /bin/sh exited with non-zero return code: 1
Unable to connect (Lookup error.).
(shell) /bin/sh exited with non-zero return code: 1
Unable to connect (Lookup error.).
(shell) /bin/sh exited with non-zero return code: 1

weetmuts, 2021-10-08:
That is your own mqtt setup that does not work.