wmbusmeters/wmbusmeters #355: Support Diehl Hydrus Type 173

zivillian, 2021-10-06:
Yesterday I've got new main water meter (Diehl Hydrus Type 173 [manual](https://www.diehl.com/cms/files/542418-DE-DE-20201103_Benutzerhandbuch_HYDRUS_2.0_HWZ_DE.pdf?download=1)) which is not yet supported. I have also got two additional IZAR RC 868 I R4, which are working fine with `NOKEY`, so I guess that my main meter also has no key.

My meter looks like the one linked in https://github.com/weetmuts/wmbusmeters/issues/338#issuecomment-923052690, but I get totally different data:

```
$  wmbusmeters --debug auto:t1 HWZ auto 72486239 NOKEY
...
(serial) received binary "A5C2033E44A5113962487276077A390030050E8B300AF0467CA94D69F7E18F484CBAFE044E5B564A81D9CB024C105952E8796D6ED9F1E4EB94E70A9B32B35007B25661594B"
(im871a) checkIM871AFrame "A5C2033E44A5113962487276077A390030050E8B300AF0467CA94D69F7E18F484CBAFE044E5B564A81D9CB024C105952E8796D6ED9F1E4EB94E70A9B32B35007B25661594B"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) rssi 97 (-74 dBm)
(im871a) got crc16 4b59 expected 4b59
(im871a) received full frame
(meter) no meter handled  checking 1 templates.
(wmbus) parseDLL @0 63
(wmbus) parseELL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(meter) HWZ: for me? 72486239 in
(meter) HWZ: yes for me
(meter) HWZ: meter detection could not find driver for id: 72486239 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x76
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) created "HWZ" "unknown" "72486239" encrypted
(meter) used meter template HWZ  auto to match 72486239
(meter) started meter 1 (HWZ 72486239 auto)
(wmbus) parseDLL @0 63
(wmbus) parseELL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(meter) HWZ: for me? 72486239 in 72486239
(meter) HWZ: yes for me
(meter) HWZ auto handling telegram from 72486239
(meter) HWZ 72486239 "3E44A5113962487276077A390030050E8B300AF0467CA94D69F7E18F484CBAFE044E5B564A81D9CB024C105952E8796D6ED9F1E4EB94E70A9B32B35007B256"
(wmbus) parseDLL @0 63
(telegram) DLL L=3e C=44 (from meter SND_NR) M=11a5 (DME) A=72486239 VER=76 TYPE=07 (Water meter) (driver unknown!) DEV=im871a[00102741] RSSI=-74
(wmbus) parseELL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(TPL) AES CBC IV decrypting "0E8B300AF0467CA94D69F7E18F484CBAFE044E5B564A81D9CB024C105952E8796D6ED9F1E4EB94E70A9B32B35007B256"
(TPL) num encrypted blocks 3 (48 bytes and remaining unencrypted 0 bytes)
(TPL) IV A5113962487276073939393939393939
(TPL) decrypted  "8A5E199CABAA042975E2525884821CDFF140CBB14E72E0071B5517CA401F7EFC4A97089CB71EAD3CAAF51A90DD4D5619"
(wmbus) decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 72486239 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x76
(telegram) TPL CI=7a ACC=39 STS=00 CFG=0530 (AES_CBC_IV nb=3 cntn=0 ra=0 hc=0)
(meter) newly created meter (HWZ 72486239 auto) did not handle telegram!
(wmbus) telegram from 72486239 ignored by all configured meters!
```

Please let me know, what information I can supply, to help supporting this meter.

weetmuts, 2021-11-06:
Thanks! Yes, the key is necessary.

weltenwort, 2021-11-20:
I just received a new Hydrus Type 173 water meter as well. The water supplier "Wasser Nord" in Brandenburg, Germany refuses to hand out the keys claiming it's the same for all meters in the area. Please update us in case you manage to convince your supplier to provide the key so we can bring some transparency into the state of the deployment.

closed 2022-05-25 by weetmuts

paul402pd2, 2022-05-26:
A six months ago I got a Type 173 meter. Last Friday I discovered that we have a boiler leak and it's dumping 18 liters/day into the cellar floor under the flooring. The leak got fixed yesterday but has probably been running for months. It's going to be expensive. After the installation I asked the city about being able to read the meter on an hourly basis and they said it wasn't possible. What do I need to do/install to be able to read the overnight flows or any indication of flows?

zivillian, 2022-05-26:
It's a dirty and unreliable hack, but maybe it works for you: https://github.com/weetmuts/wmbusmeters/issues/281#issuecomment-1130634687

paul402pd2, 2022-05-28:
Thanks! I'd like to try it. What 868MHz receiver do you recommend? I'm running Domoticz on a Raspi as a home control system.