wmbusmeters/wmbusmeters #367: Tpl status bit in json

jacek27, 2021-10-16:
Is there a way to get the status bit from tpl part of the frame in the output json?

weetmuts, 2021-10-16:
Yes, sometimes the tpl status bits are put into the status field. Like in meter_ei6500.cc line 251:
string s = decodeTPLStatusByte(tpl_sts_, &error_codes_);
and merged with another status field into the json field status="OK" or status="POWER_LOW REMOVED" etc

We also have meter_hydrus.c where the manufacturer specific bits of the tpl_sts_ field are translated.
E.g. status="AIR_IN_PIPE BATTERY" etc

What is missing is that the tpl_sts bits always should be part of the json for all meter drivers, not just hydrus,
waterstarm and ei6500.

Perhaps we can add the status field to all meters by adding a common status_ string and addPrint and decoding inside wmbus.cc and/or meters.cc in some common code.

jacek27, 2021-10-16:
I'll look into it. I see value in this as home assistant can notify me if the status bit had changed. This indicates an issue of some sort, at least in general.

weetmuts, 2022-05-21:
Hi @jacek27 ! I have finally solved the problem with the tpl status byte.

I wanted a single "status" field instead of adding a "tpl_status" field in addition to the already existing "status" field.
The existing "status" field usually picks it values from difvif entries like 02FF20 (vendor bits) or 02FD17 (second vif extension error flags) and translates the bits into a string. If no bits are set, then the "status":"OK" is generated.

So I added the PrintProperty::JOIN_TPL_STATUS this will include in the same string, the default decoded
tpl status bits (0x01 to 0x10) which can generate  BUSY, ERROR, ALARM, POWER_LOW, PERMANENT_ERROR, TEMPORARY_ERROR. Also the top 3 bits are mfct specific and you can add a lookup for these
bits in the driver constructor. 

Read the driver_ei6500.cc where the tpl sts mfct bits can be specified at  di.addMfctTPLStatusBits({..})
but as far as I know, the ei6500 does not make use of these bits so the actual lookup for these is commented out.

Then you have another lookup for the error flags 02FD17 further down for the field "status".

This is the new driver format which is much easier to write.

So in the case of ei6500 the tpl status byte could have the ALARM 0x03 bits set and the
02FD17 field could have the 0x0100 bit OBSTACLE_DETECTED set. The result would be:

"status":"ALARM OBSTACLE_DETECTED"

Whether the ei6500 or not uses both the tpl status and the error flags, I do not know.
But if it does, then this is what happens.

Now I have to go through all drivers (of the newes driver style) and add the JOIN_TPL_STATUS to the
status field. All the other older drivers are to be converted to the newest style as well.

BIBOLV, 2023-01-01:
All drivers was rewritten to new format. Closing issue. Please reopen if problem is still persistent.

closed 2023-01-01 by BIBOLV