wmbusmeters/wmbusmeters #376: oneshot and exitafter does not seems to work

lemonka, 2021-11-03:
```
wmbusmeters: 1.4.1-63-gbea17e9
bea17e9d3d63c9d70248655431176db39a269446
```
Not much to add but below command didnt worked for me - wmbusmeters process was working constantly - I have not idea if this is caused by using config (which does not have anything related to exitafter or oneshot

```

wmbusmeters --oneshot --useconfig /etc/wmbusmeters.conf
wmbusmeters  --useconfig /etc/wmbusmeters.conf --oneshot
wmbusmeters  --useconfig /etc/wmbusmeters.conf --oneshot --exitafter=3m

```

wmbusmeters.conf:

```
loglevel=normal
# Search for a wmbus device and set it to c1.
device=rtlwmbus
# But do not probe this serial tty.
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
meterfilesnaming=name
meterfilestimestamp=day
logfile=/var/log/wmbusmeters/wmbusmeters.log
shell=/usr/bin/mosquitto_pub -h localhost -u homeassistant -P  somepassword -t wmbusmeters/$METER_ID -m "$METER_JSON"
alarmshell=/usr/bin/mosquitto_pub -h localhost -u homeassistant -P somepassword -t wmbusmeters_alarm -m "$ALARM_TYPE $ALARM_MESSAGE"
alarmtimeout=1h
alarmexpectedactivity=mon-sun(00-23)
ignoreduplicates=false

weetmuts, 2021-11-03:
It is an error, that not the proper error is printed when you try the above. ;-)

Because I think that the only option that is supposed to be combined with --useconfig= is --device=
and --listento= so that you could override the device in the config. However, I see that there might be other
settings that could make sense, like combining useconfig with oneshot and exitafter, as you did.

Do you want to have a go at fixing the code? Look inside cmdline.cc 
It might be a bit tricky though, because the real fix would allow for the config files to be loaded first
then all settings could be overriden from the command line. Essentially pass the entire Config 
gotten from the parse cmdline as a parameter to start_using_config_files instead of just passing
device and listento overrides. The same for loadConfiguration (config.cc) that takes config_override
instead of device and listento_override. And then at the end of that function walk through all overridden
settings that make sense and change the config.

weetmuts, 2021-11-03:
Also the syntax is --useconfig=/
this will automatically look for /etc/wmbusmeters.conf and /etc/wmbusmeters.d

Just --useconfig defaults to --useconfig=/

lemonka, 2021-11-04:
Hi Frederik

you may ask why such approach - well... I have 3 meters (2x water ,1x energy). While Energy meter works 24/7 and send datagram every minute, water meters works only somewhere between 6:00-18:00. 

As SDR device is actually TunerTV its works constantly as I would be watching TV on it and consumes CPU (approx 10%). With other running processes it reaches in peaks up to 20% of CPU and that is the moment when my cooling is turning on. Its laptop business series so fans are not so loud but having them turn on randomly at night is little unpleasant. On the other hand constant measurement of Power Consumption is nice and we can plot fancy graphs - but I think I don't really need it :). So I decided to perform once per hour measurements.  So my plan was to to temporary move water configs between 18-06 and perform oneshot operation with timeout just in case. 

As a workaround I have found timeout command in bash. 

When I will have more free time I will try to work with code but my C is more than rust ;)

weetmuts, 2022-02-05:
I have now pushed code for this. It was a bit more complicated than expected.
You should now be able to do:
wmbusmeters --useconfig=dir_with_conf --oneshot --exitafter=5m

Please test!

ottmi, 2022-02-08:
It doesn't seem to work when more than 1 meter has been defined, see PR #473 
But with the above fix it works reliably for me.

grzegorzjj, 2022-02-09:
Hi, My issue is quite similiar. When run as a demon all meters are working fine ( 6 files in config file in standard path in /etc/wmbusmeters.d ) 6 output files are regulary updated - overwriten - every 2-4 minutes. 
When I run frm a comandline: 
wmbusmeters --useconfig=/ 
or 
wmbusmeters --useconfig=/ --exitafter=3m
it reads only ONE meter ( random 1 of 6 - probably this one that puts the first telegram ) and exits. It lasts 20-30 seconds.
The same behavior with --oneshot option in command line or oneshot=true in config file.
Compiled yesterday version.
What can be wrong - I would like to read once all the meters ( 6 of them ) and exit.

weetmuts, 2022-02-09:
I have merged @ottmi  s pull request! Please test! It looks like it should fix the problem!

grzegorzjj, 2022-02-09:
I'm not quite familiar with Your answer. Should I clone & recompile all the code ? [ Sorry for that... ]

weetmuts, 2022-02-09:
git pull; make clean; make

grzegorzjj, 2022-02-09:
Before: wmbusmeters: 1.6.0-71-gda943d3
Afetr: wmbusmeters: 1.6.0-74-gfe936f5

Works!  Exits after reading at least once all the meters.  Thank You.

closed 2022-02-26 by weetmuts