wmbusmeters/wmbusmeters #377: Docker - no data with log settings, no data with mqtt publish ??

zibous, 2021-11-03:
Docker Version: Docker version 20.10.8, build 3967b7d

```
docker run -d --privileged \
    --name=wmbusmeters \
    --restart=always \
    -v /etc/localtime:/etc/localtime:ro \
    -v /docker/wmbusmeters:/wmbusmeters_data \
    -v /dev/:/dev/ \
    weetmuts/wmbusmeters
```

Config:
```
loglevel=debug
device=auto:t1
device=/dev/ttyACM1
logtelegrams=true
format=json
## logging
##meterfiles=/var/log/wmbusmeters/meter_readings
##meterfilesaction=append
##logfile=/var/log/wmbusmeters/wmbusmeters.log
## additonal settings
##meterfilesnaming=name
##meterfilestimestamp=hour
##ignoreduplicates=true
alarmtimeout=1h
alarmexpectedactivity=mon-sun(00-23)
## MQTT Brocker settings
shell=/usr/bin/mosquitto_pub -h mbs.home -p 1883 -u "******" -P "*******" -t tele/watermeter>
alarmshell=/usr/bin/mosquitto_pub -h mbs.home -p 1883 -u "****" -P "******" -t tele/watermeter>
```

With these settings I get data in the Docker log, **but no MQTT messages**. 
```

(shell) /bin/sh exited with non-zero return code: 2,
(shell) env "METER_ID=18444343",
(shell) env "METER_MEDIA=water",
(shell) env "METER_TYPE=izar",
(shell) env "METER_NAME=watermeter",
(shell) env "METER_TIMESTAMP=2021-11-03T11:54:57Z",
(shell) env "METER_TIMESTAMP_UTC=2021-11-03T11:54:57Z",
(shell) env "METER_TIMESTAMP_UT=1635940497",
(shell) env "METER_TIMESTAMP_LT=2021-11-03 12:54.57",
(shell) env "METER_DEVICE=cul",
(shell) env "METER_RSSI_DBM=-65",
(shell) env "METER_PREFIX=",
(shell) env "METER_SERIAL_NUMBER=000000",
(shell) env "METER_TOTAL_M3=312.765",
(shell) env "METER_LAST_MONTH_TOTAL_M3=312.412",
(shell) env "METER_LAST_MONTH_MEASURE_DATE=2021-11-01",
(shell) env "METER_REMAINING_BATTERY_LIFE_Year=12",
(shell) env "METER_CURRENT_ALARMS=no_alarm",
(shell) env "METER_PREVIOUS_ALARMS=no_alarm",
(shell) env "METER_TRANSMIT_PERIOD_Second=8",
(shell) env "METER_MANUFACTURE_YEAR=0",
(shell) waiting for child 88 to complete.,
/bin/sh: syntax error: unexpected end of file,
(shell) /bin/sh: return code 2,
(shell) arg "/usr/bin/mosquitto_pub -h mbs.home -p 1883 -u "******" -P "*****" -t tele/watermeter>",
(shell) env "METER_JSON={"media":"water","meter":"izar","name":"watermeter","id":"18444343","prefix":"","serial_number":"000000","total_m3":312.765,"last_month_total_m3":312.412,"last_month_measure_date":"2021-11-01","remaining_battery_life_y":12,"current_alarms":"no_alarm","previous_alarms":"no_alarm","transmit_period_s":8,"manufacture_year":"0","timestamp":"2021-11-03T11:54:57Z","device":"cul","rssi_dbm":-65}"
```

MQTT Broker is installed on the computer and works perfectly.

When I activate the log files in the configuration, I get an error message `Could not open log file.` and no more data is evaluated.

```
(config) "logfile" "/var/log/wmbusmeters/wmbusmeters.log",
(config) "## additonal settings" "",
(config) "meterfilesnaming" "name",
(config) "meterfilestimestamp" "hour",
(config) "ignoreduplicates" "true",
(config) "alarmtimeout" "1h",
(config) "alarmexpectedactivity" "mon-sun(00-23)",
(config) "## MQTT Brocker settings" "",
(config) "shell" "/usr/bin/mosquitto_pub -h mbs.home -p 1883 -u "*****" -P "*****" -t tele/watermeter>",
(config) "alarmshell" "/usr/bin/mosquitto_pub -h mbs.home -p 1883 -u "******" -P "*****" -t tele/watermeter>",
(config) "" "",
(config) loading meter file /wmbusmeters_data/etc/wmbusmeters.d/diehl,
(config) name=watermeter,
(config) type=izar,
(config) id=18444343,
(config) key=<notprinted>,
(cmdline) setting link modes to t1 for meter watermeter,
(wmbusmeters) using log file /var/log/wmbusmeters/wmbusmeters.log,
Could not open log file.,
(config) "device" "auto:t1",
(config) "device" "/dev/ttyACM1",
(config) "logtelegrams" "true",
(config) "format" "json",
(config) "## logging" "",
(config) "meterfiles" "/var/log/wmbusmeters/meter_readings",
Cannot write meter files into dir "/var/log/wmbusmeters/meter_readings",
(config) "meterfilesaction" "append",
(config) "logfile" "/var/log/wmbusmeters/wmbusmeters.log",
(config) "## additonal settings" "",
(config) "meterfilesnaming" "name",
(config) "meterfilestimestamp" "hour",
(config) "ignoreduplicates" "true",
(config) "alarmtimeout" "1h",
(config) "alarmexpectedactivity" "mon-sun(00-23)",
(config) "## MQTT Brocker settings" "",
(config) "shell" "/usr/bin/mosquitto_pub -h mbs.home -p 1883 -u "******" -P "******" -t tele/watermeter>",
(config) "alarmshell" "/usr/bin/mosquitto_pub -h mbs.home -p 1883 -u "*****" -P "******" -t tele/watermeter>",
(config) "" "",
(config) loading meter file /wmbusmeters_data/etc/wmbusmeters.d/diehl,
(config) name=watermeter,
(config) type=izar,
(config) id=18444343,
(config) key=<notprinted>,
(cmdline) setting link modes to t1 for meter watermeter,
(wmbusmeters) using log file /var/log/wmbusmeters/wmbusmeters.log,
Could not open log file.,
(config) "device" "auto:t1",
(config) "device" "/dev/ttyACM1",
(config) "logtelegrams" "true",
(config) "format" "json",
(config) "## logging" "",
(config) "meterfiles" "/var/log/wmbusmeters/meter_readings",
Cannot write meter files into dir "/var/log/wmbusmeters/meter_readings",
(config) "meterfilesaction" "append",
(config) "logfile" "/var/log/wmbusmeters/wmbusmeters.log",
(config) "## additonal settings" "",
(config) "meterfilesnaming" "name",
(config) "meterfilestimestamp" "hour",
(config) "ignoreduplicates" "true",
(config) "alarmtimeout" "1h",
(config) "alarmexpectedactivity" "mon-sun(00-23)",
(config) loading meter file /wmbusmeters_data/etc/wmbusmeters.d/diehl,
(config) name=watermeter,
(config) type=izar,
(config) id=18444343,
(config) key=<notprinted>,
(cmdline) setting link modes to t1 for meter watermeter,
(wmbusmeters) using log file /var/log/wmbusmeters/wmbusmeters.log,
Could not open log file.,
```
**If I run wmbusmeters with the NanoCul without Docker, everything works fine.**

What can I do so that I can fix the error in the log files during the Docker installation and also receive data via MQTT?
Thanks

zibous, 2021-11-03:
> What can I do so that I can fix the error in the log files during the Docker installation and also receive data via MQTT?

Mqtt is now working:

+ Changed Mqtt Broker Host to IP Address
+ add $METER_JSON and $ALARM_TYPE to shell

```
## MQTT Brocker settings
shell=/usr/bin/mosquitto_pub -h 10.1.1.94 -p 1883 -u ****** -P ****** -t tele/watermeter -m "$METER_JSON"
alarmshell=/usr/bin/mosquitto_pub -h 10.1.1.94  -u ****** -P ****** -t   tele/watermeter/_alarm -m "$ALARM_TYPE" $ALARM_MESSAGE"
```

weetmuts, 2021-11-06:
Yes, you need to supply the data to be sent on the command line. 
Good that you got it working! Closing this now.

closed 2021-11-06 by weetmuts

zibous, 2021-11-07:
> Yes, you need to supply the data to be sent on the command line.

Hi @weetmuts,
Thanks, but i have allready the problem with the logging...

```bash
docker run -d --privileged \
    --name=wmbusmeters \
    --restart=always \
    -v /etc/localtime:/etc/localtime:ro \
    -v /docker/wmbusmeters:/wmbusmeters_data \
    -v /dev/:/dev/ \
    weetmuts/wmbusmeters
```
When i activate the `logfile=/var/log/wmbusmeters/wmbusmeters.log` or `/docker/wmbusmeters/logs/wmbusmeters.log`i get the error:

**`Could not open log file, Cannot write meter files into dir "/var/log/wmbusmeters/meter_readings",`**

```log
(cmdline) setting link modes to t1 for meter watermeter,
(wmbusmeters) using log file /var/log/wmbusmeters/wmbusmeters.log,
Could not open log file.,
```

```
/docker/wmbusmeters
├── etc
│   ├── wmbusmeters.conf
│   └── wmbusmeters.d
│       └── diehl
└── logs
    └── meter_readings

4 directories, 2 files
```
The directories are created automatically by the installation. What I don't understand is why the `logs` directory is created and not `log`. 
Could that be the mistake?