wmbusmeters/wmbusmeters #378: Telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed

filipc, 2021-11-06:
I'm trying to read wmbus T1 messages and getting error: 

```
(serial) received ascii "T1;0;0;2021-11-06 11:45:00.000;146;115;FFFFFFFF;0x6e44ff06ffffffffffff7a180060854fff15c848ff4fd3ffffffff1bffff39ffd4ffffffffffffff70ff7dff392f88ffff2bffffffffce0442bfb2ff6dff1063de7ca2d3ffffffffff25ae9fff2b51761b45ff3587ffff76ff5e8a7dffffffffff46ffffedffff5dd4c2ff60cbffdf<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2021-11-06 11:45:00.000;146;115;FFFFFFFF;0x6e44ff06ffffffffffff7a180060854fff15c848ff4fd3ffffffff1bffff39ffd4ffffffffffffff70ff7dff392f88ffff2bffffffffce0442bfb2ff6dff1063de7ca2d3ffffffffff25ae9fff2b51761b45ff3587ffff76ff5e8a7dffffffffff46ffffedffff5dd4c2ff60cbffdf<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
```

What could be the reason? Wrong key?

weetmuts, 2021-11-06:
Yeah, if rtl_wmbus fails, then it is radio wave realted and can be:
bad reception, too far away from the meter, bad antenna, radio noise disturbing
bad reception, not quite the right frequency, you need to tweak your dongle fq or ppm,  read the README.
meter not transmitting normal wmbus data.

What kind of meter are you listening to?

filipc, 2021-11-08:
it's apator162. I moved antenna closer to the meters. Now it works without such errors. 
Thanks

weetmuts, 2021-11-08:
Yay! Closing this! Check this info for how to improve the distance: https://weetmuts.github.io/wmbusmeterswiki/

closed 2021-11-08 by weetmuts

Eddie979, 2024-09-25:
Hi, I have same issue but I think it is due to a different problem source.
My meter is a WMBus Metersit (gas meter) that broadcast @169MHz: log telegram is the following:

(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.

Do you know if such a meter brand could work?

Thank you in advance