wmbusmeters/wmbusmeters #379: Unknown tpl-ci-field 81

Santa77, 2021-11-06:
Telegrams:
```
3444EE4D813929271608811D7A4F000000046D1712A62B036E000000641E426CE1F1436E00000002FF2C00000259D6D0D4090265FC0902FD66A00044C4
3444EE4D813929271608811D7A50000000046D1812A62B036E000000DF58426CE1F1436E00000002FF2C00000259D6D0D4090265FC0902FD66A00044C4
3444EE4D813929271608811D7A51000000046D1912A62B036E0000005170426CE1F1436E00000002FF2C00000259D6D0D4090265FC0902FD66A00044C4
```

Acording to document:
https://ilnas.services-publics.lu/ecnor/downloadPreview.action?documentReference=35563
CI 81h and 82h are defined and reserved for future using in relay and netrowking.

Is there possible to just ignore (skip) that CI Fields in telegrams?

weetmuts, 2021-11-06:
Hmm, the 81 ci field is followed by a single 1d. If I just remove those two bytes the telegram parses ok, sort of.
3444EE4D813929271608>>>811D<<<<7A51000000|046D1912A62B036E0000005170426CE1F1436E00000002FF2C00000259D6D0D4090265FC0902FD66A00044C4

(sontex868) 0f: 04 dif (32 Bit Integer/Binary Instantaneous value)
(sontex868) 10: 6D vif (Date and time type)
(sontex868) 11: * 1912A62B device datetime (2021-11-06 18:25)
(sontex868) 15: 03 dif (24 Bit Integer/Binary Instantaneous value)
(sontex868) 16: 6E vif (Units for H.C.A.)
(sontex868) 17: * 000000 current consumption (0.000000 hca)
After that it gets weird though.....

However going back to the 0x81....according to the wmbus standard it seems like 81 is network layer ci. The wmbusmeters source actually has an empty function parseNWL in anticipation of telegrams with network layer
information. However according to the standard there should be significantly more bytes than 1, after the 81.
So, I have pushed a fix that removes 81 1d. However, there is more to be understood/parsed here.

Do you know if your telegrams have been routed through a wmbus router that extends the range to the meters?
Do you know that routers id number?

Santa77, 2021-11-07:
No router, just ordinary Sontex 868 HCA on my desk... never used, damaged coper plate of sensor on back...

weetmuts, 2021-11-09:
Duplicate of https://github.com/weetmuts/wmbusmeters/issues/382

closed 2021-11-09 by weetmuts