wmbusmeters/wmbusmeters #380: Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path!

charugaa, 2021-11-06:
Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path!

weetmuts, 2021-11-06:
Please read the README.  Compile and install rtl_wmbus https://github.com/xaelsouth/rtl-wmbus

closed 2021-11-12 by weetmuts

chrigabr, 2023-03-15:
Hello, I have same Problem too. I don't undestand. 

"Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path!" What can I do? 

And when I do sudo wmbusmeters --debug --t1 rtlwmbus, I have this:

_(shell) return code 1
(shell) 1538 exited
(serial) EVENT thread interrupted
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>>/bin/sh: 1: rtl_wmbus: Permission denied
<<<
(shell) return code 127
(shell) 1540 exited
(main) rtl_wmbus NOT found in path
Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path!
(serial) EVENT thread interrupted_
 
Many thanks
Christian GABRIEL

franciozzy, 2023-11-04:
> (shell) output: >>>/bin/sh: 1: rtl_wmbus: Permission denied

This means you do not have permissions to execute rtl_wmbus. That could have happened if you didn't install it properly.
Try: `sudo chmod 755 $(which rtl_wmbus)`

juergen908, 2025-06-20:
> Hello, I have same Problem too. I don't undestand.
>
> "Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path!" What can I do?
>
> And when I do sudo wmbusmeters --debug [...]

Something makes me wondering:
Sometimes it is written "SDR" and sometimes "SRD".
Whats wrong with it ?

But i have still the Problem:
"Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path! "
on Raspi4b.
Can someone help me ? 
I have not much experience on Raspi...
Maybe someone in German ?

PR #385 (2021-11-12): Honor the env in child processes
fixes #380: execvpe replaces the original env with the given array. The
array was empty effectivly removing PATH from the subshell. Thus
rtl_wmbus is not found, even if located within the PATH, as stated in
the documentation.
commit 37abbcdf (2021-11-12): Honor the env in child processes

fixes #380: execvpe replaces the original env with the given array. The
array was empty effectivly removing PATH from the subshell. Thus
rtl_wmbus is not found, even if located within the PATH, as stated in
the documentation.