wmbusmeters/wmbusmeters #382: Unknown tpl-ci-field 0d

Santa77, 2021-11-07:
Another strange telegram:

```
734414867CC30B0003030D58A00EDF0700DC41343CEA390306FF0A0D44EADF07000DDF07000DDF07000DDF07000E44F9DF07000EDF07000EDF07000EDF07000E931EDF07000DDF07000DDF07000DDF0700035A4500000000374F0B20000000000000243412B31C07050C12280000000000002C002302062F19080F2F000000000000DCF5
```
similar like 81, if i remove 0D58, then telegram is parsed correctly....

```
734414867CC30B0003030D58A00EDF0700DC41343CEA390306FF0A0D44EADF07000DDF07000DDF07000DDF07000E44F9DF07000EDF07000EDF07000EDF07000E931EDF07000DDF07000DDF07000DDF0700035A4500000000374F0B20000000000000243412B31C07050C12280000000000002C002302062F19080F2F000000000000DCF5
```

```
(meter) newly created meter (merac 000bc37c apator08) did not handle telegram!
merac   000bc37c        171.951333 m3   2021-11-07 13:25.33
```

weetmuts, 2021-11-07:
What dongle are you using to receive the telegrams? Could these be dll crc remants that should have been removed?

Santa77, 2021-11-07:
I am using this:
https://busware.de/tiki-index.php?page=CUL

with +5 dBi antena and latest culfw

weetmuts, 2021-11-07:
Please collect a --debug log when you receive a telegram.

Santa77, 2021-11-07:
```
wmbusmeters 6E440106461917020507511C7A420060857C6742015FD859733277AE7C8FD141A86A9FD861E5843084CEC0266ACBFE25035F38249596A96CDEA1B9436CB4A4D6495288DD7A780CCFB92575643D7CCDBD63D1A58288DB58D7E71D04B97D3B297D1EA16206BD1FA8D898687B5DE9A2E7D111BF6EF10DD52A88338421D06FA481
No meters configured. Printing id:s of all telegrams heard!
(wmbus) Not implemented tpl-ci 51
Received telegram from: 02171946
          manufacturer: (APA) Apator, Poland (0x601)
                  type: Water meter (0x07)
                   ver: 0x05
                driver: apator162
santa@DESKTOP-HRQ5FG0:~/wsl/wmbusmeters$ wmbusmeters --debug 6E440106461917020507511C7A420060857C6742015FD859733277AE7C8FD141A86A9FD861E5843084CEC0266ACBFE25035F38249596A96CDEA1B9436CB4A4D6495288DD7A780CCFB92575643D7CCDBD63D1A58288DB58D7E71D04B97D3B297D
1EA16206BD1FA8D898687B5DE9A2E7D111BF6EF10DD52A88338421D06FA481
(wmbusmeters) version: 1.5.0-3-g02e8404 with local changes
(config) using device: <6E440106461917020507511C7A420060857C6742015FD859733277AE7C8FD141A86A9FD861E5843084CEC0266ACBFE25035F38249596A96CDEA1B9436CB4A4D6495288DD7A780CCFB92575643D7CCDBD63D1A58288DB58D7E71D04B97D3B297D1EA16206BD1FA8D898687B5DE9A2E7D111BF6EF10DD52A88338421D06FA481>:any
(config) number of meters: 0
(lookup) with file/hex "6E440106461917020507511C7A420060857C6742015FD859733277AE7C8FD141A86A9FD861E5843084CEC0266ACBFE25035F38249596A96CDEA1B9436CB4A4D6495288DD7A780CCFB92575643D7CCDBD63D1A58288DB58D7E71D04B97D3B297D1EA16206BD1FA8D898687B5DE9A2E7D111BF6EF10DD52A88338421D06FA481"
(lookup) driver: simulation hex
(main) opening <6E440106461917020507511C7A420060857C6742015FD859733277AE7C8FD141A86A9FD861E5843084CEC0266ACBFE25035F38249596A96CDEA1B9436CB4A4D6495288DD7A780CCFB92575643D7CCDBD63D1A58288DB58D7E71D04B97D3B297D1EA16206BD1FA8D898687B5DE9A2E7D111BF6EF10DD52A88338421D06FA481>:any
Started config simulation listening on any
(simulation) in
(main) regular reset of simulation  will happen every 82800 seconds
(main) added  to files
(wmbus) no alarm (expected activity) for simulation
(lookup) with file/hex "6E440106461917020507511C7A420060857C6742015FD859733277AE7C8FD141A86A9FD861E5843084CEC0266ACBFE25035F38249596A96CDEA1B9436CB4A4D6495288DD7A780CCFB92575643D7CCDBD63D1A58288DB58D7E71D04B97D3B297D1EA16206BD1FA8D898687B5DE9A2E7D111BF6EF10DD52A88338421D06FA481"
(lookup) driver: simulation hex
(main) opening <6E440106461917020507511C7A420060857C6742015FD859733277AE7C8FD141A86A9FD861E5843084CEC0266ACBFE25035F38249596A96CDEA1B9436CB4A4D6495288DD7A780CCFB92575643D7CCDBD63D1A58288DB58D7E71D04B97D3B297D1EA16206BD1FA8D898687B5DE9A2E7D111BF6EF10DD52A88338421D06FA481>:any
Started config simulation listening on any
(simulation) in
(main) regular reset of simulation  will happen every 82800 seconds
(main) added  to files
(wmbus) no alarm (expected activity) for simulation
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(simulation) from file "6E440106461917020507511C7A420060857C6742015FD859733277AE7C8FD141A86A9FD861E5843084CEC0266ACBFE25035F38249596A96CDEA1B9436CB4A4D6495288DD7A780CCFB92575643D7CCDBD63D1A58288DB58D7E71D04B97D3B297D1EA16206BD1FA8D898687B5DE9A2E7D111BF6EF10DD52A88338421D06FA481"
(wmbus) parseDLL @0 127
(telegram) DLL L=6e C=44 (from meter SND_NR) M=0601 (APA) A=02171946 VER=05 TYPE=07 (Water meter) (driver apator162) DEV= RSSI=0
(wmbus) parseELL @10 117
(wmbus) parseNWL @10 117
(wmbus) parseAFL @10 117
(wmbus) parseTPL @10 117
(wmbus) Not implemented tpl-ci 51
Received telegram from: 02171946
          manufacturer: (APA) Apator, Poland (0x601)
                  type: Water meter (0x07)
                   ver: 0x05
                driver: apator162
(wmbus) 00: 6e length (110 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 0106 dll-mfct (APA)
(wmbus) 04: 46191702 dll-id (02171946)
(wmbus) 08: 05 dll-version
(wmbus) 09: 07 dll-type (Water meter)
(wmbus) 0a: 51 tpl-ci-field (Command to device (no tplh))
telegram=||6E440106461917020507511C7A420060857C6742015FD859733277AE7C8FD141A86A9FD861E5843084CEC0266ACBFE25035F38249596A96CDEA1B9436CB4A4D6495288DD7A780CCFB92575643D7CCDBD63D1A58288DB58D7E71D04B97D3B297D1EA16206BD1FA8D898687B5DE9A2E7D111BF6EF10DD52A88338421D06FA481|+0
(serial) stopping manager
(simulation) from file "6E440106461917020507511C7A420060857C6742015FD859733277AE7C8FD141A86A9FD861E5843084CEC0266ACBFE25035F38249596A96CDEA1B9436CB4A4D6495288DD7A780CCFB92575643D7CCDBD63D1A58288DB58D7E71D04B97D3B297D1EA16206BD1FA8D898687B5DE9A2E7D111BF6EF10DD52A88338421D06FA481"
(wmbus) skipping already handled telegram.
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) event loop stopped!
(serial) TIMER thread interrupted
(wmbus) deleted simulation
(wmbus) deleted simulation
```

```
santa@DESKTOP-HRQ5FG0:~/wsl/mqtt-sender$ cat simulation_3.txt
telegram=||6E440106461917020507511C7A420060857C6742015FD859733277AE7C8FD141A86A9FD861E5843084CEC0266ACBFE25035F38249596A96CDEA1B9436CB4A4D6495288DD7A780CCFB92575643D7CCDBD63D1A58288DB58D7E71D04B97D3B297D1EA16206BD1FA8D898687B5DE9A2E7D111BF6EF10DD52A88338421D06FA481|

santa@DESKTOP-HRQ5FG0:~/wsl/mqtt-sender$
santa@DESKTOP-HRQ5FG0:~/wsl/mqtt-sender$ wmbusmeters --debug simulation_3.txt MyMeter apator162 02171946 00000000000000000000000000000000
[... 25 lines cut ...]
(wmbus) parseDLL @0 127
(wmbus) parseELL @10 117
(wmbus) parseNWL @10 117
(wmbus) parseAFL @10 117
(wmbus) parseTPL @10 117
(wmbus) Not implemented tpl-ci 51
[... cut ...]
(telegram) DLL L=6e C=44 (from meter SND_NR) M=0601 (APA) A=02171946 VER=05 TYPE=07 (Water meter) (driver apator162) DEV= RSSI=0
(wmbus) parseELL @10 117
(wmbus) parseNWL @10 117
(wmbus) parseAFL @10 117
(wmbus) parseTPL @10 117
(wmbus) Not implemented tpl-ci 51
(meter) newly created meter (MyMeter 02171946 apator162) did not handle telegram!
(wmbus) telegram from 02171946 ignored by all configured meters!
(serial) stopping manager
(serial) waiting for stop
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!
(wmbus) deleted simulation
```

weetmuts, 2021-11-08:
Thanks, yes, your telegrams that you pass on the command line contain the dll crcs.
When you run the cul driver, it automatically removes the dll crcs (look for trimCRCsFrame... A and B) 
in the cul driver. But so far, the command line hex and simulation file does not remove any dll crcs.

However, it is something I intended to add for some time. So I just pushed a fix. 
When passing a hex string on the command line or inside a simulation file 
it will automatically remove any valid dll crcs it finds.

However when passing data to stdin or stdin:hex or raw.bin or raw.bin:hex then
it still requires the dll crcs to be removed already. Perhaps I will add a driver option:
stdin:hex(dllcrcs=expected) to force check and removal of dll crcs. Any failed dll crc
will reject the telegram.

The idea is that simulation_ and command line hex are not supposed to be used in production.
So the slight security problem that valid dll crcs are removed and bad (or no) dll crcs passed along,
is not a real problem....

Please test the command line.
Also please test when using the cul device directly, the dll crcs should already be removed correctly.

Santa77, 2021-11-09:
Ok, i've made in my preprocessor changes to check and remove that crc and now everything works for me. thank you for your help....

weetmuts, 2021-11-28:
Your welcome. I have pushed an improvement to wmbusmeters so that it will auto-detect and remove dll crcs for telegrams that are supplied in a simulation_ file or on the command line.

For all other use cases the dll-crcs are either checked or assumed to not be there.

closed 2021-11-28 by weetmuts