wmbusmeters/wmbusmeters #383: encrypted payload can start with "2F2F"

zivillian, 2021-11-08:
My meter is encrypted and uses an unknown key, nevertheless a few telegrams were reported. It sounds unlikely, but all of the reported telegrams have an encrypted payload starting with "2f2f", which in [my older version](https://github.com/weetmuts/wmbusmeters/commit/6c34beabaef08d468675361ae1f025467e327c84) skipped the decryption. As far as I can tell the current master would report a warning and skip those telegrams.

```
(serial) received binary "A5C2033E44A5113962487276077AC60030252F2FE643D7D5A1AB3BF59B94EE339C243C3079CEB397873FE1B823B9AEC3A574008AA9EA730880212E4808265542FEAC55EF45"
(im871a) checkIM871AFrame "A5C2033E44A5113962487276077AC60030252F2FE643D7D5A1AB3BF59B94EE339C243C3079CEB397873FE1B823B9AEC3A574008AA9EA730880212E4808265542FEAC55EF45"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) rssi 85 (-80 dBm)
(im871a) got crc16 45ef expected 45ef
(im871a) received full frame
(wmbus) parseDLL @0 63
(wmbus) parseELL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(meter) : for me? 72486239 in 72486239
[2021-10-24_02:51:17] (meter) : meter detection did not match the selected driver hydrus! correct driver is: unknown!
(meter) Not printing this warning agin for id: 72486239 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x76
[2021-10-24_02:51:17] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2021-10-24_02:51:17] (meter) to add support for this unknown mfct,media,version combination
(meter) : yes for me
(meter)  hydrus handling telegram from 72486239
(meter)  72486239 "3E44A5113962487276077AC60030252F2FE643D7D5A1AB3BF59B94EE339C243C3079CEB397873FE1B823B9AEC3A574008AA9EA730880212E4808265542FEAC"
(wmbus) parseDLL @0 63
(telegram) DLL L=3e C=44 (from meter SND_NR) M=11a5 (DME) A=72486239 VER=76 TYPE=07 (Water meter) (driver unknown!) DEV=im871a[00102741] RSSI=-80
(wmbus) parseELL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(telegram) TPL CI=7a ACC=c6 STS=00 CFG=2530 (synchronous AES_CBC_IV nb=3 cntn=0 ra=0 hc=0)
telegram=|3E44A5113962487276077AC60030252F2F|E643D7D5A1AB3BF59B94EE339C243C3079CEB397873FE1B823B9AEC3A574008AA9EA730880212E4808265542FEAC|+467595
(hydrus) 00: 3e length (62 bytes)
(hydrus) 01: 44 dll-c (from meter SND_NR)
(hydrus) 02: a511 dll-mfct (DME)
(hydrus) 04: 39624872 dll-id (72486239)
(hydrus) 08: 76 dll-version
(hydrus) 09: 07 dll-type (Water meter)
(hydrus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(hydrus) 0b: c6 tpl-acc-field
(hydrus) 0c: 00 tpl-sts-field (OK)
(hydrus) 0d: 3025 tpl-cfg 2530 (synchronous AES_CBC_IV nb=3 cntn=0 ra=0 hc=0 )
(hydrus) 0f: 2f2f decrypt check bytes
(hydrus) 11: E6 dif (48 Bit Integer/Binary Minimum value storagenr=1)
(hydrus) 12: 43 dife (subunit=1 tariff=0 storagenr=7)
(hydrus) 13: D7 vif (Mass 10⁴ kg/h)
(hydrus) 14: D5 vife (duration of limit exceed last lower  is 1)
(hydrus) 15: A1 vife (per minute)
(hydrus) 16: AB vife (incr per output pulse on input channel 1)
(hydrus) 17: 3B vife (forward flow)
(hydrus) 18: F59B94EE339C
(hydrus) 1e: 24 dif (32 Bit Integer/Binary Minimum value)
(hydrus) 1f: 3C vif (Volume flow 10⁻² m³/h)
(hydrus) 20: 3079CEB3
(hydrus) 24: 97 dif (64 Bit Integer/Binary Maximum value)
(hydrus) 25: 87 dife (subunit=0 tariff=0 storagenr=14)
(hydrus) 26: 3F dife (subunit=0 tariff=12 storagenr=494)
(hydrus) 27: E1 vif (Temperature difference 10⁻² K)
(hydrus) 28: B8 vife (multiplied by s/A)
(hydrus) 29: 23 vife (per day)
(hydrus) 2a: B9AEC3A574008AA9
(hydrus) 32: EA dif (4 digit BCD Minimum value storagenr=1)
(hydrus) 33: 73 dife (subunit=1 tariff=3 storagenr=7)
(hydrus) 34: 08 vif (Energy J)
(hydrus) 35: 8021
(hydrus) 37: 2E dif (12 digit BCD Minimum value)
(hydrus) 38: 48 vif (Volume flow ext. mm³/s)
(hydrus) 39: 08265542FEAC
(shell) exec "/bin/sh"
```

weetmuts, 2021-11-08:
Indeed, that was odd. The first 2f2f should not be exposed when the telegram is encrypted,
since they reveal themselves only when the proper key is used. So that you can verify that
the right key was used. Thanks for this info!

Some meters keep them, some remove them, when the meter does not use an AES key, ie the telegram
is not encrypted at all.

Do you need help with how wmbusmeters treat these telegrams?

zivillian, 2021-11-08:
Just to be clear - I'm very confident that this telegram is encrypted. It is the only telegram within over 20 days which could be interpreted by wmbusmeters.

I've already opened #355, but until I've got the proper key, nothing can be done.

weetmuts, 2021-12-12:
Yeah, lets wait until we have the proper key. But I will keep this in mind. 
Thank you for the info!

closed 2021-12-12 by weetmuts