wmbusmeters/wmbusmeters #384: Driver for (ZRI) ZENNER International, Germany (0x6a49)

chroeseler, 2021-11-09:
- Using an im871a in c1 mode

 ```
Received telegram from: xxxxxxxx

          manufacturer: (ZRI) ZENNER International, Germany (0x6a49)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x14
      Concerning meter: xxxxxxxx
          manufacturer: (ZRI) ZENNER International, Germany (0x6a49)
                  type: Water meter (0x07)
                   ver: 0x00
                device: im871a[xxxxxxxx]
                  rssi: -43 dBm
                driver: unknown!
```

- With multical21 rudimentary volumes can be read. 

`MyTapWater      xxxxxxxx        0.575 m3        0 m3    0 m3/h  127 °C  127 °C  OK      2021-11-08 19:04.58
`

- Further drivers are not tested as of now

weetmuts, 2021-11-09:
You need to provide a full telegram. Use --logtelegram and post them here.

chroeseler, 2021-11-09:
```
telegram=|6644496A2463025514377234661815496A0007760050052F2F|0C1355140000026CA92B82046CA12B8C0413710000008D0493132CFBFE000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF02FD1700002F2F|+180
```

From

```
telegram=|6644496A2463025514377234661815496A0007760050052F2F|0C1355140000026CA92B82046CA12B8C0413710000008D0493132CFBFE000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF02FD1700002F2F|+180
MyTapWater      15186634        1.455 m3        0 m3    0 m3/h  127 °C  127 °C  OK      2021-11-09 17:04.53
telegram=|6644496A2463025514377234661815496A0007770050052F2F|0C1355140000026CA92B82046CA12B8C0413710000008D0493132CFBFE000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF02FD1700002F2F|+360
MyTapWater      15186634        1.455 m3        0 m3    0 m3/h  127 °C  127 °C  OK      2021-11-09 17:07.53```

trsqr, 2021-11-29:
Just out of interest, which water meter are you using?

trsqr, 2021-11-29:
I too have a Zenner meter. It's a Minomess unit that works in C1 mode. This device to be exact: https://zenner.com/products/wwz_minomess_wm-bus-2/

The telegram looks like this for me:

`telegram=|6644496A1064035514377251345015496A0007740050052F2F|0C1359000000026CBD2B82046CA12B8C0413FFFFFFFF8D0493132CFBFEFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF02FD1700002F2F|+333
`

chroeseler, 2021-11-29:
It's a 'Minomess wM-BUS C1 Mode' too.
Like I wrote, i do get readings like this: 
`MyTapWater xxxxxxxx 0.575 m3 0 m3 0 m3/h 127 °C 127 °C OK 2021-11-08 19:04.58` with the multical21 driver. 
The volumes are off by some ~0.024 and as there is no temperature I guess it just shows a default value. 
Definitely usable already.

trsqr, 2021-11-29:
Yes, the reading is off by 2 liters for me as well, but I suppose this is due to the wmbus module and the actual physical meter numbers not being in sync. I read in the service manual that with the proper installation device you could tap into the optical interface and sync them when commissioning the meter, but it's not really a big deal. 

According to the specs it should tell last months consumption figures and some other mildly interesting stuff only.

[image]

chroeseler, 2021-11-29:
Same here, we just use it for basic testing. 
Maybe another driver would show the rest, but I never checked.

trsqr, 2021-11-30:
I actually got the description of the message format and am writing a driver for this now. Should be ready for testing soon. :) It would be great to get your full telegram on the 2nd of December @chroeseler .

trsqr, 2021-11-30:
I've got a driver ready for testing here: https://github.com/trsqr/wmbusmeters/tree/minomess

Not implemented yet: status bits and the readings for months n-2 until n-15. Maybe the past months is not that interesting and I don't know of a good way to report them anyhow. Status I'll try to implement.. Will try to simulate some failures at least to understand better.

chroeseler, 2021-12-03:
> I actually got the description of the message format and am writing a driver for this now. Should be ready for testing soon. :) It would be great to get your full telegram on the 2nd of December @chroeseler .

Hey, sry just saw your message. I don't have a setup at my hands atm. I'll come back to you as quickly as possible. 
You just need a full telegram with the new driver? 

> I've got a driver ready for testing here: https://github.com/trsqr/wmbusmeters/tree/minomess
>
> Not implemented yet: status bits and the readings for months n-2 until n-15. Maybe the past months is [...]

So Last month is viewable? That sound more than enough. :D 
Thanks for the effort.

chroeseler, 2021-12-20:
> I actually got the description of the message format and am writing a driver for this now. Should be ready for testing soon. :) It would be great to get your full telegram on the 2nd of December @chroeseler .

Sry that it took so long, but I'm finally back at this project. In case the telegram is still needed here.

```
Started auto im871a[00102942] on /dev/ttyUSB0 listening on c1
telegram=|6644496A2463025514377234661815496A0007B10050052F2F|0C1359140000026CB42C82046CA12C8C0413591400008D0493132CFBFE881300000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF02FD1700002F2F|+103
MyTapWater      15186634        1.459 m3        1.459 m3        OK      2021-12-20 17:10.09
```

weetmuts, 2021-12-20:
Thanks for the telegram! It decodes nicely with the latest wmbusmeters thanks to @trsqr and https://github.com/weetmuts/wmbusmeters/pull/401 we also have an existing telegram for the regression tests.

I think we can close this issue now.

closed 2021-12-20 by chroeseler