wmbusmeters/wmbusmeters #388: Cannot get AMB8465 to read anything... Probably some missing something, somwehere...

pharmswo, 2021-11-16:
I am running Debian 10. Patched to the latest level.

I need to start using radio based HeitenKostVerteiler (quondis).

So I bought the amber dongle and it is recognised...
[117116.135409] usb 1-2: new full-speed USB device number 27 using xhci_hcd
[117116.290467] usb 1-2: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[117116.290469] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[117116.290470] usb 1-2: Product: AMB8465 Wireless USB Stick
[117116.290471] usb 1-2: Manufacturer: AMBER
[117116.290472] usb 1-2: SerialNumber: 2A01C30C
[117116.294654] usb 1-2: Detected FT232RL
[117116.294819] usb 1-2: FTDI USB Serial Device converter now attached to ttyUSB0

I have downloaded the wmbusmeters and built it.
My groups are:
... dialout cdrom floppy sudo audio dip video plugdev netdev bluetooth lpadmin scanner

I am running wmsbusmeters from the command line:
./wmbusmeters --verbose --debug auto:c1 /dev/ttyUSB0

I was just hoping to pull some random info out of the air to see it work. The trace is as follows:
(wmbusmeters) version: _
(config) using device: /dev/ttyUSB0
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0"
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input(ICRNL|IXON) output(ONLCR|OPOST) control(CLOCAL|CREAD|CSIZE|HUPCL) local(ECHO|ECHOCTL|ECHOE|ECHOK|ECHOKE|ICANON|IEXTEN|ISIG) special_chars(3,28,127,21,4,1,17,19,26,18,15,23,22) modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(serial) received binary ""
**(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary ""
(amb8465) reading response... 2
(serial) received binary ""
(amb8465) reading response... 3
(serial) received binary ""
(amb8465) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect amb8465)**
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a/im170a) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect rc1180)
(serial /dev/ttyUSB0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial /dev/ttyUSB0) sent "30"
(serial) EVENT thread interrupted
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0 (detect rc1180)
(rc1180) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect cul)
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serialtty) closed /dev/ttyUSB0 (detect cul)
(cul) are you there? no
(verbose) ignoring device /dev/ttyUSB0
(serial) EVENT thread interrupted
**No wmbus device detected, waiting for a device to be plugged in.**
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(shell) return code 1
(shell) 25370 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(serial) EVENT thread interrupted
(shell) output: >>>25365
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 25371 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted

The USB port is successfully monitored and If I pull it and reconnect it again I get the same set of messages.

if I try
./wmbusmeters --verbose --debug /dev/ttyUSB0:amb8465:c1,t1

the result is the same
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary ""
(amb8465) reading response... 2
(serial) received binary ""
(amb8465) reading response... 3
(serial) received binary ""
(amb8465) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect amb8465)

I assume that I have missed something that needs to be installed. Any help would be much appreciated.

weetmuts, 2021-12-09:
Hmm...try this simple command
./wmbusmeters auto:t1
does it say "No wmbus device detected, waiting for a device to be plugged in."?

ls -l /dev/ttyUSB0
whoami
groups

have you logged out and in again after you ran "sudo make` install"?

weetmuts, 2021-12-09:
Meh, you are right. It misbehaves sometimes when I start wmbusmeters and the amb8465 is already plugged in....
weird.... but if I unplug and replug while wmbusmeters is running, then it is detected....

weetmuts, 2021-12-09:
When I experiment with the dongle and look in syslog I see

[40064.237356] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[40064.237379] ftdi_sio 1-3:1.0: device disconnected
[40080.713618] usb 1-3: new full-speed USB device number 38 using xhci_hcd
[40080.841674] usb 1-3: Device not responding to setup address.
[40081.049633] usb 1-3: Device not responding to setup address.
[40081.257639] usb 1-3: device not accepting address 38, error -71

and I get this:
[40398.728203] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[40398.728280] ftdi_sio 1-4:1.0: device disconnected
[40407.447331] usb usb1-port3: Cannot enable. Maybe the USB cable is bad?

This happens on the right side of my laptop where the USB port feels a bit flimsy.
Can you verify your cable? Is it consistent?

docbobo, 2021-12-09:
okay, I got a stick directly connected to my laptop:

```
[1103111.326920] usb 1-1: new full-speed USB device number 42 using xhci_hcd
[1103111.483147] usb 1-1: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[1103111.483153] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1103111.483154] usb 1-1: Product: AMB8465 AMBER USB Stick
[1103111.483156] usb 1-1: Manufacturer: FTDI
[1103111.483157] usb 1-1: SerialNumber: 2A01DEB7
[1103111.488463] ftdi_sio 1-1:1.0: FTDI USB Serial Device converter detected
[1103111.488505] usb 1-1: Detected FT232RL
[1103111.489116] usb 1-1: FTDI USB Serial Device converter now attached to ttyUSB0
```

I got udev rules setup as described in /docker, which means that my device shows up like this:

```
/wmbusmeters # ls -lisa /dev/amb8465 
   1508      0 lrwxrwxrwx    1 root     root             7 Dec  9 16:48 /dev/amb8465 -> ttyUSB0
```

and here some output

```
/wmbusmeters # ./wmbusmeters --verbose --debug /dev/amb8465:amb8465:c1,t1 
(wmbusmeters) version: 1.5.0-28-gb9bac60
(config) using device: /dev/amb8465:amb8465:c1,t1 
(config) number of meters: 0
(lookup) with file/hex "/dev/amb8465"
(serial) check if /dev/amb8465 can be accessed
(serial) tty /dev/amb8465 can be accessed
(serial) before config: /dev/amb8465 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(240,37,178,97,88,217,87,47) modem(DTR|RTS|CTS)
(serial) after config:  /dev/amb8465 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(240,37,178,97,16) modem(DTR|RTS|CTS)
(serialtty) opened /dev/amb8465 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/amb8465) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary ""
(amb8465) reading response... 2
(serial) received binary ""
(amb8465) reading response... 3
(serial) received binary ""
(amb8465) are you there? no.
(serialtty) closed /dev/amb8465 (detect amb8465)
```

weetmuts, 2022-08-28:
Any news here? Have you tried without the symbolic link and double checked that the dongle is on the ttyUSB? that you expect?

docbobo, 2022-09-03:
Just tried on a completely different system - not getting this to work.

weetmuts, 2022-09-03:
Can you build the latest wmbusmeters from github? 

Check "ls -l /dev/ttyUSB0" what does it say?
Check "groups" what does it say?

Then run this command with the latest wmbusmeters.
./wmbusmeters --debug /dev/ttyUSB0:amb8465:c1,t1

weetmuts, 2023-02-03:
@docbobo @pharmswo I might be that the latest wmbus dongle from Würth that has replaced the amb8465 actually
is called TarvosI and has a quite different serial protocol.... 

Can you double check when you bought your amb8465 and if it is indeed a 
TARVOS-I USB RADIO STICK 2605056081001 that replaces the amb8465 product.

closed 2023-05-07 by github-actions[bot]