wmbusmeters/wmbusmeters #389: Support for Brunata Metrona Telemetric Smart

neuma, 2021-11-24:
Hi,

while most heat cost allocators read perfectly well out of the box - great job btw - this one seems to be a new model. The `normal` log with `logtelegrams=true` provides me with the following redacted log messages

```
[DATE] (meter) allheater: meter detection could not find driver for id: 00012345 mfct: (MWU) METRONA Union, Germany (0x36f5) type: Heat Cost Allocator (0x08) ver: 0x02
[DATE] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[DATE] (meter) to add support for this unknown mfct,media,version combination
(wmbusmeters) started meter X (allheater 00012345 auto)
[DATE] (wmbus) WARNING! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 00012345 mfct: (MWU) METRONA Union, Germany (0x36f5) type: Heat Cost Allocator (0x08) ver: 0x02
```

While I receive telegrams for other meters, this one does not seem to print any telegram. I do understand that the driver could not be found. However, the decrypt warning keeps me puzzled whether this is triggered by the missing driver or indeed an encrypted message - did a brief code review but I am not confident enough in C++ to draw any conclusion.

What else can I check on my side?

---

Further material concerning the driver

The [data sheet](https://www.brunata-metrona.de/site_data/user_upload/downloads/allgemein/alle/BRUNATA_Heizkostenverteiler_Telmetric_smart_Technisches_Datenblatt.pdf) on [their website](https://www.brunata-metrona.de/geraete-systeme/erfassungsgeraete/heizkostenverteiler.html) states OMS-T1 as protocoll.

I tried to read the meter by explicitely stating the driver in the config, basically with all heat cost related drivers listed by  `--listmeters`. No success.

weetmuts, 2021-11-29:
Can you post a telegram here? Use --logtelegram for example.

But from the message it seems that you need a decryption key. Do you have one?

neuma, 2021-11-30:
> Can you post a telegram here? Use --logtelegram for example.

`logtelegrams=true` does not produce any telegrams for the Metronas in the form of `telegram=|1234*`

However, enabling debug mode gives some more info. Please note, I tried to filter out one of the meters. Hope the debug info is complete.

```
(serial) received ascii "T1;1;1;DATE;68;68;01234567;0x3e44f5361932070002087acf00202534ac1149148d900636b7e74534183b6008602416cf39f5cad0cd99d417e8d0d60f800c00010900d585050020671b050f<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;DATE;68;68;01234567;0x3e44f5361932070002087acf00202534ac1149148d900636b7e74534183b6008602416cf39f5cad0cd99d417e8d0d60f800c00010900d585050020671b050f<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 63
(wmbus) parseELL @10 53
(wmbus) parseNWL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(meter) metrona: for me? 01234567 in 88888888
(meter) metrona: not for me: not my id
(wmbus) parseDLL @0 63
(wmbus) parseELL @10 53
(wmbus) parseNWL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(meter) metrona: for me? 01234567 in 99999999
(meter) metrona: not for me: not my id
(meter) no meter handled 01234567 checking 1 templates.
(wmbus) parseDLL @0 63
(wmbus) parseELL @10 53
(wmbus) parseNWL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(meter) metrona: for me? 01234567 in 01234*
(meter) metrona: yes for me
[DATE] (meter) metrona: meter detection could not find driver for id: 01234567 mfct: (MWU) METRONA Union, Germany (0x36f5) type: Heat Cost Allocator (0x08) ver: 0x02
[DATE] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[DATE] (meter) to add support for this unknown mfct,media,version combination
(meter) created "metrona" "unknown" "01234567" encrypted
(meter) used meter template metrona 01234* auto to match 01234567
(wmbusmeters) started meter 3 (metrona 01234567 auto)
(wmbus) parseDLL @0 63
(wmbus) parseELL @10 53
(wmbus) parseNWL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(meter) metrona: for me? 01234567 in 01234567
(meter) metrona: yes for me
(meter) metrona auto handling telegram from 01234567
(meter) metrona 01234567 "3E44F5361932070002087ACF00202534AC1149148D900636B7E74534183B6008602416CF39F5CAD0CD99D417E8D0D60F800C00010900D585050020671B050F"
(wmbus) parseDLL @0 63
(telegram) DLL L=3e C=44 (from meter SND_NR) M=36f5 (MWU) A=01234567 VER=02 TYPE=08 (Heat Cost Allocator) (driver unknown!) DEV=rtlwmbus[77771111153705700] RSSI=68
(wmbus) parseELL @10 53
(wmbus) parseNWL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
[DATE] (wmbus) WARNING! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 01234567 mfct: (MWU) METRONA Union, Germany (0x36f5) type: Heat Cost Allocator (0x08) ver: 0x02
(telegram) TPL CI=7a ACC=cf STS=00 CFG=2520 (synchronous AES_CBC_IV nb=2 cntn=0 ra=0 hc=0)
[DATE] (meter) newly created meter (metrona 01234567 auto) did not handle telegram!
(wmbus) telegram from 01234567 ignored by all configured meters!
(serial) received ascii "T1;1;1;DATE;73;73;01234567;0x3e44f5361932070002087acf00202534ac1149148d900636b7e74534183b6008602416cf39f5cad0cd99d417e8d0d60f800c00010900d585050020671b050f<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;DATE;73;73;01234567;0x3e44f5361932070002087acf00202534ac1149148d900636b7e74534183b6008602416cf39f5cad0cd99d417e8d0d60f800c00010900d585050020671b050f<0A>"
(rtlwmbus) received full frame
```

> But from the message it seems that you need a decryption key. Do you have one?

No, I don't have a key yet. Before asking for the keys of the newly installed Metronas I would like to know whether this is a driver issue or not.

weetmuts, 2021-12-02:
The telegram looks ok and since they say it is OMS-T1 that should mean that they use the standardized vif:s for HCA measurements, thus it is probably easy to add support for it. :-)

neuma, 2021-12-19:
sorry for not getting back on this earlier.

Is there anything I can help you with? Do you need more telegrams?

nillin-p, 2022-02-19:
I'm working on this, too. My meters were read by fhem and a CUL. Behind the wmbus header it appers to be an encrypted block. I would try to use wmbusmeters too to examine this. Or do you mean that it is only sensible to get the key?

neuma, 2022-03-10:
@milkpirate To my understanding, if you bought the meter, the key should be shipped together with the meter. If the meter is rented, you could ask the owner, e.g. your landlord, or try to get in contact with whatever service provider the meter-handling had been subcontracted to. Good luck with that one since this involves a lot of patience and communication skills.

Regarding the key... Mine do not have any qr code printed on them. So, my guess would be that this includes only the meter id and not the key itself - more of a convenience thing for handling the meters.

@nillin-p My current assumption is that the warning in the original post is indeed triggered by the missing decryption key. Would be great if somebody with a valid key could confirm this assumption by reading a meter without the proper key.

maerf, 2022-03-10:
I scanned to qr code on my reader (content below). Can anyone figure out
what these segments mean? Key?

4MWU0100334088
MU10;2106;131101;11313;13318;;

kongo09, 2022-03-30:
I scanned my code as well, it says:

76515014;H:167040;K:XXXXX;M:11712;2017;MPMK7FKME17MS4;LZ

The first number seems to be the meter ID which is also printed on the meter together with a separate barcode.

2017 seems to be the year of production

The rest, I don't know.

kongo09, 2022-03-30:
The second meter next to the one above reads:

77514790;H:167050;K:XXXXX;M:11721;2017;MPMW7FKME17MS4;LZ

The first again is the ID. 2017 the year of production.

The long text starts with MPMW - W could be warm as this is warm water. In the example above it was MPMK - K could be for cold. 

As that is the only difference between the two longer texts, it is rather unlikely that this represents a key.

kongo09, 2022-03-30:
Here comes a third meter, which is hot water again:

77514784;H:167050;K:XXXXX;M:11721;2017;MPMW7FKME17MS4;LZ

Essentially the same data as above apart from the ID.

milkpirate, 2022-04-10:
@neuma Hm... I see. How long is your key? Its AES right? Do you think its feasable to bruteforce it? The offical path to the key seems to take as long as bruteforcing to me 😄

kongo09, 2022-04-10:
Not sure if there is an official path at all. I've talked to the administrator managing our building. He's been in touch with Brunata before trying to get keys for meters when he changed the measurement provider at another building. The idea was to keep the old meters and allow the new company to read them. But Brunata refused and they had to physically exchange the meters to new ones. What a pain.

That seems to be a good reason to not do business with Brunata at all.

milkpirate, 2022-04-10:
Oh boy... I was already thinking "why encrypt the data"? Sure you might get info about when ppl. are home and use their heater. But yes a perlonging stonger customer "relation" might be the more resonable. They exchanged our meters a while ago. The "old" ones werent that old and wireless. The new ones are Metronas... I guess I have to discard that endevor.

Jabb0, 2022-12-24:
Did anybody have success in getting the keys for their Brunata meters?  

Having the packages encrypted is crucial IMO, otherwise the potential for misuse is high.

closed 2023-04-06 by github-actions[bot]

Krolli5, 2023-05-02:
Actual there are some used Metrona HHT offered at German ebay. Search for "Set Handterminal Ladegerät ULK - Ablesung Heizkostenverteiler Wasserzähler". Cost 35€ + shipping.
Suitable for telmetric Star - 868,3 MHz RF Version! At putting the optical reading head on allocator you can hear an reed contact clicking and transmission seems to start automatically. Menue items allow reading or programming - i did'nt test the last. Reading works!
Suggestion: if optical inteface uses similar or equal encryption, there might be a little chance for reverse engineering that unit. Or it might be set from monthly to hourly RF transmission for better investigation?

Krolli5, 2023-05-02:
minor update: R/W head has an integrated microswitch, so it's unclear who starts the protocol transmission.
II. One can choose a couple of programming modes, but to authorize programming an barcode strip is necessary. Therefore HHT comes with an reader on its top

Krolli5, 2023-05-02:
[image]

Krolli5, 2024-01-15:
[image]
[image]

Krolli5, 2024-01-15:
The main disadvantage is, that the meters will transmitt only one message per month. That corresponds with my handheld tester records and the information from the rare Metrona/Brunata documents. Imprint from Crystal = 27,13MHz gives 868,3 MHz Tx frequency, which is also confirmed by documents. Baudrate should be 2400Bd. Until now i don't have captured an dataset and my plan is connecting ASK Data and EN signals to Uno and record the data in binary direct to SD card. I hope, meter will transmit further if the intrusion contact is closed again. 
Btw the serial number differs slightly from display reading. The one, given by barcode is much longer

kongo09, 2024-01-15:
Oh, that's interesting and might explain why the devices seem invisible: if they only send data once a month, I would need to keep a listening service running for that long for any success.

Jabb0, 2024-01-15:
@Krolli5 thanks for the information!

Do you have an idea about the type of modulation used?

The monthly thing makes sense. I need to check where I got the signals from. Will get a SDR with multiple antennas to triangulate the source.

Krolli5, 2024-01-16:
According to the well documented chip information of TH72032* they are using ASK modulation. Simpliest form of AM. Carrier ON/OFF for logical H and L (or two level as eg at DCF77). I'm not sure, but this will result in a very narrow BW and not to waste much spectral energy in sidebands. So overall the transmitting range is increased. I'm in doubt, if mostly used RF frontends for SDR could be set for such small BW. If they are based on DVB-T(2) chipsets, that channel BW used is normally 5 to 8MHz! And the 866 ISM Band is full of competitors :-)
If one is familiar with, i think this is more an preferred application case for CC1100 or similar. 

* https://www.mouser.com/datasheet/2/734/TH72032-Datasheet-Melexis-953358.pdf (diagram on p. 7)

Krolli5, 2024-02-11:
As expected, i received some datasets direct from µP. Transmission starts in late hours of last day in month. And continues during the morning hours the next.
In total i count 38 (!) repetitons. Duration for receiving the 28 bytes always appox. 4 seconds. Absolutely no changes in data, so an rolling code (eg by date or time as key) can be excluded.

2024/1/31:   22:35:24; 23:01:23; 23:07:23; 23:21:24; 23:45:23; 
2024/2/01:   00:05:23; 00:11:23; 00:23:24; 00:49:24; 01:09:24; 01:11:24; 01:41:24; 01:57:24
                     02:03:24; 02:17:24; 02:35:24; 02:59:24; 03:15:24; 03:33:24; 03:37:24; 03:55:24;
                     04:09:24; 04:29:24; 04:45:24; 05:05:24; 05:23:24; 05:33:24; 05:49:24; 06:11:24;
                     06:23:24; 06:45:24; 06:59:24; 07:05:24; 07:31:24; 07:35:24; 08:05:24; 08:11:24;
                     08:25:24; 

Captured data:
0x{
 00
 FFE0
 3E
 FF93
 12
 30
 36
 1
 52
 73
 7C
 3B
 1
 12
 76
 1E
 FFE5
 17
 FF8D
 1A
 12
 32
 4
 12
 FFFF
}

There is no warranty for receiving correct data, cause i used softwareserial @ 2400,8,N,1 capturing these

Krolli5, 2024-03-07:
[image]
It seems, i generated an shift or offset in first recording by one or two byte. Aligned to the second from february gives the following data with only minor differences:

mStirner, 2025-02-13:
Anything new here?
I just stumbled across this software while trying to read my meters.
But all i get are a bunch of CRC errors: https://github.com/wmbusmeters/wmbusmeters/issues/1491