wmbusmeters/wmbusmeters #390: Support Zenner Minomess wmbus C1 water meters

trsqr, 2021-11-26:
I've got a nanoCUL stick and a Zenner Minomess water meter that works in C1 mode. This device: https://zenner.com/products/wwz_minomess_wm-bus-2/

I understand it's not supported. I can probably help, but need some pointers. I do seem to get some data in from the meter, and I can spot Zenner's manufacturer id 0x6A49 in a couple of spots in the raw data, so I think it's correctly received. However, I do seem to get the following messages:

> (cul) warning: the hex string is not proper! Ignoring telegram!
> (cul) error in received message.

Any idea why this is happening?

`$ wmbusmeters --debug --verbose /dev/ttyUSB0:cul:c1
(wmbusmeters) version: 1.5.0-8-g67e0072
(config) using device: /dev/ttyUSB0:cul:c1
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0"
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary "000A"
(amb8465) reading response... 2
(serial) received binary ""
(amb8465) reading response... 3
(serial) received binary ""
(amb8465) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a/im170a) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect rc1180)
(serial /dev/ttyUSB0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial /dev/ttyUSB0) sent "30"
(serial) EVENT thread interrupted
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0 (detect rc1180)
(rc1180) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect cul)
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary "3F2028F8807820697320756E6B6E6F776E2920557365206F6E65206F662042206220432065204620472069204B206C204D20522054207420562057205820780D0A"
(cul) probe response "? (<F8><80>x is unknown) Use one of B b C e F G i K l M R T t V W X x<0D><0A>"
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary "5620312E3637206E616E6F43554C3836380D0A"
(cul) probe response "V 1.67 nanoCUL868<0D><0A>"
(serialtty) closed /dev/ttyUSB0 (detect cul)
(cul) are you there? yes
(main) opening /dev/ttyUSB0:cul:c1
Started config cul on /dev/ttyUSB0 listening on c1
(cul) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (cul)
(main) regular reset of cul  on /dev/ttyUSB0 will happen every 82800 seconds
(cul) set link mode c
(serial /dev/ttyUSB0) sent "6272630A0D"
(serial) EVENT thread interrupted
(serial) received binary "434D4F44450D0A"
(cul) checkCULFrame "CMODE<0D><0A>"
(cul) no leading 'b' so it is text and no frame
(cul) received "CMODE"(serial /dev/ttyUSB0) sent "5832310A0D"
(wmbus) no alarm (expected activity) for cul
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted

(serial) received binary "6236363434343936413130363430333535313433374335333337323531333435303135343936413030303733423030353030354345303630374334323942414333374538453531323039424145413046334236444543384634373438374442393639374433373630393630414641324535333831334630"
(cul) checkCULFrame "b6644496A106403551437C5337251345015496A00073B005005CE0607C429BAC37E8E51209BAEA0F3B6DEC8F47487DB9697D3760960AFA2E53813F0"
(cul) no eol found yet, partial frame
(serial) received binary "454439423535354145424337434439454541463941453530414135353535433846323536463730314433354632314437324441414645313631453833453931423939344141354145324337343231303737354233453434424635453846433439303931434638370D0A"
(cul) checkCULFrame "b6644496A106403551437C5337251345015496A00073B005005CE0607C429BAC37E8E51209BAEA0F3B6DEC8F47487DB9697D3760960AFA2E53813F0ED9B555AEBC7CD9EEAF9AE50AA5555C8F256F701D35F21D72DAAFE161E83E91B994AA5AE2C74210775B3E44BF5E8FC49091CF87<0D><0A>"
(cul) checkCULFrame RSSI_RAW=-121
(cul) checkCULFrame LQI=103
(cul) bad hex "6644496A106403551437C5337251345015496A00073B005005CE0607C429BAC37E8E51209BAEA0F3B6DEC8F47487DB9697D3760960AFA2E53813F0ED9B555AEBC7CD9EEAF9AE50AA5555C8F256F701D35F21D72DAAFE161E83E91B994AA5AE2C74210775B3E44BF5E8FC49091"
(cul) warning: the hex string is not proper! Ignoring telegram!
(cul) error in received message.

(serial) received binary "62363634343439364131303634"
(cul) checkCULFrame "b6644496A1064"
(cul) no eol found yet, partial frame
(serial) received binary "303335353134333743353333373235313334353031353439364130303037334330303530303533344245324643413637444433463643323743464236424441314136454633323942334246323630463036383641363331443431443531313345343742413246443545453239363432454343413246444142454630"
(cul) checkCULFrame "b6644496A106403551437C5337251345015496A00073C00500534BE2FCA67DD3F6C27CFB6BDA1A6EF329B3BF260F0686A631D41D5113E47BA2FD5EE29642ECCA2FDABEF0"
(cul) no eol found yet, partial frame
(serial) received binary "35314430413130453744373536453037443043394243373733384646314642453046423233363334453235463541453836424434413435363534353331444136323345423731343239423033413441303531413435310D0A"
(cul) checkCULFrame "b6644496A106403551437C5337251345015496A00073C00500534BE2FCA67DD3F6C27CFB6BDA1A6EF329B3BF260F0686A631D41D5113E47BA2FD5EE29642ECCA2FDABEF051D0A10E7D756E07D0C9BC7738FF1FBE0FB23634E25F5AE86BD4A45654531DA623EB71429B03A4A051A451<0D><0A>"
(cul) checkCULFrame RSSI_RAW=81
(cul) checkCULFrame LQI=82
(cul) bad hex "6644496A106403551437C5337251345015496A00073C00500534BE2FCA67DD3F6C27CFB6BDA1A6EF329B3BF260F0686A631D41D5113E47BA2FD5EE29642ECCA2FDABEF051D0A10E7D756E07D0C9BC7738FF1FBE0FB23634E25F5AE86BD4A45654531DA623EB71429B03A4A051"
(cul) warning: the hex string is not proper! Ignoring telegram!
(cul) error in received message.

^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(serial) EVENT thread interrupted
(serial) event loop stopped!
(serial) TIMER thread interrupted
(wmbus) disconnected /dev/ttyUSB0 cul
(serialtty) closed /dev/ttyUSB0 (cul)
(wmbus) deleted cul
(serial) closing 1 devices
`

trsqr, 2021-11-26:
So this is what was received:
"b6644496A106403551437C5337251345015496A00073C00500534BE2FCA67DD3F6C27CFB6BDA1A6EF329B3BF260F0686A631D41D5113E47BA2FD5EE29642ECCA2FDABEF051D0A10E7D756E07D0C9BC7738FF1FBE0FB23634E25F5AE86BD4A45654531DA623EB71429B03A4A051A451<0D><0A>"

So when removing the line break and the RSSI and LQI from the end, we're left with: "6644496A106403551437C5337251345015496A00073C00500534BE2FCA67DD3F6C27CFB6BDA1A6EF329B3BF260F0686A631D41D5113E47BA2FD5EE29642ECCA2FDABEF051D0A10E7D756E07D0C9BC7738FF1FBE0FB23634E25F5AE86BD4A45654531DA623EB71429B03A4A051"

However, this doesn't contain an even number of digits, so it cannot be converted to hex!

```
$ wmbusmeters 6644496A106403551437C5337251345015496A00073C00500534BE2FCA67DD3F6C27CFB6BDA1A6EF329B3BF260F0686A631D41D5113E47BA2FD5EE29642ECCA2FDABEF051D0A10E7D756E07D0C9BC7738FF1FBE0FB23634E25F5AE86BD4A45654531DA623EB71429B03A4A051
Hex string must have an even lenght of hexadecimal characters.
```

If I remove one more digit, I get some results:

```
$ wmbusmeters 6644496A106403551437C5337251345015496A00073C00500534BE2FCA67DD3F6C27CFB6BDA1A6EF329B3BF260F0686A631D41D5113E47BA2FD5EE29642ECCA2FDABEF051D0A10E7D756E07D0C9BC7738FF1FBE0FB23634E25F5AE86BD4A45654531DA623EB71429B03A4A05
No meters configured. Printing id:s of all telegrams heard!
(wmbus) Unknown tpl-ci-field c5
Received telegram from: 55036410
          manufacturer: (ZRI) ZENNER International, Germany (0x6a49)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x14
                driver: unknown!
```

weetmuts, 2021-11-28:
Do you know who has developed the CUL firmware? If I look at me old comment on the other thread, I said:

>Well, if the cul is expected to send the bytes as ascii hex, then why does it send an odd number of hex characters? >Either your cul is broken or we have misunderstood the protocol.
>I am not an expert on CUL either....but the serial protocol looks pretty straightforward. So if something as weird 
>as a hex character is missing, then I am suspecting the CUL device.

And I am agreeing with what I said! :-) We could make a silly workaround adding a  0 or droping the last char.
This will work for  non-encrypted telegrams, alas when the telegram is encrypted then it might not work, depends
on if it is an extra char or a dropped char.

Do you have a good description of the CUL protocol, do you have the name of the people who developed the firmware?

weetmuts, 2021-11-28:
If you buy a cheap rtlsdr dongle, then we can compare the telegrams received with rtlsdr/rtlwmbus and cul.

trsqr, 2021-11-29:
So if the full string received from nanoCUL is here:

`
b6644496A106403551437C5337251345015496A00073C00500534BE2FCA67DD3F6C27CFB6BDA1A6EF329B3BF260F0686A631D41D5113E47BA2FD5EE29642ECCA2FDABEF051D0A10E7D756E07D0C9BC7738FF1FBE0FB23634E25F5AE86BD4A45654531DA623EB71429B03A4A051A451<0D><0A>
`

This, to my understanding means that there's 0x66 (102) bytes of wMBus data. However, between the 66 and 451 there are 221 characters, which would mean 110.5 bytes. The data should be AES encrypted as the key was provided on a sticker. But I wonder why the numbers don't match - are some headers excluded from this number?

I actually see that there's someone reporting Zenner compatibility in issue https://github.com/weetmuts/wmbusmeters/issues/384

`telegram=|6644496A2463025514377234661815496A0007760050052F2F|0C1355140000026CA92B82046CA12B8C0413710000008D0493132CFBFE000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF02FD1700002F2F|+180`

But I'm not sure how to interpret that telegram and compare it to my results.

For the firmware for my nanoCUL I compiled r568 from https://sourceforge.net/projects/culfw/ and modified board.h to disable other alternative radio protocols and enabled mbus (I needed the newer firmware as standard 1.67 firmware does not have C1 support).

http://culfw.de/culfw.html is the home page of the culfw project. The command reference is here, but it seems a bit too generic.

trsqr, 2021-11-29:
Looking at clib/rf_mbus.c in the culfw package, it seems here the serial string is constructed:

    if (rxStatus == PACKET_OK) {

      DC( 'b' );
      if (RXinfo.mode == WMBUS_CMODE) {
        if (RXinfo.frametype == WMBUS_FRAMEB) {
          DC('Y'); // special marker for frame type B to make it distinguishable
                   // from frame type A
        }
      }

      for (uint8_t i=0; i < rxLength; i++) {
        DH2( MBpacket[i] );
  //      DC( ' ' );
      }

      if (tx_report & REP_RSSI) {
        DH2(lqi);
        DH2(rssi);
      }
      DNL();
    }
    RXinfo.state = 0;
    return;
  }

Apparently DC, DH2 and DNL are functions that build the string to send. Based on this I can't understand how we could end up with an string that contains uneven number of digits, presuming that DC print just a single character, DH2 always prints two characters and DNL prints the newline.

trsqr, 2021-11-29:
I modified the firmware to print an empty space between each hexbyte and funny enough, I cannot see any issue there:

```
(serial) received binary "623636203434203439203641203130203634203033203535203134203337204335"
(cul) checkCULFrame "b66 44 49 6A 10 64 03 55 14 37 C5"
(cul) no eol found yet, partial frame
(serial) received binary "203333203732203531203334203530203135203439203641203030203037203641203030203530203035203242204230203538204632203138204536203845204633204330203746203943203236204434204642204538203644203837203030203236203036204639203537203937204344203841204246"
(cul) checkCULFrame "b66 44 49 6A 10 64 03 55 14 37 C5 33 72 51 34 50 15 49 6A 00 07 6A 00 50 05 2B B0 58 F2 18 E6 8E F3 C0 7F 9C 26 D4 FB E8 6D 87 00 26 06 F9 57 97 CD 8A BF"
(cul) no eol found yet, partial frame
(serial) received binary "20333120354120443120333220434320323520424120463220424420413920423520433120334520323720334220333920323020373020313220313520"
(cul) checkCULFrame "b66 44 49 6A 10 64 03 55 14 37 C5 33 72 51 34 50 15 49 6A 00 07 6A 00 50 05 2B B0 58 F2 18 E6 8E F3 C0 7F 9C 26 D4 FB E8 6D 87 00 26 06 F9 57 97 CD 8A BF 31 5A D1 32 CC 25 BA F2 BD A9 B5 C1 3E 27 3B 39 20 70 12 15 "
(cul) no eol found yet, partial frame
(serial) received binary "43312037302031450D0A"
(cul) checkCULFrame "b66 44 49 6A 10 64 03 55 14 37 C5 33 72 51 34 50 15 49 6A 00 07 6A 00 50 05 2B B0 58 F2 18 E6 8E F3 C0 7F 9C 26 D4 FB E8 6D 87 00 26 06 F9 57 97 CD 8A BF 31 5A D1 32 CC 25 BA F2 BD A9 B5 C1 3E 27 3B 39 20 70 12 15 C1 70 1E<0D><0A>"
(cul) bad hex for LQI and RSSI "0 1E"
(cul) warning: the LQI and RSSI hex string is not properly formatted!
Segmentation fault

```

wmbusmeters seems to segfault with this input, but that's not so critical.

trsqr, 2021-11-29:
I got it! In the board.h of the culfw there's a line that defines the buffer size:

#define TTY_BUFSIZE             224

However, the length of the message from my Zenner water meter requires a buffer size longer than that. By increasing this to 256 and recompiling the firmware did it. I'm attaching the nanoCUL firmware here just in case someone else needs it as well.

[nanoCUL_mbus_c1t1_bufsize256.zip](https://github.com/weetmuts/wmbusmeters/files/7616177/nanoCUL_mbus_c1t1_bufsize256.zip)

trsqr, 2021-11-29:
Now the output looks like this:

```
(cul) checkCULFrame "b6644496A106403551437C5337251345015496A000774005005199E2E08F043703DFB456C959E3D512E337F87A4A625D690A98A0CEE3CF592E719F64C86D78121F6F2279643E98E68F283F6DAB44673F98CA1B70309ECF35793A04AE56FC1666B39B24C3E49D56F58BB0AA4A781071F73E65DC884918041<0D><0A>"
(cul) checkCULFrame RSSI_RAW=65
(cul) checkCULFrame LQI=64
(wmbus) trimming frame A "6644496A106403551437C5337251345015496A000774005005199E2E08F043703DFB456C959E3D512E337F87A4A625D690A98A0CEE3CF592E719F64C86D78121F6F2279643E98E68F283F6DAB44673F98CA1B70309ECF35793A04AE56FC1666B39B24C3E49D56F58BB0AA4A781071F73E65DC88491"
(wmbus) ff a dll crc 0-9 c533 ok
(wmbus) ff a dll crc mid 12-27 08f0 ok
(wmbus) ff a dll crc mid 30-45 25d6 ok
(wmbus) ff a dll crc mid 48-63 f6f2 ok
(wmbus) ff a dll crc mid 66-81 b703 ok
(wmbus) ff a dll crc mid 84-99 49d5 ok
(wmbus) ff a dll crc final 102-114 8491 ok
(wmbus) trimmed 14 dll crc bytes from frame a and ignored 0 suffix bytes.
(wmbus) trimmed frame A "6644496A1064035514377251345015496A000774005005199E2E43703DFB456C959E3D512E337F87A4A690A98A0CEE3CF592E719F64C86D78121279643E98E68F283F6DAB44673F98CA109ECF35793A04AE56FC1666B39B24C3E6F58BB0AA4A781071F73E65DC8"
(cul) received full T1 frame
(wmbus) parseDLL @0 103
(wmbus) parseELL @10 93
(wmbus) parseNWL @10 93
(wmbus) parseAFL @10 93
(wmbus) parseTPL @10 93
(meter) minomess_water: for me? 55036410,15503451 in 15503451
(meter) minomess_water: meter detection did not match the selected driver multical21! correct driver is: unknown!
(meter) Not printing this warning again for id: 55036410 mfct: (ZRI) ZENNER International, Germany (0x6a49) type: Radio converter (meter side) (0x37) ver: 0x14
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) minomess_water: yes for me
(meter) minomess_water multical21 handling telegram from 15503451
(meter) minomess_water 15503451 "6644496A1064035514377251345015496A000774005005199E2E43703DFB456C959E3D512E337F87A4A690A98A0CEE3CF592E719F64C86D78121279643E98E68F283F6DAB44673F98CA109ECF35793A04AE56FC1666B39B24C3E6F58BB0AA4A781071F73E65DC8"
(wmbus) parseDLL @0 103
(telegram) DLL L=66 C=44 (from meter SND_NR) M=6a49 (ZRI) A=55036410 VER=14 TYPE=37 (Radio converter (meter side)) (driver unknown!) DEV=cul RSSI=-42
(wmbus) parseELL @10 93
(wmbus) parseNWL @10 93
(wmbus) parseAFL @10 93
(wmbus) parseTPL @10 93
(TPL) AES CBC IV decrypting "199E2E43703DFB456C959E3D512E337F87A4A690A98A0CEE3CF592E719F64C86D78121279643E98E68F283F6DAB44673F98CA109ECF35793A04AE56FC1666B39B24C3E6F58BB0AA4A781071F73E65DC8"
(TPL) num encrypted blocks 5 (80 bytes and remaining unencrypted 0 bytes)
(TPL) IV 496A5134501500077474747474747474
(TPL) decrypted  "2F2F0C1359000000026CBD2B82046CA12B8C0413FFFFFFFF8D0493132CFBFEFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF02FD1700002F2F"
(telegram) TPL CI=72 ACC=74 STS=00 CFG=0550 (AES_CBC_IV nb=5 cntn=0 ra=0 hc=0) ID=51345015 MFT=496a VER=00 TYPE=07 (Water meter)
telegram=|6644496A1064035514377251345015496A0007740050052F2F|0C1359000000026CBD2B82046CA12B8C0413FFFFFFFF8D0493132CFBFEFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF02FD1700002F2F|+333
(dvparser) warning: cannot extract uint16 from non-existant key "02FF20"
(wmbus) warning: cannot find offset -1 to add more explanation " info codes (OK)"
(multical21) 00: 66 length (102 bytes)
(multical21) 01: 44 dll-c (from meter SND_NR)
(multical21) 02: 496a dll-mfct (ZRI)
(multical21) 04: 10640355 dll-id (55036410)
(multical21) 08: 14 dll-version
(multical21) 09: 37 dll-type (Radio converter (meter side))
(multical21) 0a: 72 tpl-ci-field (EN 13757-3 Application Layer (long tplh))
(multical21) 0b: 51345015 tpl-id (15503451)
(multical21) 0f: 496a tpl-mfct (ZRI)
(multical21) 11: 00 tpl-version
(multical21) 12: 07 tpl-type (Water meter)
(multical21) 13: 74 tpl-acc-field
(multical21) 14: 00 tpl-sts-field (OK)
(multical21) 15: 5005 tpl-cfg 0550 (AES_CBC_IV nb=5 cntn=0 ra=0 hc=0 )
(multical21) 17: 2f2f decrypt check bytes
(multical21) 19: 0C dif (8 digit BCD Instantaneous value)
(multical21) 1a: 13 vif (Volume l)
(multical21) 1b: * 59000000 total consumption (0.059000 m3)
(multical21) 1f: 02 dif (16 Bit Integer/Binary Instantaneous value)
(multical21) 20: 6C vif (Date type G)
(multical21) 21: BD2B
(multical21) 23: 82 dif (16 Bit Integer/Binary Instantaneous value)
(multical21) 24: 04 dife (subunit=0 tariff=0 storagenr=8)
(multical21) 25: 6C vif (Date type G)
(multical21) 26: A12B
(multical21) 28: 8C dif (8 digit BCD Instantaneous value)
(multical21) 29: 04 dife (subunit=0 tariff=0 storagenr=8)
(multical21) 2a: 13 vif (Volume l)
(multical21) 2b: FFFFFFFF
(multical21) 2f: 8D dif (variable length Instantaneous value)
(multical21) 30: 04 dife (subunit=0 tariff=0 storagenr=8)
(multical21) 31: 93 vif (Volume l)
(multical21) 32: 13 vife (Reverse compact profile without register)
(multical21) 33: 2C varlen=44
(multical21) 34: FBFEFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
(multical21) 60: 02 dif (16 Bit Integer/Binary Instantaneous value)
(multical21) 61: FD vif (Second extension FD of VIF-codes)
(multical21) 62: 17 vife (Error flags (binary))
(multical21) 63: 0000
(multical21) 65: 2F skip
(multical21) 66: 2F skip

```

trsqr, 2021-11-29:
I think we can close this issue and continue in another topic.

closed 2021-11-29 by trsqr

weetmuts, 2021-11-29:
//// #define TTY_BUFSIZE 224

Great work in figuring this out! Will you report this bug to the upstream firmware repo?

Do you have time to create another helpful page on the CUL device here: 
https://weetmuts.github.io/wmbusmeterswiki/ ?

Describing essentially what you said here:
```
This is how I flash it (for some reason it fails if I use baudrate of 57600) for my atmega328 based board (this one to be exact: https://www.smart-home-komponente.de/nano-cul/nano-cul-868/):
avrdude -D -p atmega328p -P /dev/ttyUSB0 -b 115200 -c arduino -U flash:w:nanoCUL_r568_mbus_c1t1_bufsize300.hex

nanoCUL_r568_mbus_c1t1_bufsize300.zip
```

trsqr, 2021-11-30:
I did not report this, as there have not been updates to the nanoCUL firmware in the last 3 years. I presume the project is hibernating. I presume it's sort of a known thing, as you will often need to compile the firmware for your particular needs as the ATmega328 doesn't have enough flash nor RAM for all features at the same time.

I wrote a guide on the firmware flashing and sent you a PR.

pepsonEL, 2023-02-19:
Hi
I have a problem with nanoCUL and i can not found solution... Anybody help me please... I get this error in logs addon in HA

```
Started auto cul on /dev/ttyUSB1 listening on t1
(cul) Warning! Your cul firmware has a bug that prevents longer telegrams from being received.!
(cul) Please read: https://github.com/weetmuts/wmbusmeters/issues/390
(cul)         and: https://weetmuts.github.io/wmbusmeterswiki/nanoCUL.html
(cul) warning: the hex string is not proper! Ignoring telegram!
(cul) Warning! Your cul firmware has a bug that prevents longer telegrams from being received.!
(cul) Please read: https://github.com/weetmuts/wmbusmeters/issues/390
(cul)         and: https://weetmuts.github.io/wmbusmeterswiki/nanoCUL.html
(cul) warning: the hex string is not proper! Ignoring telegram!
(cul) Warning! Your cul firmware has a bug that prevents longer telegrams from being received.!
(cul) Please read: https://github.com/weetmuts/wmbusmeters/issues/390
(cul)         and: https://weetmuts.github.io/wmbusmeterswiki/nanoCUL.html
(cul) warning: the hex string is not proper! Ignoring telegram!
(cul) Warning! Your cul firmware has a bug that prevents longer telegrams from being received.!
(cul) Please read: https://github.com/weetmuts/wmbusmeters/issues/390
(cul)         and: https://weetmuts.github.io/wmbusmeterswiki/nanoCUL.html
(cul) warning: the hex string is not proper! Ignoring telegram!
(cul) Warning! Your cul firmware has a bug that prevents longer telegrams from being received.!
(cul) Please read: https://github.com/weetmuts/wmbusmeters/issues/390
(cul)         and: https://weetmuts.github.io/wmbusmeterswiki/nanoCUL.html
(cul) warning: the hex string is not proper! Ignoring telegram!
```