wmbusmeters/wmbusmeters #391: Support for Landis Staefa Warm Water meter

docbobo, 2021-11-27:
I have a Qundis warm water meter that got installed recently. It says "qwater 5.5" on the front and I used the ID printed on it to configure wmbusmeters last night.

Today, when coming back the following message was in the logs:

`(meter) WarmWasser: meter detection could not find driver for id: <my-id> mfct: (LSE) Landis Staefa electronic (0x3265) type: Warm Water (30°C-90°C) meter (0x06) ver: 0x18
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
`

I've now configured `logtelegrams=true`. Is that all I need to get the information required to support this?

docbobo, 2021-11-27:
Okay, here's captured telegram. 

> telegram=|244465329933961318067AC3000000|8C04134112000082046CBF2A01FD0C11046D0E0CBB2B|+15782

docbobo, 2021-11-27:
Okay, here's a completely different capture:

> telegram=|2D4465329933961318067AC5000000|0C13026400004C1300000000426CFFFF02BB560000326CFFFF046D3914BB2B|+9477

Judging form the structure, this looks quite similar to lse_07_17. However, using this as driver does't work.

gswdn, 2021-11-28:
Also have one of these ... or better: two. One for warm and one for cold water. Just started monitoring them and I'm a little bit annoyed that there's no driver (yet). Perhaps, we can figure that out? Unfortunately, I haven't received anything from the cold thing ... yet.

Also found some documentation to my device:
https://qundis.com/wp-content/uploads/2017/12/DST1_QWA5_EN0_WMEUP_V210_030317.pdf

That's what I received yesterday evenening:
`(telegram) DLL L=2d C=44 (from meter SND_NR) M=3265 (LSE) A=12417708 VER=18 TYPE=06 (Warm Water (30°C-90°C) meter) (driv(telegram) TPL CI=7a ACC=8a STS=00 CFG=0000 ()
Received telegram from: 12417708
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Warm Water (30°C-90°C) meter (0x06)
                   ver: 0x18
                device: rtlwmbus[00000001]
                  rssi: 73 dBm
                driver: unknown!
telegram=||2D4465320877411218067A8A0000000C13092100004C1300000000426CFFFF02BB560000326CFFFF046D1515BB2B|+0
(wmbus) skipping already handled telegram.`

Values from the display were:
- current consumption: 2.109 ... should be at the time of receiving the telegram
- mode: FS
- checksum: 5756
- due date: 31.12
- last year's consumption: 0.000 (was just installed a few weeks ago)

Already got some more during the night:
- ||2D4465320877411218067A8B0000000C13122100004C1300000000426CFFFF02BB560000326CFFFF046D3A00BC2B|+0
- ||2D4465320877411218067A8C0000000C13122100004C1300000000426CFFFF02BB560000326CFFFF046D2104BC2B|+0

Both should be a consumption value of 2.112 ... need to check with my wife later if she used warm water during the night.

I continue monitoring both meters closely and provide values. Best during the night. Hope, we can use this to reverse engineer the telegrams with some help of other's here?

By the way: I also have that heat cost allocation devices from qundis with a "kalo k1.5x" print on them. Receiving their telegrams with an unknown driver as well. Will also monitor them and raise another ticket here once I have some more data. 

@docbobo : What values do you have for the received telegrams? We need to figure out the pattern behind this.

docbobo, 2021-11-28:
At least for my variant, just adding the mapping `meter_detection.h` seems to be sufficient.

docbobo, 2021-11-28:
@gswdn, if I take the telegrams from your comment, the update build from the PR produces this:

> /wmbusmeters # /wmbusmeters/wmbusmeters --format=json "2D4465320877411218067A8B0000000C13122100004C1300000
000426CFFFF02BB560000326CFFFF046D3A00BC2B" 
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: <your-id>
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Warm Water (30°C-90°C) meter (0x06)
                   ver: 0x18
                driver: lse_07_17
/wmbusmeters # /wmbusmeters/wmbusmeters --format=json "2D4465320877411218067A8B0000000C13122100004C1300000
000426CFFFF02BB560000326CFFFF046D3A00BC2B" WarmWasser auto <your-id> NOKEY
{"media":"warm water","meter":"lse_07_17","name":"WarmWasser","id":"<your-id>","total_m3":2.112,"due_date_m3":0,"due_date":"2127-15-31","error_code":"OK","error_date":"2127-15-31","device_date_time":"2021-11-28 00:58","timestamp":"2021-11-28T07:51:38Z"}

gswdn, 2021-11-28:
@docbobo 👍 Nice! That data looks accurate.

Can you also please add and try for the regular water-meter? Probably, data structure is similar.
 
`X(LSE_07_17,  MANUFACTURER_LSE, 0x07,  0x18) \`

`(telegram) DLL L=2d C=44 (from meter SND_NR) M=3265 (LSE) A=12362945 VER=18 TYPE=07 (Water meter) (driver unknown!) DEV=rtlwmbus[00000001] RSSI=93
(telegram) TPL CI=7a ACC=8d STS=00 CFG=0000 ()
Received telegram from: 12362945
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Water meter (0x07)
                   ver: 0x18
                device: rtlwmbus[00000001]
                  rssi: 93 dBm
                driver: unknown!
telegram=||2D4465324529361218077A8D0000000C13560501004C1300000000426CFFFF02BB560000326CFFFF046D2909BC2B|+0
`

Unfortunately, I'm not that deep c++/linux/raspi dev and only running wmbusmeter from within a docker container. So, don't have all the dev tools installed.

closed 2021-11-28 by weetmuts

weetmuts, 2021-11-28:
Many thanks! Yes the lse_07_17 driver picks up all the values correctly in the telegram.
Great work!
Many thanks!

PR #393 (2021-11-28): Add Qundis watermeter 5.5 version 0x0x/0x18 to lse_07_17 
Fixes #391