wmbusmeters/wmbusmeters #392: Techem Warmwater Meter

SamLowrie, 2021-11-27:
I have a Techem warmwater meter (rado 4?) with following example telegram:
```
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2021-11-27 14:04:47.000;84;62;81651968;0x3644685068196581953772002664602423fe06a0002025e76beca215d30564ba6f75b052fadd4e2c875a84f29cb58b8a3c5dd7a0e04639<0A>"
```
```
 ./wmbusmeters 3644685068196581953772002664602423fe06a0002025e76beca215d30564ba6f75b052fadd4e2c875a84f29cb58b8a3c5dd7a0e04639
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 81651968
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x95
      Concerning meter: 60642600
          manufacturer: (HYD) Hydrometer (0x2324)
                  type: Warm Water (30Â°C-90Â°C) meter (0x06)
                   ver: 0xfe
                driver: unknown!
```
Any chance of decoding this telegram?

SamLowrie, 2021-11-28:
Some more infos:
[Product-Sheet](https://www.techem.com/content/dam/techem/downloads/geraete/wasserzaehler/84847-02-0520-MK-WasserZ-radio4-IN-D-E.pdf.coredownload.pdf)

Besides the ID another serial?/key? is printed on the display: 10162300

There is also a 2D QR-Code which reads:
```
QR-Code: 9DME0060642600 E0010162300000023
Printed:       60642600    10162300
```

It tried to use the QR-Code as key in various combination, but the telegram fails with a bad CRC. However according to the datasheet it should be OMS compliant. Even when they embed proprietary data, the CRC should work?!

weetmuts, 2021-11-28:
The QR code seems to contain the printed serial 10162300 and the telegram embedded 60642600.
You need another AES key to decode this telegram.

weetmuts, 2022-06-19:
When you have the key, please open a new issue and we can proceed to add support for this meter!

closed 2022-06-19 by weetmuts