wmbusmeters/wmbusmeters #394: Support for Qundis Q heat 5 heat meter

docbobo, 2021-11-28:
We had Qundis Q heat 5 installed in our apartment building recently. While I can see quite a few Qundis devices broadcasting, none of their IDs seem to match the meter ID I have. I believe this might be similar to #369 where there is a radio transmitter device that sends that data on behalf of the actual meter. However, in my case, all the Qundis devices seem to be detected as smoke detectors:

> Received telegram from: 45797086
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Smoke detector (0x1a)
                   ver: 0x21
                device: rtlwmbus[00000001]
                  rssi: 38 dBm
                driver: unknown!
telegram=||3E44934486707945211A7801FD08F081027C034955230082026CFFFF81037C034C41230082036CFFFF03FD17000000326CFFFF046D0F0ABC2B02FDAC7E1100|+46

I've already tried adding `X(QHEAT,      MANUFACTURER_QDS, 0x1a,  0x21)` to the meters detection, but apparently this is not sufficient and the resulting json does not seem to contain any readings.

weetmuts, 2021-11-28:
Thanks! This meter used variable length vifs, I have added support for this and pushed. 
Now we need to figure out which fields indicate smoke etc, there are several fields in this meter so it is not obvious.
Run `./build/wmbusmeters --format=json --debug 3E44934486707945211A7801FD08F081027C034955230082026CFFFF81037C034C41230082036CFFFF03FD17000000326CFFFF046D0F0ABC2B02FDAC7E1100 Smokey auto 45797086 NOKEY`

and you will get this output
```
(qsmoke) 0b: 01 dif (8 Bit Integer/Binary Instantaneous value)
(qsmoke) 0c: FD vif (Second extension FD of VIF-codes)
(qsmoke) 0d: 08 vife (Access Number (transmission count))
(qsmoke) 0e: F0
(qsmoke) 0f: 81 dif (8 Bit Integer/Binary Instantaneous value)
(qsmoke) 10: 02 dife (subunit=0 tariff=0 storagenr=4)
(qsmoke) 11: 7C vif (VIF in following string (length in first byte))
(qsmoke) 12: 03 viflen (3)
(qsmoke) 13: 49 vif (I)
(qsmoke) 14: 55 vif (U)
(qsmoke) 15: 23 vif (#)
(qsmoke) 16: 00
(qsmoke) 17: 82 dif (16 Bit Integer/Binary Instantaneous value)
(qsmoke) 18: 02 dife (subunit=0 tariff=0 storagenr=4)
(qsmoke) 19: 6C vif (Date type G)
(qsmoke) 1a: FFFF
(qsmoke) 1c: 81 dif (8 Bit Integer/Binary Instantaneous value)
(qsmoke) 1d: 03 dife (subunit=0 tariff=0 storagenr=6)
(qsmoke) 1e: 7C vif (VIF in following string (length in first byte))
(qsmoke) 1f: 03 viflen (3)
(qsmoke) 20: 4C vif (L)
(qsmoke) 21: 41 vif (A)
(qsmoke) 22: 23 vif (#)
(qsmoke) 23: 00
(qsmoke) 24: 82 dif (16 Bit Integer/Binary Instantaneous value)
(qsmoke) 25: 03 dife (subunit=0 tariff=0 storagenr=6)
(qsmoke) 26: 6C vif (Date type G)
(qsmoke) 27: FFFF
(qsmoke) 29: 03 dif (24 Bit Integer/Binary Instantaneous value)
(qsmoke) 2a: FD vif (Second extension FD of VIF-codes)
(qsmoke) 2b: 17 vife (Error flags (binary))
(qsmoke) 2c: 000000
(qsmoke) 2f: 32 dif (16 Bit Integer/Binary Value during error state)
(qsmoke) 30: 6C vif (Date type G)
(qsmoke) 31: FFFF
(qsmoke) 33: 04 dif (32 Bit Integer/Binary Instantaneous value)
(qsmoke) 34: 6D vif (Date and time type)
(qsmoke) 35: * 0F0ABC2B device datetime (2021-11-28 10:15)
(qsmoke) 39: 02 dif (16 Bit Integer/Binary Instantaneous value)
(qsmoke) 3a: FD vif (Second extension FD of VIF-codes)
(qsmoke) 3b: AC vife (Duration since last readout [second(s)])
(qsmoke) 3c: 7E vife (Reserved)
(qsmoke) 3d: 1100
{"media":"smoke detector","meter":"qsmoke","name":"Smokey","id":"45797086","status":"OK","counter_int":0,"device_date_time":"2021-11-28 10:15","timestamp":"2021-11-28T17:25:47Z"}
```

Please do a few tests, provoking the smoke detector and watching which fields change.
#AL perhaps means number of alarms? This is one of the new variable length vifs in there.

weetmuts, 2021-11-28:
>However, in my case, all the Qundis devices seem to be detected as smoke detectors:

Hmm, in this case I think you actually captured a telegram from a smoke detector, given the content of the telegram.
If it is not your detector, then we add all the field as is to the json, and wait for someone to figure out the content.

docbobo, 2021-11-28:
Hmm. Interesting. I see lots of Techem smoke detectors being detected, though I wasn't able to associate any of them with the ones in my apartment. Given the RSSI I am seeing, I am not convinced that the Qundis devices are actually coming from a completely different building... 

Let me switch back to scanning then. Perhaps I really missed a device there.

docbobo, 2021-11-28:
That said - did you notice any  vif that could theoretically hold heat allocation or similar? Maybe it's just a buggy implementation.

docbobo, 2021-11-28:
And just for completeness:

> Received telegram from: 48128850
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Smoke detector (0x1a)
                   ver: 0x23
                device: rtlwmbus[00000001]
                  rssi: 29 dBm
                driver: unknown!
telegram=||3744934450881248231A7A5C00002081027C034955230082026CFFFF81037C034C41230082036CFFFF02FD170000326CFFFF046D2514BC2B|+160

That seems to be a different version

weetmuts, 2021-11-28:
>That said - did you notice any vif that could theoretically hold heat allocation or similar? 

No, I do not think so. The HCA value has its own standardized vif, so its pretty easy to get that right.
Also smoke detectors have no standardized vif, so I suppose this is one way of doing it.

Thanks for the additional telegram. I have added it to the test suite.

docbobo, 2021-11-29:
Okay, I think you've convinced me. It seems that I might need an additional module for this particular meter.

That aside - here's one more unknown entry from the logs this morning:

> Received telegram from: 46398739
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Smoke detector (0x1a)
                   ver: 0x22
                device: rtlwmbus[00000001]
                  rssi: 18 dBm
                driver: unknown!
telegram=||3E44934439873946221A7801FD088481027C034955230082026CFFFF81037C034C41230082036CFFFF03FD17000000326CFFFF046D1B06BD2B02FDAC7E1100|+34947

BIBOLV, 2023-01-01:
Closing due to inactivity. Driver was implemented and last provided telegram parses successfully.

closed 2023-01-01 by BIBOLV