wmbusmeters/wmbusmeters #395: Qundis caloric 5.5 Heat Cost Allocator not recognized

RalfJL, 2021-11-28:
Hi,
thanks for this wonderful software and sorry for opening an issue without having a solution.

I bought two popular Qundis caloric 5.5 Heat Cost Allocator but they are not recognized by wmbusmeters. All software is freshly pulled from github last week.
rtl_wmbus drops all telegrams from this Heat Cost Allocator; aka. no output at all. 
Using rtl_433 produces some output, but it is not recognized by wmbusmeters and rtl_433 is misinterpreting the wmbus values.
Both Qundis caloric do send two different telegrams. Here are some telegrams from one of them and the warning from wmbusmeters:
 wmbusmeters --logtelegrams rtl433:c1

Started config rtl433[1234] listening on c1
No meters configured. Printing id:s of all telegrams heard!
(rtl433) dll_len adjusted to 75 from 71. Fix rtl_433? This warning will not be printed again.
Received telegram from: 96615935
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x35
                device: rtl433[1234]
                  rssi: 999 dBm
                driver: qcaloric
telegram=||4B449344355961963508780DFF5F3500822200000F0007B06EFFFF00000000FFFF00000000FFFF000000000080008000800080008000800080008000800080008000802F046D310BBC2B04D4|+33
Received telegram from: 96615935
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x35
                device: rtl433[1234]
                  rssi: 999 dBm
                driver: qcaloric
telegram=||334493443559619635087A230000200B6E0000004B6E000000426CFFFFCB086E000000C2086CFFFF326CFFFF046D310BBC2BFA9B|+49

Currently I am puzzeling if the new rtl_433 is sending something useful, because the Techem is not recognized anymore with the rtl_433 driver.
Received telegram from: 72500832
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat Cost Allocator (0x80)
                   ver: 0x69
                device: rtl433[1234]
                  rssi: 999 dBm
                driver: fhkvdataiii
telegram=||34446850320850726980A0919F290000C0370A007909C40B0A0000000000000000000000000000000000000000000000000000FFFF|+3091

But the rtl_433 driver is much better. With rtl_sdr the Techem has a lot of checksum errors 

Any help would be perfect.
Thanks
Ralf

weetmuts, 2021-11-28:
Hmm, how far away are you from the hca device? 

Both rtl_433 and rtl_wmbus acquire radio samples using rtl_sdr. Interesting that rtl_433 performs better than rtl_wmbus. @merbanan any comment, do you have some smart logic fixing read errors in rtl_433?

Your second telegram looks fine, it is expected that the qcaloric device sends two completely different telegrams.
This is handled by wmbusmeters. (I think one telegram is old proprietary format and a second wmbus compliant telegram.)

`./build/wmbusmeters --format=json --debug 334493443559619635087A230000200B6E0000004B6E000000426CFFFFCB086E000000C2086CFFFF326CFFFF046D310BBC2BFA9B HCA auto '*' NOKEY`

outputs
```
{"media":"heat cost allocation","meter":"qcaloric","name":"HCA","id":"96615935","current_consumption_hca":0,"set_date":"2127-15-31","consumption_at_set_date_hca":0,"set_date_1":"2127-15-31","consumption_at_set_date_1_hca":0,"set_date_17":"2127-15-31","consumption_at_set_date_17_hca":0,"error_date":"2127-15-31","device_date_time":"2021-11-28 11:49","timestamp":"2021-11-28T17:51:00Z"}
```

Likewise the Techem telegram looks fine.
{"media":"heat cost allocator","meter":"fhkvdataiii","name":"Foof","id":"72500832","current_hca":10,"current_date":"2021-11-28T02:00:00Z","previous_hca":0,"previous_date":"2020-12-31T02:00:00Z","temp_room_c":24.25,"temp_radiator_c":30.12,"timestamp":"2021-11-28T17:55:50Z"}

merbanan, 2021-11-28:
rtl_433 uses 1MS by default compared to 1.6MS for rtl_wmbus. This correspond almost 1 to 1 with the sampled bandwidth. If there is a stationary signal it is very likely it will be affecting the received signal. So that could be one reason rtl_433 is better in this case.

Decoding wise there is nothing in rtl_433 to error correct thing. We just verify the checksums and thats it.

RalfJL, 2021-11-29:
I did more investigation, so we might have to rename the topic here.
With rtl_sdr I am recording several Mbytes per minute. I guess that this is just noise. rtl_433 only shows a few weatherstation telegrams and FS20. Beside the 2 Qunids and the one Techem Heat cost allocator it only shows an Interlogix Security device. 
I will try to convince rtl_sdr to send less noise.
The datasheet of the Qundis caloric explains the second telegram. The Qundis is in mode FC 2.2 which means „FC“
stands for walk-by & OMS, C-Modus
walk-by = consumption values and 13 Statistics
OMS is what is correctly decoded by wmbusmeters.

So actually I thought wmbusmeters would work with rtl_433 but I think it doesn't.
Here is my wmbusmeters.conf:
loglevel=debug
device=rtl433:c1,t1
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=append
logfile=/var/log/wmbusmeters/wmbusmeters.log
shell=/usr/bin/mosquitto_pub -h 172.16.1.100 -t wmbusmeters -m "$METER_JSON" > /tmp/t 2>&1

I would expect that the HCA's would appear in mqtt, but they do not. and the debug output in the log file shows, that there is something wrong. Please see also above a value of 999db is not plausible.
rtl_433 is producing csv output which is consumed by wmbusmeters. 
There might be a misunderstanding between rtl_433 and wmbusmeters with csv data. 

This is what the debug output shows in the logfile for the second Qundis HCA:
(rtl433) checkRTL433Frame "2021-11-29 09:31:35,,,Wireless-MBus,,26867779,,,,CRC,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,47449344797786263508780dff5f350082e900000e0007b06effff05000000ffff00000000ffff000000000080008000800080008000800080008000800080008000802f046d2009bd2bc9a2,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,8,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,C,53,Heat Cost Allocator,120,0,0,0,,,,,,,,,QDS,68,74,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,<0A>"
(wmbus) parseDLL @0 76
(telegram) DLL L=4b C=44 (from meter SND_NR) M=4493 (QDS) A=26867779 VER=35 TYPE=08 (Heat Cost Allocator) (driver qcaloric) DEV=rtl433[1234] RSSI=999
(wmbus) parseELL @10 66
(wmbus) parseNWL @10 66
(wmbus) parseAFL @10 66
(wmbus) parseTPL @10 66
(dvparser) warning: unexpected end of data (dife expected)
(dvparser) warning: unexpected end of data (vif expected)
(telegram) TPL CI=78
Received telegram from: 26867779
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x35
                device: rtl433[1234]
                  rssi: 999 dBm
                driver: qcaloric
(wmbus) 00: 4b length (75 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 9344 dll-mfct (QDS)
(wmbus) 04: 79778626 dll-id (26867779)
(wmbus) 08: 35 dll-version
(wmbus) 09: 08 dll-type (Heat Cost Allocator)
(wmbus) 0a: 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
(wmbus) 0b: 0D dif (variable length Instantaneous value)
(wmbus) 0c: FF vif (Vendor extension)
(wmbus) 0d: 5F vife (duration of limit exceed last upper  is 3)
(wmbus) 0e: 35 varlen=53
(wmbus) 0f: 0082E900000E0007B06EFFFF05000000FFFF00000000FFFF000000000080008000800080008000800080008000800080008000802F
(wmbus) 44: 04 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 45: 6D vif (Date and time type)
(wmbus) 46: 2009BD2B
(wmbus) 4a: C9 dif (2 digit BCD Instantaneous value storagenr=1)
(wmbus) 4b: A2 dife (subunit=0 tariff=2 storagenr=5)
telegram=||4B449344797786263508780DFF5F350082E900000E0007B06EFFFF05000000FFFF00000000FFFF000000000080008000800080008000800080008000800080008000802F046D2009BD2BC9A2|+499

Thanks a lot for your help.

RalfJL, 2021-11-30:
Today I received a new RTL2832U USB Stick with an aluminium case and reception is much better now. Both Qundis HCA's and the Techem are now correctly reported by the wmbus driver. No matter if it is close (1.5 meter) or far (more than 10 meters with brick walls between
Here are the reports from mqtt. Values are correct.
{"media":"heat cost allocation","meter":"qcaloric","name":"QDS_26867779","id":"26867779","current_consumption_hca":7,"set_date":"2127-15-31","consumption_at_set_date_hca":0,"set_date_1":"2127-15-31","consumption_at_set_date_1_hca":0,"set_date_17":"2127-15-31","consumption_at_set_date_17_hca":0,"error_date":"2127-15-31","device_date_time":"2021-11-30 18:03","timestamp":"2021-11-30T17:02:32Z","device":"rtlwmbus[77771111153705700]","rssi_dbm":126}

{"media":"heat cost allocation","meter":"qcaloric","name":"QDS_96615935","id":"96615935","current_consumption_hca":1,"set_date":"2127-15-31","consumption_at_set_date_hca":0,"set_date_1":"2127-15-31","consumption_at_set_date_1_hca":0,"set_date_17":"2127-15-31","consumption_at_set_date_17_hca":0,"error_date":"2127-15-31","device_date_time":"2021-11-30 18:05","timestamp":"2021-11-30T17:00:19Z","device":"rtlwmbus[77771111153705700]","rssi_dbm":82}

{"media":"heat cost allocator","meter":"fhkvdataiii","name":"TCM_72500832","id":"72500832","current_hca":13,"current_date":"2021-11-30T02:00:00Z","previous_hca":0,"previous_date":"2020-12-31T02:00:00Z","temp_room_c":23.35,"temp_radiator_c":27.49,"timestamp":"2021-11-30T16:57:54Z","device":"rtlwmbus[77771111153705700]","rssi_dbm":122}
So that is great.

Using the rtl433 driver does not work for me. It reports strange values and does that in standalone mode as well. I think a HCA does not have something like: inst_power_jh_1 or inst_volume_flow_min_1
and of course there is no such thing as 999dB reception. Even with the new USB stick rtl_433 is reporting these strange values. On the other hand the values I am looking for inst_hca_0 has always been correctly reported by rtl_433. But the Techem was always wrong and showed no usable value.
The rtl_433 example rtl_433_json_to_rtlwmbus.py again is misleading because it does not calculate the checksum but always reports that the  checksum is correct which I doubt. 

If you don't mind I will close that topic and open an issue with rtl_433. For now I think the rtlwmbus driver is more reliable.
rtl_433 has it's strength in another field.

Thanks a lot for the prompt and professional help.
Ralf

closed 2021-11-30 by RalfJL