wmbusmeters/wmbusmeters #397: Supercom 581   433,82MHz Parameters?

XiaoJun9, 2021-11-29:
I tried to setup the system for scanning a watermeter with Supercom 581 at  433,82MHz. I managed the 433,82MHz which are confirmed in the logfile.
In the datasheet, the speed is mentioned to be 2400 baud. Because I did not reveive any data, I wonder how to setup the config. 
I use wmbusmeters on a ubuntu machine. The dongle is detected, the protocols are s1, t1, c1.

weetmuts, 2021-11-30:
So, wmbusmeters is mostly about the decoding side of the binary telegrams after they have been received.
What radio dongle are you using to listen? How are you invoking it?

XiaoJun9, 2021-11-30:
Thanks for your reply!
   
  
  
   I used a RTL dongle. wmbusmeters seemed to activate it, but obviously we do not have signals to receive on 866 or 433Mhz. In the meanwhile I found, that this Supercom 581 works bidirectionally and needs a kind of wakeup signal. Biside this I would be interested how to setup frequency AND baudrate in the config file.

weetmuts, 2021-12-01:
So, bidirectional wmbus is not yet supported out of the box with wmbusmeters.
But all the infrastructure is in place to do so, so it should be relatively easy to do
as long as you have documentation on what message the supercom 581 wants. 
(It might be a standardized readout request.)

But first you have to buy a dongle that can transmit the wakeup signal. The im871a can transmit telegrams.

> frequency AND baudrate
The radio data baudrate is implicit from wmbus C1/T1/S1 mode, defined by the standard and cannot be changed.

Wmbusmeters can be used for mbus as well, then you set a baudrate, but I do not think that is what want?

XiaoJun9, 2021-12-02:
> frequency AND baudrate
> The radio data baudrate is implicit from wmbus C1/T1/S1 mode, defined by the standard and cannot be changed.
Ok, thats importatnt. So as long as we are talking about wmbus, we don't have to care about the baudrate, because its implicit.

Actually I dont have the respective dongle for bidirectional transfers. I would consider this topic again later.
Thanks for your support!

closed 2021-12-02 by XiaoJun9