wmbusmeters/wmbusmeters #398: AT-WMBUS-MR-01

ma73u53, 2021-11-30:
Hello,
At the beginning I would like to thank you for the great package! It works perfectly for apator162 (AT-WMBUS-16-2) and FC0012 tuner (Realtek RTL2838UHIDIR)

I have tried to integrate AT-WMBUS-MR-01 water heat meter but with no success.
Tried listening rtwmbus on various frequencies, from 868.0MHz to 868.95MHz on c1, s1, t1.
I receive only telegrams from AT-WMBUS-16-2 meters.
Name of the water heat meter is APATOR ELF.
Attached you can find technical description of that device:
[i-en-009-2017-elf-13-01.pdf](https://github.com/weetmuts/wmbusmeters/files/7624787/i-en-009-2017-elf-13-01.pdf)
In the description there is information, that operating frequency of the device is 868.95MHz, mode T1.

Is there any posibility to integrate AT-WMBUS-MR-01 with wmbusmeters at the moment, or maybe in the future?

Thank you in advance for the information and for help.

weetmuts, 2021-11-30:
We sure can try! But we need some telegrams to start with and potentially an encryption key if the meter sends encrypted telegrams. You can for example collect telegrams with --logtelegram

weetmuts, 2021-11-30:
It seems like the heat meter will only transmit Wireless MBus if the proper interface module is installed in the meter. Can you verify this?

ma73u53, 2021-11-30:
Attached you will find logged telegrams related to query:
**wmbusmeters --logtelegrams --listento=c1,s1,t1 rtlwmbus:868.950M**
[logtelegram.txt](https://github.com/weetmuts/wmbusmeters/files/7625471/logtelegram.txt)

I am also attached photos of meters - seems like wmbus interfaces are installed - there are IDs on them.
[image]
[image]

weetmuts, 2021-12-09:
Cool, the telegrams look ok but you have to have a decryption key, since the telegrams are encrypted.

weetmuts, 2021-12-09:
Ah, your log only contains the water meter. Which is already supported by wmbusmeters you just have to use
the key 00000000000000000000000000000000

Try running longer, like a week or so, and look for telegrams that are not the water meter.

weetmuts, 2022-01-13:
Any decrypted telegrams from your AT-WMBUS-MR-01 ?

BIBOLV, 2023-01-01:
Closing due to inactivity. Please reopen issue if you manage to capture more unencrypted telegrams.

closed 2023-01-01 by BIBOLV

inatlewski, 2023-03-13:
@ma73u53, I'm not sure if you noticed but the driver for Apator Elf has been implemented some time ago, and the heat meter is now supported in wmbusmeters. Could you verify it once again? I have the same heat meter and I'm planning to buy AT-WMBUS-MR-01 module.

Commit: https://github.com/wmbusmeters/wmbusmeters/commit/e02c985821e4411c144fc8dcf3dd1de7fd323e23