wmbusmeters/wmbusmeters #402: Diehl driver support

Nimtek123, 2021-12-02:
Hi I am looking for support for drivers for Diehl meter

[image]
[image]
[image]
r

weetmuts, 2021-12-02:
Cool, have you captured any telegrams so far? Do you know the model/name/version of the meter?

weetmuts, 2021-12-02:
Or rather the addon that seems to listen to pulses from the actual water meter and resend them.

Nimtek123, 2021-12-02:
Hi thanks for the quick response.
Here are some captured telegrams below 
        "manufacturer": "(HYD) Hydrometer (0x2324)",
        "data": "19442423860775035048A261520015BE57C53B10004E72915FA2",
        "data": "19442423860775035048A251520015BEB6B2E1ED623A18FC74A5",

Nimtek123, 2021-12-02:
Make and Model attached
[image]

Nimtek123, 2021-12-02:
Another Model 
[image]

weetmuts, 2021-12-02:
It seems like the izar driver works for this meter. This are the values from your telegrams, do they make sense?
521.05	519.147
521.602	519.147
521.072	519.147

Note that the reported values might not correlate with what is written on the mechanical meters, that depends
on if they synced the mechanics with the wmbus attachement.

See if you run:
wmbusmeters --format=json auto:t1 MyWater izar 48500375 NOKEY
and monitor the output when you use some water and not, to see if your water usage triggers the counter.

Nimtek123, 2021-12-02:
Thanks @weetmuts 
I will run the izar one. 
Any update on the Hydro Model?

weetmuts, 2021-12-02:
Should be the same driver.

Nimtek123, 2021-12-03:
Hi @weetmuts 
With every response we getting the below
(meter) MyWater: meter detection did not match the selected driver izar! correct driver is: unknown!
(meter) Not printing this warning again for id: 48500375 mfct: (HYD) Hydrometer (0x2324) type: Water meter (0x07) ver: 0x86
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination

{"media":"water","meter":"izar","name":"MyWater","id":"48500375","prefix":"","serial_number":"000000","total_m3":521.05,"last_month_total_m3":519.147,"last_month_measure_date":"2021-11-15","remaining_battery_life_y":9,"current_alarms":"no_alarm","previous_alarms":"leakage","transmit_period_s":8,"manufacture_year":"0","timestamp":"2021-12-02T18:17:24Z"}

Nimtek123, 2021-12-06:
Hi @weetmuts 
Checking if there is Any update on the above. Thanks much appreciated

Nimtek123, 2021-12-07:
Hi @weetmuts hope you are well. We are using the wmbusmeters service differently

The data concentrator sends a file to our file server. The fileserver then run the decoding hex string command to decode the data and we store it on our database.

Then when we run the below on our VM sitting on the cloud server It takes a bit of time to process the data. So we run the risk of timing out before i can send the decoded data out.

 `wmbusmeters --format=json 1944A511780700963750A211140013BAAA735B5973B5919781C6 auto meter izar 50379600 NOKEY`

weetmuts, 2021-12-07:
Thanks, I added the new mapping to wmbuseters and pushed it to github. So now it should no longer complain.
Your command has an auto too many, try this line instead which executes quickly:
```
wmbusmeters --format=json 1944A511780700963750A211140013BAAA735B5973B5919781C6 Water izar 50379600 NOKEY
```

Nimtek123, 2021-12-08:
@weetmuts  Thank you very much much appreciated for your assistance

closed 2021-12-09 by weetmuts