wmbusmeters/wmbusmeters #403: lse_07_17: reporting 0 as total

docbobo, 2021-12-02:
In addition to the regular telegrams, e.g.

`telegram=|2D4465329933961318067ADA000000|0C13567100004C1300000000426CFFFF02BB560000326CFFFF046D2307A12C|+117507`

I am also seeing short ones, e.g.

`telegram=|2A4465329933961318067AD8800000|8C04130070000082046CBE2B01FD0C11046D1800A12C02FDAC7E9B2E|+91644`
`telegram=|244465329933961318067AE1000000|8C04130070000082046CBE2B01FD0C11046D010CA22C|+8101`

Those are actually messing up my statistics, as they are reporting the total as `0`. Are those legit telegrams?

weetmuts, 2021-12-02:
They do look legit. Some meters sends some data in one telegram and other that in a second telegram.
wmbusmeters remembers the values between telegrams so you should not be getting zeroes if you are running wmbusmeters as a daemon or other long running process. But the first telegram might be zero though before wmbusmeters has seen a telegram with a total_m3 value.

When I run them through wmbusmeters, without restarting wmbusmeters I get:

 ./build/wmbusmeters simulation_oddones.txt Vatten auto 13963399 NOKEY
Vatten	13963399	7.156 m3	0 m3	2127-15-31	OK	2127-15-31	2021-12-01 07:35	2021-12-02 18:19.05
Vatten	13963399	7.156 m3	0 m3	2127-15-31	OK	2127-15-31	2021-12-01 00:24	2021-12-02 18:19.05
Vatten	13963399	7.156 m3	0 m3	2127-15-31	OK	2127-15-31	2021-12-02 12:01	2021-12-02 18:19.05

Looks ok to me.

weetmuts, 2021-12-02:
What is your meter name/brand/model?

LukasACRIOS, 2021-12-07:
Hello @weetmuts and thank you for pointing me in this direction! I am currently dealing with remote Qundis readouts and from what I could see it is sending 2 telegrams as mentioned above. The walk-by one is transmitted every 112 seconds, the OMS is transmitted every 450 seconds. 

I am attaching below different telegrams for different modes. The next challenge is to add any sort of logic on the concentrator side (as I use the WMBUS to NB-IoT concentrator) so it sends only the OMS telegram. The current logic is that it sends the data from a given ID (the ID's are set as a filter) and then it ignores it. This is not sufficient because it can send the wrong telegram and I won't have any data. 

One idea could be to receive OMS telegrams only between 6pm - 8am, because the walk-by transmits only from 8am to 6pm. [https://oms-group.org/fileadmin/products/qundis/q_caloric_5_5_2017/DST2-QCAL-GB0-P2P3_Qcal_55_260717_V240.pdf](source)

 {
    "timestamp": "2021-12-07T11:03:14.441270",
    "unit": "901288002328172",
    "wmbusId": "28177654",
    "found": true,
    "data": "SUSTRFR2Fyg1CHgN/181AIIkAADwAAewbv//UgMAAP//AAAAAL4rJwMAAACAAIAAgACAAIAAgAAAAAAAABMAiACsAC8EbQUMpyw=",
    "owner": "Lukas",
    "parsed": {
      "media": "heat cost allocation",
      "meter": "qcaloric",
      "name": "WMBUS",
      "id": "28177654",
      "current_consumption_hca": 0,
      "set_date": "",
      "consumption_at_set_date_hca": 0,
      "set_date_1": "",
      "consumption_at_set_date_1_hca": 0,
      "set_date_17": "",
      "consumption_at_set_date_17_hca": 0,
      "error_date": "",
      "device_date_time": "2021-12-07 12:05",
      "timestamp": "2021-12-07T16:13:41Z"
    }
    
    {
    "timestamp": "2021-12-07T14:25:28.050225",
    "unit": "901288002328172",
    "wmbusId": "28177654",
    "found": true,
    "data": "SUSTRFR2Fyg1CHgN/181AII/AABeAQewbv//UwMAAP//AAAAAL4rJwMAAACAAIAAgACAAIAAgAAAAAAAABMAiACsAC8EbRsPpyw=",
    "owner": "Lukas",
    "parsed": {
      "media": "heat cost allocation",
      "meter": "qcaloric",
      "name": "WMBUS",
      "id": "28177654",
      "current_consumption_hca": 0,
      "set_date": "",
      "consumption_at_set_date_hca": 0,
      "set_date_1": "",
      "consumption_at_set_date_1_hca": 0,
      "set_date_17": "",
      "consumption_at_set_date_17_hca": 0,
      "error_date": "",
      "device_date_time": "2021-12-07 15:27",
      "timestamp": "2021-12-07T16:14:33Z"
    }
    
      {
    "timestamp": "2021-12-07T12:38:10.276406",
    "unit": "901288002328172",
    "wmbusId": "28177654",
    "found": true,
    "data": "MUSTRFR2Fyg1CHoxAAAgC25TAwBLbgAAAEJs///LCG4nAwDCCGy+KzJs//8EbScNpyw=",
    "owner": "Lukas",
    "parsed": {
      "media": "heat cost allocation",
      "meter": "qcaloric",
      "name": "WMBUS",
      "id": "28177654",
      "current_consumption_hca": 353,
      "set_date": "2127-15-31",
      "consumption_at_set_date_hca": 0,
      "set_date_1": "2127-15-31",
      "consumption_at_set_date_1_hca": 0,
      "set_date_17": "2021-11-30",
      "consumption_at_set_date_17_hca": 327,
      "error_date": "2127-15-31",
      "device_date_time": "2021-12-07 13:39",
      "timestamp": "2021-12-07T16:14:11Z"
    }

LukasACRIOS, 2021-12-08:
Just a quick update  - it seems like I will be able to filter it by CI field which should be "7A". If anyone wants to take a look at it as well I have around 1500 telegrams from around 110 IDs taken every hour which I can share with no problem.

docbobo, 2021-12-09:
Yes, you are absolutely right, @weetmuts. Once the large telegram is received, the smaller ones won't send zero. So it was merely a timing issue in my case.

That said, could you add an option to process those diagrams unless a full one was received? Apparently, Home Assistant treats that was a meter reset. So every time I restart wmbusmeter, I risk scerwing up my statistics.

[image]

LukasACRIOS, 2021-12-09:
In my case, I was getting 3 different telegram lengths - 24B, 31B, and 49B. Only the 31B one was sending data, however, I checked the data from other Qundis HCAs (from what I could find) and the lengths were different, so I would not rely solely on that. 

What I found is that no matter the length, the control field within the OMS telegrams (with data) is always "7A".

weetmuts, 2021-12-09:
I have pushed some new functionality, can you build the latest wmbusmeters and do this command on your different types of telegrams?

wmbusmeters --analyze 2D446532....

and post the output here? Just out of curiosity, I do not think it will give is significantly more info.

It seems to me the problem with meters sending multiple telegrams with different information has no generic solution.

Sometimes you want data asap (even if it is not all data) sometimes you want some data asap and other can wait,
sometimes you want to ignore some data. Sometimes you can wait until all telegram types has arrive and then you can start relaying the meter state.

It seems like we could add setting: 

meterstate=relay_immediately 
meterstate=waitforall_before_relay 
metersend=waitforimportant_before_relay (typically the actually measurement, but do not care about model/version etc)
metersend=relay_only_important (ignore anything that does not have the total measurement)

The important values are those selected to be printed when you run wmbusmeters without --format=...
It is selected in the call to addPrint and is the  second to last boolean.

This setting could be applied globally or per meter.

What do you think?

docbobo, 2021-12-09:
I would always want tine information as quickly as possible 😉 But it has to have value - and in my case that's not given.

Theoretically, there is also another option in addition to the ones you've been offering (which I like a lot, btw). And that is persisting the meter state between the runs and using the last persisted state as a basis. That would mean that I wouldn't be receiving 0 in those conditions but whatever the previous value was.

If you are going with the settings, I suspect this needs to be done per meter, doesn't it? For example, I really want accurate information from my water meter, but if there's a fire...

LukasACRIOS, 2021-12-09:
I will give it a try @weetmuts. Just to be clear, does the current: latest tag contains the changes? Thank you

LukasACRIOS, 2021-12-09:
I have tried the latest build with SHA sha256-6b4cc0fa2df45d662a7ffd22bb3ccd9f186d62d6a009e262f964f810a8d16f35 and the output seems to be the same as before. Am I missing something? 

```
{
  "media": "heat cost allocation",
  "meter": "qcaloric",
  "name": "WMBUS",
  "id": "28177654",
  "current_consumption_hca": 353,
  "set_date": "2127-15-31",
  "consumption_at_set_date_hca": 0,
  "set_date_1": "2127-15-31",
  "consumption_at_set_date_1_hca": 0,
  "set_date_17": "2021-11-30",
  "consumption_at_set_date_17_hca": 327,
  "error_date": "2127-15-31",
  "device_date_time": "2021-12-07 13:39",
  "timestamp": "2021-12-09T14:11:59Z"
}

{
  "media": "heat cost allocation",
  "meter": "qcaloric",
  "name": "WMBUS",
  "id": "28177654",
  "current_consumption_hca": 0,
  "set_date": "",
  "consumption_at_set_date_hca": 0,
  "set_date_1": "",
  "consumption_at_set_date_1_hca": 0,
  "set_date_17": "",
  "consumption_at_set_date_17_hca": 0,
  "error_date": "",
  "device_date_time": "2021-12-07 15:27",
  "timestamp": "2021-12-09T14:12:18Z"
}

{
  "media": "heat cost allocation",
  "meter": "qcaloric",
  "name": "WMBUS",
  "id": "28177654",
  "current_consumption_hca": 0,
  "set_date": "",
  "consumption_at_set_date_hca": 0,
  "set_date_1": "",
  "consumption_at_set_date_1_hca": 0,
  "set_date_17": "",
  "consumption_at_set_date_17_hca": 0,
  "error_date": "",
  "device_date_time": "2021-12-07 12:05",
  "timestamp": "2021-12-09T14:12:36Z"
}
```

weetmuts, 2021-12-09:
Did you use --analyze as I wrote?

LukasACRIOS, 2021-12-10:
Hello @weetmuts , sorry for the delay. I did try it with --analyze and it works like a charm. I will do some further testings as I found a similar case to be with a Techem water meter as well. If I find anything, I will continue with a matching issue thread.

Are you thinking about implementing this --analyze feature within a JSON parser as well? Meaning that for the proprietary telegram it would give you the "Cannot parse the data", since there is no value anyway. 

Thank you! 

[image]
[image]

weetmuts, 2021-12-10:
I am not sure I understand your question about json.

The lines above are green because, the content has been understood
and copied to the json. Except for the varlen, where it is a bug
that the data has been marked with addMoreExplanation in meter_qcaloric.cc
check for the last addMoreExplanation in the processContent.
It should be red.

However most other fields that have not been understood or copied 
to the json, are marked as red.

weetmuts, 2022-05-26:
Hello @LukasACRIOS @docbobo 

I have now improved the driver format so that it is much easier to write.
Now fields are declared, and if no corresponding data has been found
in the telegram, the field will be reported with the value null.

When a value finally arrives it will switch from null to the proper value.

If the field is marked as OPTIONAL, then the "key":null will not
even be printed in the json.

Does this help you understand your telegrams better?

weetmuts, 2022-05-26:
If you read the refactored driver:
https://github.com/weetmuts/wmbusmeters/blob/master/src/driver_lse_07_17.cc

You can see that fields are marked with:
PrintProperty::IMPORTANT
PrintProperty::OPTIONAL 

I can now implement:
resendmeterstate=always 
I.e. non-optional fields with no value is reported as null. optional fields with no value is not printed at all.

resendmeterstate=waitforimportant
I.e. no mqtt/shell invocation happens as long as the fields marked IMPORTANT have no value.
Other fields might be null though.

resendmeterstate=waitforall
I.e. no nulls allowed. Only optional fields are permitted to have no value (and thus not be printed.)

You can test the analyze function online, and see the null values in the first two telegrams:
https://wmbusmeters.org/analyze/244465329933961318067AE10000008C04130070000082046CBE2B01FD0C11046D010CA22C

https://wmbusmeters.org/analyze/2A4465329933961318067AD88000008C04130070000082046CBE2B01FD0C11046D1800A12C02FDAC7E9B2E

https://wmbusmeters.org/analyze/2D4465329933961318067ADA0000000C13567100004C1300000000426CFFFF02BB560000326CFFFF046D2307A12C

weetmuts, 2022-05-26:
With this solution, you can either drop telegrams with null values.

Or you can ask wmbusmeters to drop them for you with 
resendmeterstate=waitforimportant
assuming that the right fields in lse_07_17 has been marked IMPORTANT,
I am not so sure, please check.

weetmuts, 2022-05-26:
If you use resendmeterstate=waitforall
then the initial mqtt message might take a while (depending on how often the other radio telegrams are sent), 
but after the internal meter state is filled with the values, then every single telegram will trigger a mqtt message.

weetmuts, 2022-05-26:
Or perhaps everything that is not IMPORTANT is OPTIONAL?
What is the easiest thing to do in HASS? 
Ignore null values or ignore json objects lacking the necessary keys?

BIBOLV, 2023-01-01:
Closing due to inactivity.

closed 2023-01-01 by BIBOLV