wmbusmeters/wmbusmeters #405: Bug in dockerfile?

gswdn, 2021-12-02:
Hi.

I tried to build the docker images for my raspi on my own using
`docker buildx build --platform linux/arm . --tag wmbusmeters:testing --file Dockerfile.armhf --load`

Got an error message. Seems, line
`COPY docker/docker-entrypoint.sh /wmbusmeters/docker-entrypoint.sh` 

is not correct. Needed to change it to
`COPY ./docker-entrypoint.sh /wmbusmeters/docker-entrypoint.sh`

Assuming, this needs to be changed in all dockerfiles

Did I do anything wrong or is this really a bug?

Also, as I am quite new to this all, it took me a while to figure out, how to build on an Intel machine for raspi ARM CPU. Also suggest to add the docker buildx build command from above to docker readme. Ok? Can create a PR for this.

BIBOLV, 2021-12-22:
Hi,
Thanks for reporting issue - fix and PR has been created. After fix has been applied it will be possible to build docker images also locally by running `docker build https://github.com/weetmuts/wmbusmeters.git#:docker`. Why build was working on CI/CD platforms but not locally is a mystery, but might be something to do with versions.
Existing docker implementation is intended to be used with CI/CD systems for building docker images for wmbusmeters users to be able to use docker advantages by simply running containers. Local build, of course, is possible by experienced users with command provided above or one that you came up with. So with that in mind there have not been created manual for local docker image build.

BIBOLV, 2023-01-01:
Closing due to inactivity.

closed 2023-01-01 by BIBOLV