wmbusmeters/wmbusmeters #406: (LSE) Landis Staefa electronic (0x3265) type: Warm Water (30Â°C-90Â°C) meter (0x06)

safrandee, 2021-12-02:
`
Started auto rtlwmbus[00000001] listening on c1,s1,t1
(meter) Warmwasser: meter detection could not find driver for id: 12xxxxx mfct: (LSE) Landis Staefa electronic (0x3265) type: Warm Water (30Â°C-90Â°C) meter (0x06) ver: 0x16
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
`
I don't know why, because my cold water meter looks similar. But the driver for the warm water meter is unknown.

Telegram for the cold water meter:
`Received telegram from: 83xxxxx
manufacturer: (LSE) Landis Staefa electronic (0x3265)
type: Heat Cost Allocator (0x08)
ver: 0x34
device: rtlwmbus[00000001]
rssi: 90 dBm
driver: whe5x`

Telegram for warm water:
`Received telegram from: 12569528
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Warm Water (30°C-90°C) meter (0x06)
                   ver: 0x16
                device: rtlwmbus[00000001]
                  rssi: 68 dBm
                driver: unknown!`

Any suggestions?

gswdn, 2021-12-03:
I also have LSE cold- and warm water meters. They report with the same manufacturer id as yours.
Perhaps, you not just need to set the driver manually, but also add a mapping to [meter_detection.h](https://github.com/weetmuts/wmbusmeters/blob/master/src/meter_detection.h)

Needed to do this for my cold-water meter reporting type 0x07 and ver 0x18. [My pull request](https://github.com/weetmuts/wmbusmeters/pull/404) is still open for that. So, I don't know if this is correct and accepted. But it works for me in my fork.

safrandee, 2021-12-03:
Hallo Thomas,
danke erstmal für deine Antwort.
Du hast nicht zufällig ein fertiges Dockerimage? Ich habe nämlich absolut keine Ahnung, wie ich das anstellen soll.
Gruß!

Hello Thomas,
thank you first for your answer.
You don't happen to have a finished Docker image? Because I have absolutely no idea how to do it.
Greeting!

gswdn, 2021-12-03:
I'm also quite new to all of this. Took me a while to figure out how to build docker images.
You can try https://github.com/gswdn/wmbusmeters/blob/safredee/docker/safrandee.tar
Includes my changes and also added a mapping for your device.
Please let me know, if this works. Then, I can do a pull request again, so it becomes part of the official code.

safrandee, 2021-12-03:
Many thanks for your effort.
After trying a lot to find out how to pack the image in a container, the error message came up: `WARNING: The requested image's platform (linux / arm / v7) does not match the detected host platform (linux / amd64) and no specific platform was requested.`
Wrong platform ;-)

gswdn, 2021-12-03:
Normally, I do Linux things only on my raspi. But also wasn't able to build the docker image there - don't remember the error message.
So, installed Debian in a Virtual Box VM on my Intel box and experienced the same platform challenge as you.

Solved it by creating the image like this in my Debian Intel VM.
`docker buildx build --platform linux/arm . --tag testing --file Dockerfile.armhf --load`
This takes a quite long time - 40 mins or so. Don't like to know, how long it would take on my rapsi. Or it's faster there, because it's then a native architecture?

Anyway. Then, you can export the image and import on your raspi.

And also needed to adjust the dockerfiles; see discussion https://github.com/weetmuts/wmbusmeters/issues/405

safrandee, 2021-12-05:
Thank you very much, but you misunderstood something. I have a small server running Linux and I want to run the Docker image on it. In the meantime I have tried to compile the program myself according to the instructions. Unfortunately, this did not lead to success because I encountered the error: #188 
I followed WEETMUTS 'problem solving, but it didn't work for me ...

gswdn, 2021-12-06:
Ups. Could be. For whatever reason, I assumed, you run that docker image on your raspi ... as I do. And you tried to create the raspi-docker-image on a regular Intel system ... like I did. And that doesn't work.
Seems to be different for you. So you have AMD64 and I provided an ARM container. Right? That cannot work.

Ok. I can provide an AMD64 docker image for testing?! The raspi-image, which I created for myself, is working and I added your mapping there as well.
But that must not solve #188 for sure.

safrandee, 2021-12-06:
no idea what happened. but when I reinstalled the docker image, the hot water meter also works correctly.
Thank you very much, dear thomas, for taking care of my concerns. The community lives through people like you.
I close this issue.

closed 2021-12-06 by safrandee

gswdn, 2021-12-07:
Thank you :). Nice, that it's now working for you, @safrandee.

One more question: do you now use the the docker image which I provided (https://github.com/gswdn/wmbusmeters/blob/safredee/docker/safrandee.tar)? 
I'm asking because this image contains a new mapping for your device and I had to raise a pull request to get this included into the regular wmbusmeters image.
If you don't use and need this additional mapping: also fine. Then I can just delete this branch in my repo.

safrandee, 2021-12-07:
i couldn't use your dockerfile because i don't have an arm system.

gswdn, 2021-12-09:
Ok. Makes sense. Thank you. Then, the meter mapping is not required and I can delete this branch.