wmbusmeters/wmbusmeters #407: No telegrams at all

raszuk, 2021-12-03:
Hi, 

I got nanoCUL868. Trying (so far via console) to listen to t1 telegrams sent my water meters. Nothing shows up. 

I flashed it with many different firmware-s including those in https://www.smart-home-komponente.de/support/firmware/ no effect. 

Flashing is correct as when on the console I see correct version after "V" command. 

Set X01 or X21 ... even tried Ar Zr etc ... nothing shows up. 

Only on "X<enter>" I see counter slowly incrementing to 900 then stops. No idea what this counter shows ... number of received telegrams, but why they are not shown on the serial port ? 

Any help would be highly appreciated as I already spent many hours on this and I am close to give up :) 

Kind regards,
R.

weetmuts, 2021-12-09:
If you run wmbusmeters --debug auto:t1 
and post the output here, then we can take a look.

raszuk, 2021-12-09:
Thank you for looking at this issue. Below is the output requested:

robert@nuc1eo:~/wmbus/wmbusmeters$
robert@nuc1eo:~/wmbus/wmbusmeters$ sudo wmbusmeters --debug auto:t1
(wmbusmeters) version: 1.5.0-3-g02e8404
(config) number of meters: 0
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary ""
(amb8465) reading response... 2
(serial) received binary ""
(amb8465) reading response... 3
(serial) received binary ""
(amb8465) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a/im170a) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect rc1180)
(serial /dev/ttyUSB0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial /dev/ttyUSB0) sent "30"
(serial) EVENT thread interrupted
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0 (detect rc1180)
(rc1180) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect cul)
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary "3F2028F8807820697320756E6B6E6F776E2920557365206F6E65206F662042206220432065204620472069204B206C204D20522054207420562057205820780D0A"
(cul) probe response "? (<F8><80>x is unknown) Use one of B b C e F G i K l M R T t V W X x<0D><0A>"
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary "5620312E3637206E616E6F43554C3836380D0A"
(cul) probe response "V 1.67 nanoCUL868<0D><0A>"
(serialtty) closed /dev/ttyUSB0 (detect cul)
(cul) are you there? yes
(main) opening t1
(serial) EVENT thread interrupted
Started auto cul on /dev/ttyUSB0 listening on t1
(cul) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (cul)
(main) regular reset of cul  on /dev/ttyUSB0 will happen every 82800 seconds
(cul) set link mode t
(serial /dev/ttyUSB0) sent "6272740A0D"
(serial) EVENT thread interrupted
(serial) received binary "544D4F44450D0A"
(cul) checkCULFrame "TMODE<0D><0A>"
(cul) no leading 'b' so it is text and no frame
(cul) received "TMODE"(serial /dev/ttyUSB0) sent "5832310A0D"
(wmbus) no alarm (expected activity) for cul
(serial) EVENT thread interrupted
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop

and nothing shows up. 

Many thx, R.

weetmuts, 2021-12-09:
It looks good from the perspective of wmbusmeters it talks to it and sets T1 mode and get TMODE reply from the cul.
What meter are you listening to? Is it far away? In a cellar, lots of heavy walls?
Are you sure it is transmitting on 868MHz T1 linkmode?

raszuk, 2021-12-09:
I am testing this is residential building where there is about 100 of meters present which do transmit on t1. Those are known as BMeters meters. Yes those should all be 868 MHz. The closest meters are just 10 m away with no walls - still nothing shows up irrespective for how long I run it. 

What is interesting in this nanoCUL is the X counter increase to 900 then stops. Also nothing shows up on the console when I serial into the nanoCUL. 

Also I found about 6 different flavors of firmware which is supposed to work with nanoCUL868 and xM-BUS ... but I have no clue which one to use.

weetmuts, 2021-12-09:
Another user wrote this page on the CUL. Perhaps it can help?
https://weetmuts.github.io/wmbusmeterswiki/nanoCUL.html

raszuk, 2021-12-13:
That was really helpful !!!! Huge thx ... now I am getting first telegram - at least from one device - heat meters. Water meters radios are still dead :( 

But for the heat meter apparently the driver is missing ... it is APATOR Radio Model: APT-ELF2-WMBUS-1 

Is there any guide how to define a driver if not already part of the distro ? Or maybe I need to specify it somehow in the device file ? 

(wmbusmeters) logging started 2021-12-13 16:41:01 using 1.5.0-3-g02e8404
[2021-12-13_16:41:03] Started auto cul on /dev/ttyUSB0 listening on t1
(wmbusmeters) waiting for telegrams
[2021-12-13_16:41:05] (memory) rss 3100672 peak 2.95 MiB
[2021-12-13_16:41:16] (meter) ApatorCieplomierz1: meter detection could not find driver for id: 21400916 mfct: (APA) Apator, Poland (0x601) type: Heat volum
e at inlet meter (0x0c) ver: 0x42
[2021-12-13_16:41:16] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2021-12-13_16:41:16] (meter) to add support for this unknown mfct,media,version combination
(wmbusmeters) started meter 1 (ApatorCieplomierz1 21400916 auto)
telegram=|4E44010616094021420C7AD70040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A27020A5E3004046D2630AD2C02FD1700002F2F2F2F2
F2F2F2F2F2F2F|+15
telegram=|4E44010616094021420C7AD80040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A27020A5E3004046D2630AD2C02FD1700002F2F2F2F2
F2F2F2F2F2F2F|+31
telegram=|4E44010616094021420C7AD90040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A27020A5E3004046D2630AD2C02FD1700002F2F2F2F2
F2F2F2F2F2F2F|+47
robert@nuc1eo:/var/log/wmbusmeters/meter_readings$

I have defined the meter in: 

robert@nuc1eo:/etc/wmbusmeters.d$ more APATOR_HEAT1
name=ApatorCieplomierz1
id=21400916
key=<XXXXXXXXXXXXXXXXXXXXXXXXXX> (replaced by the one received from vendor)

Many thx,
R.

weetmuts, 2021-12-13:
Neat! There is support for the elf heatmeter already. But it is hardcoded to expect kWh in the telegram
and this version of the heat meter sends MJ. Do you want to try to fix meter_elf.cc
yourself? This commit shows an identical fix.

```
git show e92ebe3d9a70ae5809d1cb5baf380d8d1f4f4c3d
```

You also need to add:
   X(ELF,       MANUFACTURER_APA,  0x0c,  0x42) \

to meter_detection.h

raszuk, 2021-12-13:
Hello Fredrik,

Fantastic !  I recompiled it just adding the detection part. It works !!!!
I am not worried about GJ vs kWh as heat meter can be configured to report
required value - in any case I am testing the radio not the actual heat
amount or water volume. The meters are not connected as I want to carry
them around :)

The wmbus reports this now:

(wmbusmeters) logging started 2021-12-13 20:42:07 using 1.5.0-30-gd609d64
with local changes
[2021-12-13_20:42:09] Started config cul on /dev/ttyUSB1 listening on t1
using fq 868.95M
(wmbusmeters) waiting for telegrams
[2021-12-13_20:42:11] (memory) rss 3014656 peak 2.87 MiB
(wmbusmeters) started meter 1 (MyELF2 21400916 auto)
telegram=|4E44010616094021420C7A5D0040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A35020A5E2702046D2934AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+8

***@***.***:/var/log/wmbusmeters/meter_readings$ more MyELF2
{"media":"heat volume at
inlet","meter":"elf","name":"MyELF2","id":"21400916","meter_date":"","total_energy_consumption_kwh":0,"cur
rent_power_consumption_kw":0,"total_volume_m3":0,"total_energy_consumption_at_date_kwh":0,"flow_temperature_c":23.5,"return_tempera
ture_c":22.7,"external_temperature_c":127,"status":"OK","operating_time_h":0,"version":"","battery_v":0,"timestamp":"2021-12-13T19:
42:15Z","device":"cul","rssi_dbm":-63}

Actually what I am testing is ELF2 not ELF which has following registries:

[image: image.png]

But for now I have much bigger worries:

#1 - When the ELF2 meter is about 1 meters away from nanoCUL receiver the
radio signal is -50 dBm when it is 5 meters away it drops to -63 dBm when
it is 10 meters away no reception ... again all open rooms no walls.
Something is very odd with this nanoCUL receiver or antenna.

#2 - The water meter is not sending anything or I am not receiving anything
from it. It is also Apator  AT-WMBUS-16-2 ... likewise I am not receiving
anything from BMeters radio-s.  All in open rooms direct line of sight.

Do you know what could be the issue ? Is there some other USB receiver I
should purchase which is rock solid and proven to work well with
wmbusmeters software ?  And does not cost a fortune :) ?

Many many thx for your help,
R.

weetmuts, 2021-12-13:
I believe the apator can be configured only to send during busines hours and not weekends. Could that be it?

Also reception distance with the nanocul is not as good as the other dongles. I have done some simple tests and written down the results here: https://weetmuts.github.io/wmbusmeterswiki/ANTENNA.html

weetmuts, 2021-12-13:
the im871a is my recommendation. an rtlsdr metal case is also good, but you spend a lot of cpu decoding the radio signals.

raszuk, 2021-12-13:
Yes they are supposed to send between 7:00 am and 18:00 local time. I was
trying earlier during the day - nothing was received.

Another interesting observation ... after wmbusmeters service restart I
resume getting more telegrams from ELF2. Previously it was stuck in getting
only one.

(wmbusmeters) logging started 2021-12-13 20:42:07 using 1.5.0-30-gd609d64
with local changes
[2021-12-13_20:42:09] Started config cul on /dev/ttyUSB1 listening on t1
using fq 868.95M
(wmbusmeters) waiting for telegrams
[2021-12-13_20:42:11] (memory) rss 3014656 peak 2.87 MiB
(wmbusmeters) started meter 1 (MyELF2 21400916 auto)
telegram=|4E44010616094021420C7A5D0040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A35020A5E2702046D29
34AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+8
(wmbusmeters) shutting down
(wmbusmeters) logging started 2021-12-13 22:45:06 using 1.5.0-30-gd609d64
with local changes
[2021-12-13_22:45:08] Started config cul on /dev/ttyUSB1 listening on t1
using fq 868.95M
(wmbusmeters) waiting for telegrams
[2021-12-13_22:45:10] (memory) rss 2912256 peak 2.77 MiB
(wmbusmeters) started meter 1 (MyELF2 21400916 auto)
telegram=|4E44010616094021420C7A2A0040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A28020A5E2102046D2A
36AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+6
telegram=|4E44010616094021420C7A2B0040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A28020A5E2102046D2A
36AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+22
telegram=|4E44010616094021420C7A320040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A28020A5E2102046D2E
36AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+135
telegram=|4E44010616094021420C7A330040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A28020A5E2102046D2E
36AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+151
telegram=|4E44010616094021420C7A350040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A28020A5E2102046D2E
36AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+184
telegram=|4E44010616094021420C7A360040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A28020A5E2102046D2E
36AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+200
telegram=|4E44010616094021420C7A370040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A28020A5E2102046D2E
36AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+216
telegram=|4E44010616094021420C7A380040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A28020A5E2102046D2E
36AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+232
telegram=|4E44010616094021420C7A390040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A28020A5E2102046D2E
36AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+248
telegram=|4E44010616094021420C7A3C0040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A28020A5E2102046D2E
36AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+296
telegram=|4E44010616094021420C7A3E0040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A28020A5E2102046D2E
36AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+328
telegram=|4E44010616094021420C7A440040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A28020A5E2102046D32
36AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+424
telegram=|4E44010616094021420C7A460040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A28020A5E2102046D32
36AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+456
telegram=|4E44010616094021420C7A490040252F2F|0C0E000000000C13000000004C0E00000000426CBE2B0B3B0000000B2D0000000A5A28020A5E2102046D32
36AD2C02FD1700002F2F2F2F2F2F2F2F2F2F2F|+504
***@***.***:/var/log/wmbusmeters/meter_readings$

Irrespective of the above in log I only see one decoded (perhaps latest):

***@***.***:/var/log/wmbusmeters/meter_readings$ more MyELF2
{"media":"heat volume at
inlet","meter":"elf","name":"MyELF2","id":"21400916","meter_date":"","total_energy_consumption_kwh":0,"cur
rent_power_consumption_kw":0,"total_volume_m3":0,"total_energy_consumption_at_date_kwh":0,"flow_temperature_c":22.8,"return_tempera
ture_c":22.1,"external_temperature_c":127,"status":"OK","operating_time_h":0,"version":"","battery_v":0,"timestamp":"2021-12-13T21:
55:21Z","device":"cul","rssi_dbm":-64}
***@***.***:/var/log/wmbusmeters/meter_readings$

Is there some knob to turn on to keep all decoded telegrams for a meter ?

Last it seems that im871a is out of stock for the last few months. So for
now I will just try with few new antenas for nanoCUL.

Many thx !

weetmuts, 2022-01-09:
>Is there some knob to turn on to keep all decoded telegrams for a meter ?

--logtelegrams

weetmuts, 2022-01-09:
Ah, sorry you meant logtelegrams but only for decoded ones. No, not right now. Should be easy to add though....

raszuk, 2022-01-09:
Actually I mean only for not decoded ones. 

See the situation today if I have no meter defined it is logging all telegrams and decoding the header. That is cool - but only if no meters is configured. 

The moment I add first meter log file does not contain any telegram headers .. it only contains raw telegrams. Maybe there is an easy way to pipe it in shell to decode telegram headers ? 

See when I am trying to set up the env I am receiving about 100 telegrams from different meters. Btw I am using RTLSDR now as nano cul with the very same antenna was a complete disaster. So I know about some meters and I am still discovering new hence the need to see both. 

Btw is there some way to setup all meter definitions in single file instead of 100 files (one in each file) ? 

Many thx !