wmbusmeters/wmbusmeters #410: Device Support requested - Relay WMBus Counter

vju42, 2021-12-13:
Hi, 
i am building my first MBUS connection between HomeAssitant and a GAS Meter.

I use a GAS meter that provides a simple counter signal, and use a counter device that creates the WMBUS Connection.
This device seems to be unsupported yet.

Please let me know what I can provide - I have access to telegrams and the product documentation.
I'm also ready to test or support the driver building process - just let me know.

Manufacturer: Relay
Website: https://www.relay.de/en/
Product name: Padpuls M2
Product page: https://www.relay.de/en/products/m-bus-slaves/padpuls-m2 
Product page contains also Manual 
The Counter supports two counters and allows to assign type and data conversions.

Here's the log output:
[2021-12-13_17:12:05] (meter) Gas: meter detection could not find driver for id: 00172801 mfct: (REL) Relay, Germany (0x48ac) type: Gas meter (0x03) ver: 0x50
[2021-12-13_17:12:05] (meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
[2021-12-13_17:12:05] (meter) to add support for this unknown mfct,media,version combination
(wmbusmeters) started meter 1 (Gas 00172801 auto)

Sample Telegrams from Log:
telegram=|2644AC482817000050377201281700AC48500387000000|2F2F0C01000000002F2F2F2F2F2F2F2F|+24
telegram=|2644AC482817000050377201281700AC48500388000000|2F2F0C01000000002F2F2F2F2F2F2F2F|+82
telegram=|2644AC482817000050377201281700AC48500389000000|2F2F0C01000000002F2F2F2F2F2F2F2F|+140

weetmuts, 2021-12-13:
Are you using PadPuls M2W ? Ie, you have a gas meter pulsing and m2w translates the pulses directly to wmbus?
(I.e. no plain m-bus involved here?)

When I look at the telegram it contains a single content field:
025   : 0C dif (8 digit BCD Instantaneous value)
026   : 01 vif (Energy 10⁻² Wh)
027 C!: 00000000 total energy consumption (0.000000 kWh)

Did you pick the translation factor to energy from the gas meter? I do not know much about
gas meters but I sort of would have assumed they reported a volume? 

We can easily write a driver for the PadPuls.

vju42, 2021-12-13:
Correct, its m2W not m2.
At the first catches telegram, I had not configured the energy yet.
Now I  configured the energy, but its supposed to be 10 L since they send one signal for 0.01 m^3 and I guess I have to translate to kWh myself.
telegram=|2644AC482817000050377201281700AC485003D9000000|2F2F0C14000000002F2F2F2F2F2F2F2F|+2355
I also succeeded reading the telegram this way (as newbie).

I would be interested in a driver and as said I can assist testing it.

I still have to go out and mount it - encryption did not work yet - they offer encryption levels 5 and 7.

weetmuts, 2021-12-13:
Yeah, volume sounds better, since that is what the meter is actually measuring?

Encryption should work both 5 and 7. Try 5 first.

Did you pick the Gasmeters version 0x50 ?
Apart from GasMeter which other meter types can you select?

vju42, 2021-12-13:
For completeness, here's the software for configuration:
https://www.relay.de/en/products/software/mbconf

It shows lots of options - I need to get the device again and connect it to get the list.
Its a longer list, 10-15 entries at least. Same for the units.

weetmuts, 2021-12-13:
Cool. It is very easy just to add a gasmeter with a volume field. However I need to think a bit on how  to deal with this kind of configurable relay which is in fact many meters in one.

In the meantime you can use the unismart driver. It will produce some additional fields, but they will be empty.
The total_m3 should be correct though.

vju42, 2021-12-13:
The driver seems to work.
Please let me know on case you want me to try all settings and see what types are generated. I now start to try get the json into HA.
Thanks

rikke-rp, 2022-05-30:
Hello
How is the progress on PadPuls M2W driver? I am interested in this driver aswell, and I can provide some telegrams if needed

jens62, 2023-06-16:
Hi,
I am using the relay PadPuls M2W for my gas meter as well.
As suggested I tried with the unismart driver.

This is a telegram I got: "2644AC487953000050377201795300AC4850038D0000202F2F0C16511450172F2F2F2F2F2F2F2F"

And the result is:

```
[SERIAL] select timeout 1 s
Auto driver  : not found!
Best driver  : apator08 04/08
Using driver : unismart 00/00
000   : 26 length (38 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : ac48 dll-mfct (REL)
004   : 79530000 dll-id (00005379)
008   : 50 dll-version
009   : 37 dll-type (Radio converter (meter side))
010   : 72 tpl-ci-field (EN 13757-3 Application Layer (long tplh))
011   : 01795300 tpl-id (00537901)
015   : ac48 tpl-mfct (REL)
017   : 50 tpl-version
018   : 03 tpl-type (Gas meter)
019   : 8d tpl-acc-field
020   : 00 tpl-sts-field (OK)
021   : 0020 tpl-cfg 2000 (synchronous )
023   : 2F skip
024   : 2F skip
025   : 0C dif (8 digit BCD Instantaneous value)
026   : 16 vif (Volume m³)
027 C?: 51145017
031   : 2F skip
032   : 2F skip
033   : 2F skip
034   : 2F skip
035   : 2F skip
036   : 2F skip
037   : 2F skip
038   : 2F skip

{
    "media":"gas",
    "meter":"unismart",
    "name":"",
    "id":"00537901",
    "timestamp":"2023-06-16T15:27:30Z"
}

```

Unfortunately the value for Volume is null.
But the correct value is in the bytes 027 - 030: 027 C?: 51145017.
When I look at my physical meter the value is 17501451.
I.e. the information is there but not correctly interpreted.

Using the driver suggested by the application ("apator08") I get a value, but different compared to what I read on my meter:
```

Using driver : apator08 04/08
000   : 26 length (38 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : ac48 dll-mfct (REL)
004   : 79530000 dll-id (00005379)
008   : 50 dll-version
009   : 37 dll-type (Radio converter (meter side))
010   : 72 tpl-ci-field (EN 13757-3 Application Layer (long tplh))
011   : 01795300 tpl-id (00537901)
015   : ac48 tpl-mfct (REL)
017   : 50 tpl-version
018   : 03 tpl-type (Gas meter)
019   : 94 tpl-acc-field
020   : 00 tpl-sts-field (OK)
021   : 0020 tpl-cfg 2000 (synchronous )
023   : 2F skip
024   : 2F skip
025   : 0C dif (8 digit BCD Instantaneous value)
025 C!: *** 10|2f2f0c16 total consumption (123299.087667 m3)
026   : 16 vif (Volume m³)
027 C?: 51145017
031   : 2F skip
032   : 2F skip
033   : 2F skip
034   : 2F skip
035   : 2F skip
036   : 2F skip
037   : 2F skip
038   : 2F skip

{
    "media":"gas",
    "meter":"apator08",
    "name":"",
    "id":"00537901",
    "total_m3":123299.087667,
    "timestamp":"2023-06-16T15:37:19Z"
}

```
Do you have any recommendation for me how to get the correct value "total_m3"?

weetmuts, 2023-06-16:
Ok, I have a solution for this thing. :-)  Drivers that can be written in a text file by the user. @vju42
I am working in this right now.

@jens62 in the meantime you can use the iperl driver to get the volume out.

jens62, 2023-06-16:
Thanks for the fast reply.
iperl works!

```
./wmbusmeters --analyze=iperl  2644AC487953000050377201795300AC4850038D0000202F2F0C16511450172F2F2F2F2F2F2F2F
Auto driver  : not found!
Best driver  : apator08 04/08
Using driver : iperl 00/00
000   : 26 length (38 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : ac48 dll-mfct (REL)
004   : 79530000 dll-id (00005379)
008   : 50 dll-version
009   : 37 dll-type (Radio converter (meter side))
010   : 72 tpl-ci-field (EN 13757-3 Application Layer (long tplh))
011   : 01795300 tpl-id (00537901)
015   : ac48 tpl-mfct (REL)
017   : 50 tpl-version
018   : 03 tpl-type (Gas meter)
019   : 8d tpl-acc-field
020   : 00 tpl-sts-field (OK)
021   : 0020 tpl-cfg 2000 (synchronous )
023   : 2F skip
024   : 2F skip
025   : 0C dif (8 digit BCD Instantaneous value)
026   : 16 vif (Volume m³)
027 C!: 51145017 ("total_m3":17501451)
031   : 2F skip
032   : 2F skip
033   : 2F skip
034   : 2F skip
035   : 2F skip
036   : 2F skip
037   : 2F skip
038   : 2F skip

{
    "media":"gas",
    "meter":"iperl",
    "name":"",
    "id":"00537901",
    "total_m3":17501451,
    "timestamp":"2023-06-16T20:20:27Z"
}
```
My issue is solved.

Nevertheless:
What about an option that adds the telegram to the output?
The user might do some gymnastics on the telegram on his own then.

closed 2023-11-15 by github-actions[bot]