wmbusmeters/wmbusmeters #411: new Diehl  IZAR RC 868 i W G4 meter not recognised as izar

atti-dall-atto, 2021-12-15:
A 2021 model UK water meter with a Diehl OMS Vol. 2 v3.0.1 IZAR RC 868 i W G4 radio module sends the following telegrams

`Started config rtlwmbus[00000008] listening on c1,t1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 10901234
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Water meter (0x07)
                   ver: 0x7c
                device: rtlwmbus[00000008]
                  rssi: 140 dBm
                driver: unknown!`

I assume the driver should be izar, so I create the following meter

`name=MyWaterMeter
driver=izar
id=10901234
key=NOKEY`

and see this in the logs

`Registering meters ...
Adding meter-0001 ...
Generating MQTT configuration ... 
Broker core-mosquitto will be used.
Running wmbusmeters ...
Started config rtlwmbus[00000008] listening on c1,t1
(meter) MyWaterMeter: meter detection did not match the selected driver izar! correct driver is: unknown!
(meter) Not printing this warning again for id: 10901234 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x7c
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) newly created meter (MyWaterMeter 10901234 izar) did not handle telegram!`

This is running as a VirtualBox image of Home Assistant (latest OS 7 Core 2021.12.1) with the ha-addon from this repo (0.2.19)

[image]

weetmuts, 2021-12-16:
Thanks can you post a telegram here? By using for example --logtelegrams

atti-dall-atto, 2021-12-16:
Sure thing - do I remove too much if I anonymise the meter id? Looks encrypted, sadly. I tried a simulation run with the OMS driver supercom587 and got nothing that decrypted to a field starting with 0x2f or ending with it if I used NOKEY.

```
telegram=||5E44A511xxxx90107C077A7D0050059CE4EB950F99A893B5ABA0168059FADA6A3AE7EAAE11F1C647D7481EDF7EC3C0333D74EE1963D344A9F1652A14FB6A0DDC60A8EB479D3853A05FF1CFF0892D9E44C70AA23F83EF655C33F954461AD1E8|
telegram=||2E44A511xxxx90107C077A7E0020251FF059A1D0AA4AA923A0EDC5A9EB22FFB07150893EA3060A4B9D92DB08F818A8|
telegram=||5E44A511xxxx90107C077A7E0050050FB805EDDBF5A96E15B09CB07C2C0ED7A6053F158C7597762A5A0B0AC058DF0217F00F9E1429D201A5268446AF9F0959D30E64AA46210652205D8AFAEFCE43E3644489EA4FE4FF6A11A0DB85F3D07623|
telegram=||2E44A511xxxx90107C077A7F00202523A6E8B0FB2403811E153810F4BC3A198A7659671D88453AD73A6A7D9B84B11F|
telegram=||2E44A511xxxx90107C077A80002025ED51470F0380E02C3855647069A7A22638CB7D7C76CC2E11CEEC0FF8BF7933C1|
telegram=||5E44A511xxxx90107C077A80005005DC7B83AD6B1CDB00C1AD37C66D822BABA5BE8125E7DC292C9032DF00FE25F32E33CB71BC453765725BBC838BC7B193F6BE8AE27AB356AE0C947D0AE94E2E5F2C0B48AE47541B7FDBA87584EFDDF602F0|
telegram=||2E44A511xxxx90107C077A81002025D87A0CD7131B1A9A7022BC2F2ECBB5AE5451F6BC6CFCCBC2C29431E835356040|
telegram=||5E44A511xxxx90107C077A81005005E911548CC1EE84765707B1D748E88F85890293C5B01817CC1A852EC31F744359F592B375D00B05CC616ED52E9625B5537F4AB0B786BC13CFB49EE0EAD9D6F9262C54CA7C5F2CF0A321AE509F4D1CAA29|
telegram=||5E44A511xxxx90107C077A82005005F20CFB27573E30904440834730E690BD013D9EA8A241CEEE20AA42C6C4BC04064A84CCC764D84EA9795FE505576D70AB169A0BFF56B001617D582D6F5160A247E89A2FE5B402CE54E495418B8345437E|
telegram=||5E44A511xxxx90107C077A8400500542089BD882E3446A99CAAA213B8B41EE90CE072352376F7CCDC30976D3CF5389310DB41DD9333891C013C2EB0BADC6A77D6C1E4DF864BC84C8A3AC850C4E48C5F26584C52357C72BB4D61A38EC3F4D5A|
telegram=||2E44A511xxxx90107C077A850020254D7A6819E69C52EEE83057930D785813BEC94C9A78EFE966B2865CA65D31773C|
telegram=||5E44A511xxxx90107C077A8500500541AFCF6110B801BA9CF588316E93FD128308A9D3B7263962F39C1C8B971AA230513C48B0C94AE8F2C187F183A55EE411C290D7C2D05EE292E139BA36E54F6D25702C82A6861E1C6C1804C3F5A8D39516|
telegram=||2E44A511xxxx90107C077A86002025E0CDBC60CF2296DA7862459B3D3C90A3FCAA6C7DE2B78468A923B43E89D6B3F7|
telegram=||5E44A511xxxx90107C077A870050057D5DA87B59FC479427C2676A3A86331C3727F4CE9605A792A5BB8DD19B408401130C690E6164CD8496CF962FD3E2F31D438996BB0EA21D0495BEBC9D9B23EFC705B0D7A9032854510B4994BEF3857B0D|
telegram=||5E44A511xxxx90107C077A8800500514E441901EA32ED4684ABF8CD82BC029F3F246F9464A935793EA0CD3E7FBD3AFB1493403E52888DA193F59B34B250C6CB83724F83262FAA6D08E1F68E15A32C2AEAF41184B7EA1D7FBD79ED3F685F40C|
telegram=||2E44A511xxxx90107C077A89002025B5FC9E48EDF28124167C4EBD086F61B6D57D68CE0ED15DB009424BFF39CBD8AF|
telegram=||5E44A511xxxx90107C077A8A00500517A2DAF99815B26BB632044DFBF9B454021775844103CC8D25F9B28212EF9454F61422CA6965CA1EDA37F1C4B06D3E6557DA3E9EB26AC18539C0DE3F4AA904F26B5FFD000142112D5EC3B49F7CEE0CA9|
telegram=||5E44A511xxxx90107C077A8B0050059AD60CD0BA5C239500A8F654C35693884BFF50DBDC8E1A89D6C9732B0CAB7AE6084D4201777BD231B68ED881740CE88A23E82F697FEE195AB844544160A90D15CB73AF54FF73CCDD6218AFD36761DEAE|
telegram=||5E44A511xxxx90107C077A8C0050058EDE5ED8EFB29EB392E835C639D49C79BCB5EC02D130A15E7CD1B2C39586D4A683A40D20A064A056E4AD64E1E6CC7EA73490E146B5029B7B0054655C0C113BEF6CA19BBE72DF64BFC2FFD77635D3802A|

```

atti-dall-atto, 2021-12-16:
```
(meter) MyMeter supercom587 handling telegram from 10905xxx
(meter) MyMeter 10905xxx "5E44A511xxxx90107C077A7D0050059CE4EB950F99A893B5ABA0168059FADA6A3AE7EAAE11F1C647D7481EDF7EC3C0333D74EE1963D344A9F1652A14FB6A0DDC60A8EB479D3853A05FF1CFF0892D9E44C70AA23F83EF655C33F954461AD1E8"
(wmbus) parseDLL @0 95
(telegram) DLL L=5e C=44 (from meter SND_NR) M=11a5 (DME) A=10905xxx VER=7c TYPE=07 (Water meter) (driver unknown!) DEV= RSSI=0
(wmbus) parseELL @10 85
(wmbus) parseNWL @10 85
(wmbus) parseAFL @10 85
(wmbus) parseTPL @10 85
(TPL) AES CBC IV decrypting "9CE4EB950F99A893B5ABA0168059FADA6A3AE7EAAE11F1C647D7481EDF7EC3C0333D74EE1963D344A9F1652A14FB6A0DDC60A8EB479D3853A05FF1CFF0892D9E44C70AA23F83EF655C33F954461AD1E8"
(TPL) num encrypted blocks 5 (80 bytes and remaining unencrypted 0 bytes)
(TPL) IV A511xxxx90107C077D7D7D7D7D7D7D7D
(TPL) decrypted  "24D093E35F31CB851286121A166226E1001BBDFAC2999ABD10D6FC95ECBDA3CEC967B50EDF6B3BE6AF4A1766F9AF295578768555EF123C049C19E009A2E15D9C2F3A84D592635932601158A0FD9BAE4D"
```

weetmuts, 2022-01-13:
Thanks! This look like a standard wmbus encryption, ie it is no longer the old izar driver.  This is expected, since the meter is marked OMS compliant.

Unfortunately it does not look like the key you have is correct. Otherwise decrypted should start with 2F2F.

weetmuts, 2022-01-14:
You can also email me the telegrams and the key you have.

weetmuts, 2022-03-28:
@atti-dall-atto Did you get a better key?

atti-dall-atto, 2022-03-28:
@weetmuts  no - I contacted the utility (eOn next) but they said their systems only had the meter ID which is printed on the bill. They have contracted out meter reading operations to another company. I concluded that _they_ could either

a) easily be using the same key for every meter and would be unwilling to divulge this fact, OR

b) they use the Diehl street reader kit as a black box too, and Diehl have a fixed algorithm of their own devising for generating the meter AES key from a hashing function on the known meter number .

The installer had an app on his phone that did a barcode scan of the label on the meter at the time of install. I doubt whether that matches to a lookup in a spreadsheet of truly random AES keys for the meter batch somehow - too much to go wrong if it is lost.

 I briefly considered trying to write a brute-forcer that would use a grammar of likely functions (SHA1, SH256,...) and operands (shift, extend, fill, swap) before concluding that I had gone too far down this particular rabbit hole.

weetmuts, 2022-05-25:
Yes, the keys cannot be brute-forced. Closing this for now. Good luck!

closed 2022-05-25 by weetmuts