wmbusmeters/wmbusmeters #412: Decoding issue with Ei6500 OMS

oreilly199, 2021-12-16:
Hi!

I own a Ei6500 OMS smoke detector and try do decode the payload:

`wmbusmeters --debug 5E442515636879000C1A7A1000502510DB35C6D554358FBD8FAE2A5F4E9A57284CFAFCE502A2A6F0EDE71BEB733261113E892337DF1699D266B880E48E33EE2FBAEAEA593E048310419E4DEDC0E7B48B693318EEC004BCA53936D2FC095A29 MyEI auto 00796863 0FF53ED30713CA17926CD24A1B8B5F1B`

But somehow the decryption failes:

`(wmbusmeters) version: 1.5.0-30-gd609d64
(config) using device: <5E442515636879000C1A7A1000502510DB35C6D554358FBD8FAE2A5F4E9A57284CFAFCE502A2A6F0EDE71BEB733261113E892337DF1699D266B880E48E33EE2FBAEAEA593E048310419E4DEDC0E7B48B693318EEC004BCA53936D2FC095A29>:any
(config) number of meters: 1
(lookup) with file/hex "5E442515636879000C1A7A1000502510DB35C6D554358FBD8FAE2A5F4E9A57284CFAFCE502A2A6F0EDE71BEB733261113E892337DF1699D266B880E48E33EE2FBAEAEA593E048310419E4DEDC0E7B48B693318EEC004BCA53936D2FC095A29"
(lookup) driver: simulation hex
(main) opening <5E442515636879000C1A7A1000502510DB35C6D554358FBD8FAE2A5F4E9A57284CFAFCE502A2A6F0EDE71BEB733261113E892337DF1699D266B880E48E33EE2FBAEAEA593E048310419E4DEDC0E7B48B693318EEC004BCA53936D2FC095A29>:any
Started config simulation listening on any
(simulation) in
(main) regular reset of simulation  will happen every 82800 seconds
(main) added  to files
(wmbus) no alarm (expected activity) for simulation
(lookup) with file/hex "5E442515636879000C1A7A1000502510DB35C6D554358FBD8FAE2A5F4E9A57284CFAFCE502A2A6F0EDE71BEB733261113E892337DF1699D266B880E48E33EE2FBAEAEA593E048310419E4DEDC0E7B48B693318EEC004BCA53936D2FC095A29"
(lookup) driver: simulation hex
(main) opening <5E442515636879000C1A7A1000502510DB35C6D554358FBD8FAE2A5F4E9A57284CFAFCE502A2A6F0EDE71BEB733261113E892337DF1699D266B880E48E33EE2FBAEAEA593E048310419E4DEDC0E7B48B693318EEC004BCA53936D2FC095A29>:any
Started config simulation listening on any
(simulation) in
(main) regular reset of simulation  will happen every 82800 seconds
(main) added  to files
(wmbus) no alarm (expected activity) for simulation
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(simulation) from file "5E442515636879000C1A7A1000502510DB35C6D554358FBD8FAE2A5F4E9A57284CFAFCE502A2A6F0EDE71BEB733261113E892337DF1699D266B880E48E33EE2FBAEAEA593E048310419E4DEDC0E7B48B693318EEC004BCA53936D2FC095A29"
(wmbus) ff a dll crc first (calculated 988c) did not match (expected 7a10) for bytes 0-10!
(meter) no meter handled  checking 1 templates.
(wmbus) parseDLL @0 95
(wmbus) parseELL @10 85
(wmbus) parseNWL @10 85
(wmbus) parseAFL @10 85
(wmbus) parseTPL @10 85
(meter) MyEI: for me? 00796863 in
(meter) MyEI: yes for me
(meter) created "MyEI" "ei6500" "00796863" encrypted
(meter) used meter template MyEI  auto to match 00796863
(meter) started meter 1 (MyEI 00796863 auto)
(wmbus) parseDLL @0 95
(wmbus) parseELL @10 85
(wmbus) parseNWL @10 85
(wmbus) parseAFL @10 85
(wmbus) parseTPL @10 85
(meter) MyEI: for me? 00796863 in 00796863
(meter) MyEI: yes for me
(meter) MyEI ei6500 handling telegram from 00796863
(meter) MyEI 00796863 "5E442515636879000C1A7A1000502510DB35C6D554358FBD8FAE2A5F4E9A57284CFAFCE502A2A6F0EDE71BEB733261113E892337DF1699D266B880E48E33EE2FBAEAEA593E048310419E4DEDC0E7B48B693318EEC004BCA53936D2FC095A29"
(wmbus) parseDLL @0 95
(telegram) DLL L=5e C=44 (from meter SND_NR) M=1525 (EIE) A=00796863 VER=0c TYPE=1a (Smoke detector) (driver ei6500) DEV= RSSI=0
(wmbus) parseELL @10 85
(wmbus) parseNWL @10 85
(wmbus) parseAFL @10 85
(wmbus) parseTPL @10 85
(TPL) AES CBC IV decrypting "10DB35C6D554358FBD8FAE2A5F4E9A57284CFAFCE502A2A6F0EDE71BEB733261113E892337DF1699D266B880E48E33EE2FBAEAEA593E048310419E4DEDC0E7B48B693318EEC004BCA53936D2FC095A29"
(TPL) num encrypted blocks 5 (80 bytes and remaining unencrypted 0 bytes)
(TPL) IV 2515636879000C1A1010101010101010
(TPL) decrypted  "4A0921CFFAA0E8BB9FBB2E80C3B837659BE54CCA26192B5E2724790672259236935B5295817E3DFAB395455CA592A5000931AC069D574648F33BDECADB3C0C1B0E2B69096CB47FF2B5BF67FECE4DBCAC"
(wmbus) WARNING! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 00796863 mfct: (EIE) Ei Electronics, Ireland (0x1525) type: Smoke detector (0x1a) ver: 0x0c
(telegram) TPL CI=7a ACC=10 STS=00 CFG=2550 (synchronous AES_CBC_IV nb=5 cntn=0 ra=0 hc=0)
(meter) newly created meter (MyEI 00796863 ei6500) did not handle telegram!
(wmbus) telegram from 00796863 ignored by all configured meters!
(serial) stopping manager
(simulation) from file "5E442515636879000C1A7A1000502510DB35C6D554358FBD8FAE2A5F4E9A57284CFAFCE502A2A6F0EDE71BEB733261113E892337DF1699D266B880E48E33EE2FBAEAEA593E048310419E4DEDC0E7B48B693318EEC004BCA53936D2FC095A29"
(wmbus) ff a dll crc first (calculated 988c) did not match (expected 7a10) for bytes 0-10!
(wmbus) skipping already handled telegram.
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) event loop stopped!
(serial) TIMER thread interrupted
(wmbus) deleted simulation
(wmbus) deleted simulation`

I double-checked the decryption key for Mode5 AES decryption. But i believe there is an error on an earlier point: 

`(wmbus) ff a dll crc first (calculated 988c) did not match (expected 7a10) for bytes 0-10!`

Could you help me to investigate the problem?

Best regards

weetmuts, 2021-12-16:
I think you just have the wrong key (or you have the key bytes in the wrong order instead of 112233...  try ...332211)
or you have the right key but listening to the wrong meter.

The log line "(TPL) decrypted "4A0921...." is wrong since the telegram should log to "(TPL) decrypted "2F2F..."
where 2f2f proves that it was properly decrypted.

The final crc warning is a read herring, ignore that, it should go away when I fix the code in the future. It is a side
effect of the ability of wmbusmeters to strip dll crc:s from hex telegrams on the command line.

oreilly199, 2021-12-16:
Curius... I got the key this way from the manufacturers package, where serial number and key are clearly listet together. I will try to manually "reverse" the key and then write again.

oreilly199, 2021-12-20:
You were right. The vendor gave me a wrong AES-key. I contacted eielectronics and they helped me out.

Closed!

closed 2021-12-20 by oreilly199