wmbusmeters/wmbusmeters #413: Nooelec NESDR SMArt v4 RTL-SDR + CUL_V3 : Zenner Minomess no telegrams at all

DannyBoyKN, 2021-12-19:
Our water meters were successively exchanged by new once over the last years and I recently learned that they're capable to be read out over-the-air. Many of our meters are Zenner Minomess (I believe same as [here](https://github.com/weetmuts/wmbusmeters/issues/384#issuecomment-981352567)) as being already used by others here as to mention some examples #384, #390...

I was curious what type of data they would/can transmit. I have already several of RaspberryPis so I bought also the [Nooelec NESDR SMArt v4 RTL-SDR](https://smile.amazon.de/gp/product/B01GDN1T4S/ref=ppx_yo_dt_b_asin_title_o00_s01?ie=UTF8&psc=1) and trying to follow the explanations given in the german magazine[ c't "Energiekosten: Energiekostenverteiler selbst auslesen"](https://www.heise.de/select/ct/2021/25/2028616030220371058).
I'm no expert on radio transmissions at all, but I'm willing to learn :smile:

My first tries failed :disappointed: wmbusmeters - executed inside a docker container - detected the RTL SDR device and was waiting for telegrams to be received. The config was this:
```
loglevel=normal
device=auto
listento=c1,t1,s1
logtelegrams=true
format=json
meterfiles=/wmbusmeters_data/logs/meter_readings
meterfilesaction=overwrite
logfile=/wmbusmeters_data/logs/wmbusmeters.log
```
but no telegrams were received over several nights :disappointed:

```
./wmbusmeters --verbose --debug auto:c1
(wmbusmeters) version: 1.5.0-30-gd609d64
(config) number of meters: 0
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) check if /dev/ttyAMA0 can be accessed
(serial) tty /dev/ttyAMA0 can be accessed
(serial) EVENT thread interrupted
(serialtty) device /dev/ttyAMA0 is already in use and locked.
(amb8465) could not open tty /dev/ttyAMA0 for detection
(serial) EVENT thread interrupted
(serialtty) device /dev/ttyAMA0 is already in use and locked.
(im871a) could not open tty /dev/ttyAMA0 for detection
(serial) EVENT thread interrupted
(serialtty) device /dev/ttyAMA0 is already in use and locked.
(serial) EVENT thread interrupted
(serialtty) device /dev/ttyAMA0 is already in use and locked.
(main) ignoring /dev/ttyAMA0, it does not respond as any of the supported wmbus devices.
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	filename (a '-' dumps samples to stdout)

<<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 3497 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(serial) EVENT thread interrupted
(shell) output: >>><<<
(shell) return code 0
(shell) 3499 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 77771111153705700 not currently used.
(main) opening rtlwmbus
Started auto rtlwmbus[77771111153705700] listening on c1
(rtlwmbus) using command: rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | rtl_wmbus -s
(rtlwmbus) opening 77771111153705700
(serial) EVENT thread interrupted
[... 97 lines cut ...]
(serial) EVENT thread interrupted
(serial) received binary "434D4F44450D0A"
(cul) checkCULFrame "CMODE<0D><0A>"
(cul) no leading 'b' so it is text and no frame
loglevel=debug
(cul) received "CMODE"(serial /dev/ttyACM0) sent "5832310A0D"
(wmbus) no alarm (expected activity) for cul
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted

```

However, after one day still no telegrams :worried:

I'm stuck now !
What could be the Problem?
Are my antennas to "small" ? Lambda/4 for CUL and the standard antennas for the RTL-SDR ?
Is the Zenner Minomess telegram transmission actually enabled ? How to know ?

It's a residential building with several devices transmitting over 868MHz.
I should be receiving something, no ?

I would appreciate any help :grinning:

chroeseler, 2021-12-20:
Hey, I was just testing the new driver for the Zenner Minomess. Works just fine for me. Can you see any other wmbus devices? In my place all the heaters have wmbus devices that pollute my output. The Zenner meter sends roughly every 10 min.

> Is the Zenner Minomess telegram transmission actually enabled ? How to know ?

We activated with the flashlight method. The manual reads:
Activation is carried out automatically as soon as the function 'pulse enabled' is switched on and the radio module has detected >30L from the counter. Alternatively, the radio interface can be activated after removel of the protective film on the IR interface with a flashlight (>8s)

Added my own debug output, maybe you can get something out of it. We use a 
[im871a-USB](https://shop.imst.de/wireless-modules/usb-radio-products/10/im871a-usb-wireless-m-bus-usb-adapter-868-mhz) dongle
```
(wmbusmeters) version: 1.5.0-30-gd609d64
(config) number of meters: 0
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary ""
(amb8465) reading response... 2
(serial) received binary ""
(amb8465) reading response... 3
(serial) received binary ""
(amb8465) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A581100833001301023800000D3E"
(im871a) checkIM871AFrame "A581100833001301023800000D3E"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 3e0d expected 3e0d
(im871a) received full frame
(device info bytes) "3300130102380000"
(im871a/im170a) info: im871a other firmware=13 hci=01 uid=00003802
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000600B32542291000010001FF0702320001000000C1C1DB"
(im871a) checkIM871AFrame "A5810617FF000600B32542291000010001FF0702320001000000C1C1DB"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 dbc1 expected dbc1
(im871a) received full frame
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(device config bytes) "FF000600B32542291000010001FF0702320001000000C1"
(im871a/im170a) config: other link_mode=c1a id=00102942 media=00 version=01 c_field=00 auto_rssi=01
(im871a/im170a) are you there? yes 00102942 im871a firmware: 13
(main) opening c1
Started auto im871a[00102942] on /dev/ttyUSB0 listening on c1
(im871a) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (im871a)
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
[... 101 lines cut ...]
(wmbus) 002   : 9344 dll-mfct (QDS)
(wmbus) 004   : 68603313 dll-id (13336068)
(wmbus) 008   : 17 dll-version
(wmbus) 009   : 06 dll-type (Warm Water (30°C-90°C) meter)
(wmbus) 010   : 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
(wmbus) 011   : 0D dif (variable length Instantaneous value)
(wmbus) 012   : FF vif (Vendor extension)
(wmbus) 013   : 5F vife (duration of limit exceed last upper  is 3)
(wmbus) 014   : 35 varlen=53
(wmbus) 015 C?: 0082C600007E0007C113FFFF488306009F2C73370300BE2B976606005801730176014C0142011301F200F1005F01FA00CC004A012F
(wmbus) 068   : 04 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 069   : 6D vif (Date and time type)
(wmbus) 070 C?: 1011B42C
telegram=||49449344686033131706780DFF5F350082C600007E0007C113FFFF488306009F2C73370300BE2B976606005801730176014C0142011301F200F1005F01FA00CC004A012F046D1011B42C|+2
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(wmbus) disconnected /dev/ttyUSB0 im871a
(serialtty) closed /dev/ttyUSB0 (im871a)
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!
(wmbus) deleted im871a
(serial) closing 1 devices
```

DannyBoyKN, 2021-12-20:
Well, no I don't get anything ... nothing, nada, njente 😞 

It's this one ?
[image]

Because at Zenner online I found this:
[image]

!!! There's a difference "LoRa  868MHz" vs. "wM-Bus C1 Mode" !!!

I found additional info material, unfortunately only in german:

> **Aktivierung Funkschnittstelle:**
>
>wM-Bus-Ausführung:
>Die Aktivierung des Funks und der Spulenabtastung des Moduls erfolgt mittels
>der ZENNER Software MSS, dem Universalinterface MinoConnect (USB oder Bluetooth) und dem >speziellen ZENNER Infrarot Optokopf IrCombiHead. Bitte aktivieren Sie dazu in der Software MSS >das Feld „SetOpe- ratingMode“. Alternativ kann die Schnittstelle auch nach Entfernung der >Schutzfolie an der IR-Schnittstelle **mit einer Lichtquelle (> 8s) aktiviert werden (als Leuchtmittel >bitte Standard-Glühbirne verwenden, keine LED-Leuchte).** Nach erfolgter Aktivierung sendet >das Modul für einen Zeitraum von einer Stunde mit einem erhöhten Sendeintervall von 20 s (Inbetriebnahmeszenario).
>
>LoRaWAN® (OTAA):
>Die Aktivierung des Funks und der Spulenabtastung des Moduls erfolgt mittels der ZENNER >Software MSS, dem Universalinterface MinoConnect (USB oder Bluetooth) und dem speziellen >ZENNER Infrarot Optokopf IrCombiHead. Alternativ kann die Schnittstelle auch nach Entfernung >der Schutzfolie an der IR-Schnittstelle mit **einer Lichtquelle (> 8s) aktiviert werden (als >Leuchtmittel bitte Standard-Glühbirne verwenden, keine LED-Leuchte).**
>Das Gerät sendet eine Beitrittsanfrage (join request) an den LNS und wartet auf die Annahme (join accept). Falls keine Verbindung erfolgt, wird jede Minute eine weitere Anfrage versendet (max. >5 Anfragen).
>LED zeigt Status des Verbindungsvorgangs:
>schnell blinkend: noch keine Antwort erhalten
>langsam blinkend: erfolgreiche Verbindung.
>Bei erfolglosem Beitritt, sendet das Gerät jeden weiteren Tag eine zufällige Beitrittsanfrage bis eine erfolgreiche Verbindung erfolgt ist (LED zeigt spätere Verbindungsversuche nicht an).

In my case I suppose the over-the-air activation is sending join requests and expects join accepts. There should be a LED blinking (fast=no accept, slow=accept success). I tried to illuminate but I don't see any LED, or the accept failed, so join request is resend each 2nd day until accept success, but no LED is blinking at all.

I tried this same time as executing wmbusmeters ... no telegrams... no reaction ... nada

BIBOLV, 2021-12-20:
If the first one is actual picture of your meter then it is LoRa and it is not the same as Wireless MBUS, so you will not be able to read it with wmbusmeters. Maybe you can still use RTL-SDR to capture LoRa traffic, but you will need to research that. I only know that years ago someone tried to do it. Long time ago I played around with rakwireless LoRa gateway as it was cheapest at the time and chirpstack server.

DannyBoyKN, 2021-12-21:
I thought LoRa was for the physical / transport layer, wM-Bus for protocol / application layer 😞
However, I'm not sure if the picture from Zenner with the wM-Bus C1 Mode is just a marketing gag or it really does exist with that inscription ... maybe someone can confirm / post a 'real' picture.
 
So, yes, I'll go for some researches using the RTL-SDR USB dongle...

weetmuts, 2021-12-21:
I am pretty sure that it is possible to send mbus application data (dif vif values) inside a lora telegram. Maybe the meter does that, but if we cant get the lora telegram content, we cant decode it anyway.

weetmuts, 2021-12-21:
Many meters have plugin modules that can be inserted, to use either wmbus, lora, gsm (soon 5g low bitrate as well i suppose) or some other radio type. So i am not at all surprised that this water meter can be bought in two different configurations.

chroeseler, 2021-12-22:
> I thought LoRa was for the physical / transport layer, wM-Bus for protocol / application layer 😞
> However, I'm not sure if the picture from Zenner with the wM-Bus C1 Mode is just a marketing gag or [...]

Though you already managed to find the differences. Just for the sake of the question here is ours. [image]

trsqr, 2021-12-22:
Yep, the LoRaWAN device is not compatible with wMBus as you found out. I've got the Zenner Minomess that looks exactly like the Zenner marketing picture (wM-Bus C1 Mode). The radio on the thing had to be activated by shining some light on it for several seconds. And a LED bulb just did not work, I had to dig out an old flashlight with halogen bulb to get the radio activated.

BIBOLV, 2023-01-01:
Closing due to inactivity.

closed 2023-01-01 by BIBOLV