wmbusmeters/wmbusmeters #414: Dongle not working until replug after home assistant restart

greinet, 2021-12-21:
Hallo,

I am using the wmbusmeter addon for home assistant and am quite happy with it, but every time the addon is restarted (e.g. when home assistant core or the server itself is restarted) the dongle stops working and only works again after unplugging and then plugging it in again.

**Configuration**

```
data_path: /config/wmbusmeters
conf: |-
  loglevel=debug
  device=rtlwmbus:t1
  logtelegrams=true
  format=json
  logfile=/dev/stdout
  shell=/wmbusmeters/mosquitto_pub.sh wmbusmeters/$METER_NAME "$METER_JSON"
meters:
  - |-
    name=MainWater
    id=*******
    driver=izar
mqtt:
  *****
```

**Output**
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
Syncing wmbusmeters configuration ...
Registering meters ...
Adding meter-0001 ...
Generating MQTT configuration ... 
Broker 192.168.178.34 will be used.
Running wmbusmeters ...
(config) "device" "rtlwmbus:t1"
(wmbusmeters) version: 1.5.0-8-g67e0072
(config) using device: rtlwmbus:t1 
(config) number of meters: 1
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers
Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	filename (a '-' dumps samples to stdout)
<<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 269 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>><<<
(shell) return code 0
(serial) EVENT thread interrupted
(shell) 270 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(main) found specified device (rtlwmbus:t1) that matches detected device (:rtlwmbus[00000001]:0/0)
(main) opening rtlwmbus:t1
Started config rtlwmbus[00000001] listening on t1
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 00000001
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serialcmd) opened /bin/sh pid 275 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 21 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
```

After replugging it works as expected again. I already tried the resetafter configuration option but it didn't solve the problem.

weetmuts, 2022-01-11:
I am not an expert on HASS, @to-masz , do you have any input?
The dongle is taken by rtlsdr process and maybe restarting HA does not properly 
shut down the rtlsdr (or wait long enough) before trying to start wmbusmeters again?

Is the wmbusmeters started as a daemon or as a command line. I know for sure the daemon
will start the rtlsdr/rtlwmbus in their own process group which will be killed, if wmbusmeters dies.
But within HA, I do not know?

to-masz, 2022-01-11:
@greinet we need some more details
1. Is that also a case when you will hit "restart" button on the add-on page? 
2. Is that also a case when you will hit "stop" button on the add-on page and then wait (let's say 60 seconds) and "start" it again? 
3. What is your hardware you use to run hass os? 

@wajo666 @philips1923 I remember from recent issues that you have rtl dongle. Do you have similar problem? 

@weetmuts homeassistant stops the docker container and runs the new one in that case... and as you can find in `/ha-addon/run.sh` wmbusmeters is started as a command line.

wajo666, 2022-01-11:
Hi all, yes I have had HA in docker wmbusmeter in docker z2m in docker too :) I have had only one small problem when wmbusmeter is started z2m docker has to restart again because in wmbusmeter has setup for all /dev/ and z2m sticker is mounted in dev. When I restarting HA in docker or in HA internal wothout problem woth wmbusmeter

philips1923, 2022-01-11:
> @wajo666 @philips1923 I remember from recent issues that you have rtl dongle. Do you have similar problem?

Yep, I have the same issue (HA on proxmox). When restarting only HA instance everything is OK. When my terminal has power-off electricity and restarts whole proxmox with HA I have to unplug and plug again RTL stick. Stick as on image, bought directly from China)
[image]

greinet, 2022-01-20:
Sorry for the late response, a couple of days after the problem occured I switched from running HassOS to home assistant as docker container and also ran the standalone wmbusmeter docker container. With this setup everything works fine and i don't need to replug the dongle anymore.

> 1. Is that also a case when you will hit "restart" button on the add-on page?
> 2. Is that also a case when you will hit "stop" button on the add-on page and then wait (let's say 60 seconds) and "start" it again?
> 3. What is your hardware you use to run hass os?
1. Yes.
2. I tried stopping and starting and it didn't solve the problem, but I never really waited longer than 10 seconds at max.
3. I ran it on a Raspberry Pi 4 4GB. 

Sadly I don't have any more hardware to do further testing on, so I won't really be able to contribute any more information on this issue.

Btw, from what I can tell by the picture I have the same exact dongle like @philips1923.

to-masz, 2022-01-27:
@weetmuts 
> The dongle is taken by rtlsdr process and maybe restarting HA does not properly
shut down the rtlsdr (or wait long enough) before trying to start wmbusmeters again?

It looks like you are right. How can we enforce rtlsdr to shutdown and free the dongle? We can either add it only to addon or handle the SIGTERM / SIGKILL in wmbusmeters

weetmuts, 2022-03-29:
When rtlsdr/rtlwmbus are started by wmbusmeters it creates its own process group with a shell that starts rtlsdr and rtlwmbus. To stop them it sends sigterm to this group. (It tooke quite a time to get this working properly for plain linux, but now it seems to work.) Assuming that wmbusmeters gets a nice SIGTERM (and not a SIGKILL) it should terminate this process group nicely, and the programs should stop. 

When it is hung. Can you run:
lsof | grep rtl

You will see many lines, but this line indicates that the process 126588 is holding the usb device.

```
rtl_sdr   126588                           fredrik   10u      CHR             189,19       0t0        615 /dev/bus/usb/001/020
```

in different shells, like inside the docker, outside the docker etc.
Check the pids that you see if they still exists using 
ps axcf
```
 126573 pts/2    Sl+    0:00  |       \_ wmbusmeters
 126587 pts/2    S      0:00  |           \_ sh
 126588 pts/2    Sl     0:01  |               \_ rtl_sdr
 126589 pts/2    S      0:41  |               \_ rtl_wmbus
```

Then try kill 126588, kill -9 126588
in different shells until it releases the dongle.

closed 2023-04-06 by github-actions[bot]

weetmuts, 2023-08-18:
@to-masz Finally! I have found the explanation to rtl_sdr not exiting properly!

In my default setup I pipe the stderr of rtl_sdr to /dev/null and this causes rtl_sdr to exit nicely.
This was done by a long painful trial and error, I never knew why this was necessary.

However the reason was an infinite signal loop when the writing to stderr at exit fails because stderr 
is closed at shutdown and they have fixed it in rtl_sdr now!

https://github.com/osmocom/rtl-sdr/commit/142325a93c6ad70f851f43434acfdf75e12dfe03