wmbusmeters/wmbusmeters #415: Support for Maddalena microCLIMA heat meter

MightyPork, 2021-12-22:
https://www.maddalena.it/en/products/thermal-energy/microclima-u/95

payloads received using a rtl-sdr stick, they are from different meters as I'm testing this in a large apartment building fitted with these.

```
494424343124579300047a5a0000202f2f046d2720b62c04060d07000001fd170004130a8c0400043b00000000042b00000000025b1500025f15000261d0ff03fd0c05000002fd0b1011

A44424348630579300047ADD0000202F2F046D0721B62C04064708000004135DB0030001FD1700426C9F2C4406C6040000C40106C0070000C4020637070000C4030611070000C404060B070000C405060B070000C406060B070000C407060B070000C40806A5060000C40906F7050000C40A067A050000C40B060F050000C40C06C6040000C40D063F040000C40E06BB030000C40F06A502000003FD0C05000002FD0B1111

494424343124579300047A600000202F2F046D0821B62C04060D07000001FD170004130A8C0400043B00000000042B00000000025B1500025F15000261D7FF03FD0C05000002FD0B1011

A44424346030579300047A420000202F2F046D0921B62C0406EF1B00000413CBD70D0001FD1740426C9F2C4406A30D0000C40106791A0000C40206A8180000C40306F5170000C40406F5170000C40506F5170000C40606F5170000C40706F4170000C408068B170000C4090622160000C40A06CF130000C40B0622110000C40C06A30D0000C40D068D0A0000C40E062A080000C40F068B06000003FD0C05000002FD0B1111

A44424348230579300047AE80000202F2F046D0B21B62C0406960D00000413F4F8040001FD1700426C9F2C440643080000C40106780D0000C40206760D0000C40306750D0000C40406750D0000C40506750D0000C40606750D0000C40706750D0000C408062C0D0000C40906560C0000C40A06760B0000C40B06EC090000C40C0643080000C40D0623060000C40E0601050000C40F068204000003FD0C05000002FD0B1111
```

I'm not good with C++, but I might try my hand at implementing this, it should be somewhat similar to evo868.

weetmuts, 2022-01-13:
Thanks I pushed support for microclima, you can try this:
/build/wmbusmeters --format=json 494424343124579300047a5a0000202f2f046d2720b62c04060d07000001fd170004130a8c0400043b00000000042b00000000025b1500025f15000261d0ff03fd0c05000002fd0b1011 Heato microclima ANYID NOKEY
{"media":"heat","meter":"microclima","name":"Heato","id":"93572431","total_energy_consumption_kwh":1805,"total_energy_consumption_tariff1_kwh":0,"total_volume_m3":297.994,"total_volume_tariff2_m3":0,"volume_flow_m3h":0,"power_kw":0,"flow_temperature_c":21,"return_temperature_c":21,"temperature_difference_c":654.88,"status":"OK","device_date_time":"2021-12-22 00:39","timestamp":"2022-01-13T11:00:19Z"}

Ehum the temp difference is out of spec. This is because it is reported as 10^-2 Kelvin in the telegram.
Have to see why the unit conversion fails here....

MightyPork, 2022-01-13:
Hmm, isn't it possible you have wrong byte order for the temp difference field?

```
(auto) 059   : 02 dif (16 Bit Integer/Binary Instantaneous value)
(auto) 060   : 61 vif (Temperature difference 10⁻² K)
(auto) 061 C?: D0FF
```

FFD0 would be 0.47 K

weetmuts, 2022-01-13:
Ehum, is it -48 or -47....

weetmuts, 2022-01-13:
> Maybe you forgot to commit a part? I can decode it with the "sharky" driver now, but the "microclima" driver shows that it isn't defined.

I have pushed it. Please do a "make clean" to rebuild everything.

MightyPork, 2022-01-13:
You're, negative, I did the conversion on a calculator and forgot the sing. Don't see the new commit yet

weetmuts, 2022-01-13:
The lasest commit on github is 14f59384014d26773088487945a7f4d791a79a80 
this includes commit e3346133f827e94e4fe2d89d42ac7b972c629d2a   Added microclima heat meter.

weetmuts, 2022-01-13:
I think it should be -0.48 K

MightyPork, 2022-01-13:
Oh, I thought you added some new commit that corrected the parsing.

-0.48 K sounds right, yes. I found something very confusing about negative values in BCD fields here: https://m-bus.com/assets/downloads/MBDOC48.PDF page 61. 

If I read it right, a leading "F" can (but does not have to?) mean the rest of the number is to be interpreted as signed, in hex form.

Maybe the meaning is specific to a particular manufacturer's implementation. What a messy encoding, why couldn't they just use 2's complement everywhere? ...

weetmuts, 2022-01-13:
Thanks for the link! That looks like an older document and the part you are referring to is titled "New proposal".
And it is about how to use illegal BCD numbers that start with E or F to signal overflow/underflow and error.
Perhaps this is also the case in the current standard, I will have to look for that.

In this case it is more simple. This field is hexadecimal field using two complement form to represent a negative value.

I have not seen that many Temperature difference fields so far, sharky has one and microclima has one.
Perhaps the temp difference is always signed, sort of would make sense. Right now however, I added support for VifScaling::AutoSigned when extracting the field (check meter_microclima.cc).

Also for temperature differences K and C is the same thing! So now it reports the diff as _c instead.
Fix pushed please test.

weetmuts, 2022-01-13:
Now I have pushed the new meter_microclima.cc as well! Sorry about that!

weetmuts, 2022-01-13:
Interesting, that longer telegram contains historical data, but not the temperature difference. (But it does contain the total energy consumption and total volume. )

So the temp diff is initialized to zero and stored in a variable with unit Kelvin, so yeah, 0 becomes -273.15 Celsius, quite correctly. :-)

weetmuts, 2022-01-13:
Try: build/wmbusmeters --analyze A44424348630579300047ADD0000202F2F046D0721B62C04064708000004135DB0030001FD1700426C9F2C4406C6040000C40106C0070000C4020637070000C4030611070000C404060B070000C405060B070000C406060B070000C407060B070000C40806A5060000C40906F7050000C40A067A050000C40B060F050000C40C06C6040000C40D063F040000C40E06BB030000C40F06A502000003FD0C05000002FD0B1111 Any auto ANYID NOKEY

MightyPork, 2022-01-16:
I have been testing the new version for a while and haven't seen any problems, just once where it stopped receiving and I had to re-connect the rtl-sdr dongle. That could be a hardware glitch of some kind.

Just one thing that could be better, if the JSON didn't include fields that were not in the frame, like the zero temperatures? Although I think I will just ignore them, if short frames are so rare... I never saw the short frame again.

The meter in my apartment has UNKNOWN_ERROR_FLAGS(0x40), but still seems to measure fine, the kWh number grows.

Here are some new samples:

```
telegram=|A44424347830579300047A19000020#2F2F046D1533D0210406810B00000413DCE3060001FD1700426CBF2C44062E0B0000C401062E0B0000C40206F1090000C4030685080000C40406EC070000C40506EC070000C40606EC070000C40706EC070000C40806EC070000C40906EC070000C40A06EC070000C40B0655070000C40C06DC060000C40D064F060000C40E06C2050000C40F065905000003FD0C05000002FD0B1111|+90
{"media":"heat","meter":"microclima","name":"Name","id":"93573078","total_energy_consumption_kwh":2945,"total_energy_consumption_tariff1_kwh":0,"total_volume_m3":451.548,"total_volume_tariff2_m3":0,"volume_flow_m3h":0,"power_kw":0,"flow_temperature_c":0,"return_temperature_c":0,"temperature_difference_c":-273.15,"status":"OK","device_date_time":"2022-01-16 19:21","timestamp":"2022-01-16T18:21:34Z","device":"rtlwmbus[77771111153705700]","rssi_dbm":16}
telegram=|A44424342724579300047A7C000020#2F2F046D1933D0210406B6100000041347C5070001FD1740426CBF2C4406930F0000C40106930F0000C40206E80D0000C40306940C0000C40406580C0000C40506580C0000C40606580C0000C40706580C0000C40806580C0000C40906440C0000C40A06C60B0000C40B06DA0A0000C40C06C0090000C40D06F7070000C40E06D9060000C40F062A06000003FD0C05000002FD0B1111|+11
{"media":"heat","meter":"microclima","name":"Name","id":"93572427","total_energy_consumption_kwh":4278,"total_energy_consumption_tariff1_kwh":0,"total_volume_m3":509.255,"total_volume_tariff2_m3":0,"volume_flow_m3h":0,"power_kw":0,"flow_temperature_c":0,"return_temperature_c":0,"temperature_difference_c":-273.15,"status":"UNKNOWN_ERROR_FLAGS(0x40)","device_date_time":"2022-01-16 19:25","timestamp":"2022-01-16T18:24:27Z","device":"rtlwmbus[77771111153705700]","rssi_dbm":114}
telegram=|A44424349530579300047AD2000020#2F2F046D1733D02104068A1300000413F6420B0001FD1700426CBF2C440691120000C4010691120000C4020612100000C40306990E0000C40406440E0000C40506440E0000C40606440E0000C40706440E0000C40806410E0000C40906C90D0000C40A061B0D0000C40B06040C0000C40C06BF0A0000C40D0666090000C40E06A1070000C40F065C06000003FD0C05000002FD0B1111|+23
{"media":"heat","meter":"microclima","name":"Name","id":"93573095","total_energy_consumption_kwh":5002,"total_energy_consumption_tariff1_kwh":0,"total_volume_m3":738.038,"total_volume_tariff2_m3":0,"volume_flow_m3h":0,"power_kw":0,"flow_temperature_c":0,"return_temperature_c":0,"temperature_difference_c":-273.15,"status":"OK","device_date_time":"2022-01-16 19:23","timestamp":"2022-01-16T18:24:39Z","device":"rtlwmbus[77771111153705700]","rssi_dbm":19}
telegram=|A44424342024579300047A94000020#2F2F046D2033D02104067A03000004133E78030001FD1740426CBF2C44067A030000C401067A030000C402066F030000C403066D030000C404066D030000C405066D030000C406066D030000C407066D030000C408066D030000C409066D030000C40A066D030000C40B0664030000C40C065E030000C40D064D030000C40E0645030000C40F063B03000003FD0C05000002FD0B1111|+503
{"media":"heat","meter":"microclima","name":"Name","id":"93572420","total_energy_consumption_kwh":890,"total_energy_consumption_tariff1_kwh":0,"total_volume_m3":227.39,"total_volume_tariff2_m3":0,"volume_flow_m3h":0,"power_kw":0,"flow_temperature_c":0,"return_temperature_c":0,"temperature_difference_c":-273.15,"status":"UNKNOWN_ERROR_FLAGS(0x40)","device_date_time":"2022-01-16 19:32","timestamp":"2022-01-16T18:32:39Z","device":"rtlwmbus[77771111153705700]","rssi_dbm":19}
```

closed 2022-01-16 by MightyPork