wmbusmeters/wmbusmeters #416: Support for LSE Water Meter

jczmok-iag, 2021-12-22:
Hello there, 
thanks for all your efforts! I started using the wmbusmeters on my Raspberry to get more data in my apartment,
however i have some meters not supported.

I got:

Received telegram from: 14133936
          manufacturer: (LSE) Unknown (0xb265)
                  type: Water meter (0x07)
                   ver: 0x41
                device: rtlwmbus[00000001]
                  rssi: 108 dBm
                driver: unknown!
telegram=||324465B23639131441077AA5000000046D1412B62C01FD0C05326CFFFF0DFF5F0C0008313000010613080BFFFC0C7887292459|+0

I would assume this is Landis Staefa but has a different coding ?!?

weetmuts, 2022-01-13:
Yeah, this was an odd telegram running:
wmbusmeters --debug 324465B23639131441077AA5000000046D1412B62C01FD0C05326CFFFF0DFF5F0C0008313000010613080BFFFC0C7887292459

Prints:
(wmbus) 000   : 32 length (50 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : 65b2 dll-mfct (LSE)
(wmbus) 004   : 36391314 dll-id (14133936)
(wmbus) 008   : 41 dll-version
(wmbus) 009   : 07 dll-type (Water meter)
(wmbus) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 011   : a5 tpl-acc-field
(wmbus) 012   : 00 tpl-sts-field (OK)
(wmbus) 013   : 0000 tpl-cfg 0000 ( )
(wmbus) 015   : 04 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 016   : 6D vif (Date and time type)
(wmbus) 017 C?: 1412B62C
(wmbus) 021   : 01 dif (8 Bit Integer/Binary Instantaneous value)
(wmbus) 022   : FD vif (Second extension FD of VIF-codes)
(wmbus) 023   : 0C vife (Model/Version)
(wmbus) 024 C?: 05
(wmbus) 025   : 32 dif (16 Bit Integer/Binary Value during error state)
(wmbus) 026   : 6C vif (Date type G)
(wmbus) 027 C?: FFFF
(wmbus) 029   : 0D dif (variable length Instantaneous value)
(wmbus) 030   : FF vif (Vendor extension)
(wmbus) 031   : 5F vife (duration of limit exceed last upper  is 3)
(wmbus) 032   : 0C varlen=12
(wmbus) 033 C?: 0008313000010613080BFFFC
(wmbus) 045   : 0C dif (8 digit BCD Instantaneous value)
(wmbus) 046   : 78 vif (Fabrication no)
(wmbus) 047 C?: 87292459

So where is the water consumption? This looks like a telegram describing the meter.

Can you collect some more telegrams when the water meter actually consumes water.

It could be that the water meter sends multiple types of telegrams, some often with water data,
and some more seldom that describe the water meter, like the one above.

weetmuts, 2022-05-25:
ping @jczmok-iag

weetmuts, 2022-09-21:
Can you collect some more telegrams? Sometimes the meters sends the consumption telegram often and then intermittently they send a meta-data telegram like this one.

BIBOLV, 2023-01-01:
Closing due to inactivity. Please reopen issue if you manage to collect more telegrams.

closed 2023-01-01 by BIBOLV