wmbusmeters/wmbusmeters #417: Support for Powerplus Electricity Meter

jczmok-iag, 2021-12-22:
And here's the next one which is unsupported:

Received telegram from: 34600086
          manufacturer: (PPC) Power Plus Communications, Germany (0x4203)
                  type: Electricity meter (0x02)
                   ver: 0x01
                driver: unknown!

debug:

(simulation) from file "2E4403428600603401027AC90020050A079223FA9AF7B3F4D3AE7BB32DFAD7EA99CABEF06BA2440E0FBE483A967423" to trigger at relative time 0
(wmbus) ff a dll crc first (calculated f216) did not match (expected 7ac9) for bytes 0-10!
(wmbus) parseDLL @0 47
(telegram) DLL L=2e C=44 (from meter SND_NR) M=4203 (PPC) A=34600086 VER=01 TYPE=02 (Electricity meter) (driver unknown!) DEV= RSSI=0
(wmbus) parseELL @10 37
(wmbus) parseNWL @10 37
(wmbus) parseAFL @10 37
(wmbus) parseTPL @10 37
(telegram) TPL CI=7a ACC=c9 STS=00 CFG=0520 (AES_CBC_IV nb=2 cntn=0 ra=0 hc=0)
Received telegram from: 34600086
          manufacturer: (PPC) Power Plus Communications, Germany (0x4203)
                  type: Electricity meter (0x02)
                   ver: 0x01
                driver: unknown!
(wmbus) 000   : 2e length (46 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : 0342 dll-mfct (PPC)
(wmbus) 004   : 86006034 dll-id (34600086)
(wmbus) 008   : 01 dll-version
(wmbus) 009   : 02 dll-type (Electricity meter)
(wmbus) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 011   : c9 tpl-acc-field
(wmbus) 012   : 00 tpl-sts-field (OK)
(wmbus) 013   : 2005 tpl-cfg 0520 (AES_CBC_IV nb=2 cntn=0 ra=0 hc=0 )
telegram=||2E4403428600603401027AC90020050A079223FA9AF7B3F4D3AE7BB32DFAD7EA99CABEF06BA2440E0FBE483A967423|+0

**telegrams:**
telegram=||2E4403428600603401027ABF002005925EAC174603230B5BA331F8F4213E7A6ACACA2A6E20CAE8AE26CED3BB09142A|+0
telegram=||2E4403428600603401027AC2002005A0C4AD26694E2D52779165909563DA243ABA7D719774CB47BF11240833707D77|+0
telegram=||2E4403428600603401027AC30020057FD7937EEF32694ED6A956CC94CF68040BAF2DB72EDCE623A2407190D2D67DE2|+0
telegram=||2E4403428600603401027AC60020058F909E388F0406803B65C5B4D7A281AB9D05C3E5FD8F374A4FA5B5C7AA50022B|+0
telegram=||2E4403428600603401027AC7002005A0FC4E281F46CCD267108FD0554E9B422F58C0AB523C9B6B5746F574DEE9A457|+0
telegram=||2E4403428600603401027AC90020050A079223FA9AF7B3F4D3AE7BB32DFAD7EA99CABEF06BA2440E0FBE483A967423|+0

weetmuts, 2022-01-13:
Thank you! But those telegrams are still encrypted. Do you have the key for the meter? If so run:
wmbusmeters --logtelegrams simulation_readings.txt MyElectricity omnipower 34600086 yourkey
and post the output here.

BIBOLV, 2023-01-01:
Closing due to inactivity. Please reopen issue if you manage to collect unencrypted telegrams.

closed 2023-01-01 by BIBOLV