wmbusmeters/wmbusmeters #421: Add support for uart via tcp

ficueu, 2021-12-28:
Hello,
I made ESP32 - AMB8426 gateway to read mbus packets and send it over LAN. Currently I use socat to make virtual com port on debian, but socat isn't very stable for long time. Maybe nice feature will be native support for tcp client?

Gateway is based on ESP32 + LAN8720A + POE + AMB8426 with ESPHome and stream server from github://oxan/esphome-stream-server

weetmuts, 2022-01-27:
This is a good idea and I have thought about it before. Do you have any suggestion on how to proceed?

ficueu, 2022-01-28:
You can check tcp client for Z2M: https://github.com/Koenkk/zigbee2mqtt/blob/07a8a78a551f95ef345d97b4f5db79cb85318189/scripts/zigbee2socat_installer.sh

SzczepanLeon, 2022-11-16:
> This is a good idea and I have thought about it before. Do you have any suggestion on how to proceed?

Maybe option CMD for stdin can be added (just like for rtlwmbus).

`stdin:CMD(nc -lku 4444)`

BIBOLV, 2023-01-01:
Feature implemented - `rawtty:CMD(nc -lk 4444)`

closed 2023-04-03 by github-actions[bot]