wmbusmeters/wmbusmeters #422: LSE Heat Cost Allocator driver unknown

sheineme, 2021-12-29:
Hi, I get following driver "unknown" messages in the wmbusmeters.log:

Received telegram from: 86337967
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x2c
                device: rtlwmbus[77771111153705700]
                  rssi: 95 dBm
                driver: unknown!

(I also recieve telegrams from other devices with a different version "0x35", for which the dirver can be resolved as "qcaloric")

Can support for the "0x2c" version be added?

Thanks!

weetmuts, 2022-01-13:
Yes, please post a telegram when using --logtelegrams here. Thanks!

sheineme, 2022-02-07:
Finally had some time to play around a little bit with "wmbusmeters --analyze" and looked around in the issue list...
Found that #203 might be related to this one. There a HCA labeled "Kalo 1.5X" is mentioned. Mine are actually "Kalo 1.5M", which according to www.kalo.de is the previous version of the "X" one.

In my logs, I also see 2 (maybe even 3?) "types" of telegrams with different lengths (31, 36 and 49 bytes long). I didn't make much progress fiddeling around with drivers qcaloric, lse_08 or wh46x.

Will come back with more telegram data when I find the time to capture some more.

BIBOLV, 2023-01-01:
Closing due to inactivity. Please reopen issue when telegrams collected.

closed 2023-01-01 by BIBOLV