wmbusmeters/wmbusmeters #423: wmbusmeters stdin:rtlw not fully working?

MaxPeal, 2021-12-29:
i captured 15m signals with rtl_sdr
`timeout --preserve-status 15m rtl_sdr -f 868.625M -s 1600000 - 2>/dev/null | tee rtl_wmbus-samples_2021-12-29-14.bin | rtl_wmbus -s | tee rtl_wmbus-samples_2021-12-29-14.bin.decode`

weetmuts, 2021-12-29:
What do the lines in rtl_wmbus-samples_2021-12-29-14.bin.decode start with?

If they start with T1;0 or C1;0 or S1:0 then a signal was received but the crc failed,
so wmbusmeters ignores them, rightly so, since they are corrupt.

Only lines with T1;1 or C1;1 or S1;1 are used.

weetmuts, 2021-12-29:
Ah, you have created two issues.... check the https://github.com/weetmuts/wmbusmeters/issues/424
for the solution. I think we can close this as a duplicate.

MaxPeal, 2021-12-30:
closing in favor of https://github.com/weetmuts/wmbusmeters/issues/426

closed 2021-12-30 by MaxPeal