wmbusmeters/wmbusmeters #424: telegram with failed CRC checks trigger wrongly the emergency exit

MaxPeal, 2021-12-29:
telegram with failed CRC checks trigger wrongly the emergency exit
```
$ cat rtl_wmbus-samples_2021-12-29-14.bin.decode.sample | $wmbusmeters stdin:rtlwmbus 
Started config rtlwmbus on stdin listening on any
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 72172368
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Water meter (0x07) encrypted
                   ver: 0x76
                device: rtlwmbus[]
                  rssi: 16 dBm
                driver: unknown!
Received telegram from: 93843999
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Smoke detector (0xf0)
                   ver: 0x76
                device: rtlwmbus[]
                  rssi: 124 dBm
                driver: tsd2
$
```
but with the CRC errors filtered:
```
$ cat rtl_wmbus-samples_2021-12-29-14.bin.decode.sample | grep -vE 'T1;0;0;|C1;0;0;|S1;0;0;' | $wmbusmeters stdin:rtlwmbus 
Started config rtlwmbus on stdin listening on any
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 72172368
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Water meter (0x07) encrypted
                   ver: 0x76
                device: rtlwmbus[]
                  rssi: 16 dBm
                driver: unknown!
Received telegram from: 93843999
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Smoke detector (0xf0)
                   ver: 0x76
                device: rtlwmbus[]
                  rssi: 124 dBm
                driver: tsd2
Received telegram from: 03727567
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat Cost Allocator (0x08) encrypted
                   ver: 0x94
                device: rtlwmbus[]
                  rssi: 94 dBm
                driver: fhkvdataiv
Received telegram from: 91540946
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Smoke detector (0xf0)
                   ver: 0x76
                device: rtlwmbus[]
                  rssi: 128 dBm
                driver: tsd2
Received telegram from: 72262375
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Water meter (0x07) encrypted
                   ver: 0x76
                device: rtlwmbus[]
                  rssi: 61 dBm
                driver: unknown!
Received telegram from: 03684646
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat Cost Allocator (0x08) encrypted
                   ver: 0x94
                device: rtlwmbus[]
                  rssi: 67 dBm
                driver: fhkvdataiv
Received telegram from: 02671580
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Radio converter (meter side) (0x37) encrypted
                   ver: 0x95
      Concerning meter: 68324475
          manufacturer: (HYD) Hydrometer (0x2324)
                  type: Warm Water (30°C-90°C) meter (0x06) encrypted
                   ver: 0xfe
                device: rtlwmbus[]
                  rssi: 14 dBm
                driver: unknown!
Received telegram from: 15735440
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat Cost Allocator (0x08) encrypted
                   ver: 0x94
                device: rtlwmbus[]
                  rssi: 7 dBm
                driver: fhkvdataiv
Received telegram from: 03714189
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat Cost Allocator (0x08) encrypted
                   ver: 0x94
                device: rtlwmbus[]
                  rssi: 58 dBm
                driver: fhkvdataiv
$
```

MaxPeal, 2021-12-29:
versions i tried:
wmbusmeters: 1.5.0-35-g6a4266e
wmbusmeters: 1.3.0
wmbusmeters: 1.4.0
wmbusmeters: 1.4.1
wmbusmeters: 1.5.0

```
export wmbusmeters="docker run --rm -i weetmuts/wmbusmeters:release-1.3.0 /wmbusmeters/wmbusmeters"
user@box:~/GIT-REPROs/docker-wmbusmeters-maxpeal-private$ $wmbusmeters --version
wmbusmeters: 1.3.0
da690f2afdc26bf96caf418ccf06543e5689da01
```

MaxPeal, 2021-12-29:
debug output and input file rtl_wmbus-samples_2021-12-29-14.bin.decode.sample
https://gist.github.com/MaxPeal/fb5f4db2b1459e45fc936b6cc2c844a8

weetmuts, 2021-12-29:
Thanks! You are absolutely right. The bug was that when piping an archive file with multiple rtlwmbus telegrams, then
the frame chunking would discard everything in the buffer, when an T0 C0 or S0 was found....

For realtime usage rtl_sdr | rtl_wmbus | wmbusmeters this is not a problem since the buffer is tiny and only contains the currently received telegram or perhaps another one. However when piping a large file into wmbusmeters, then the buffer that was discarded after a crc error could be large, depending on how big chunks is read from stdin.

Also note that wmbusmeters defaults to --ignoreduplicates=true because rtlwmbus outputs duplicate telegrams because it has two decoding methods. If both decoding methods succeeds, then you get duplicates. So, with the fixed wmbusmeters,  if you do:
cat rtl_wmbus-samples_2021-12-29-14.bin.decode.sample | grep "^.1;1" | wc -l
You get 18 lines.
When run through wmbusmeters like this:
cat rtl_wmbus-samples_2021-12-29-14.bin.decode.sample | ./build/wmbusmeters --ignoreduplicates=false stdin:rtlwmbus 2>&1 | grep Received | wc -l
You also get 18 lines.

I have pushed the fix. Please test to see if you agree that it is fixed now.

MaxPeal, 2021-12-30:
LGFM
thanks for the fix.
with my testing its working, now im crunching my data for the unsupported meters.

closed 2021-12-30 by MaxPeal