wmbusmeters/wmbusmeters #426: rla;C1;1;1;support for rtl_wmbus -v (show used algorithm in the output) t2a;T1;1;1; / rla;T1;1;1; / t2a;C1;1;1; /

MaxPeal, 2021-12-30:
i make the rtl_wmbus capture first, and save as much data as possible, but without the "rtl_sdr raw data":
```rtl_sdr -d1 -f 868.625M -s 1600000 - 2>./rtl_sdr_v5_-13-error.log | rtl_wmbus -sv 2>./rtl_wmbus_v5_-13-error.log | tee rtl_wmbus_2021-12-30_v5_-13.bin.decode```

and after this the wmbusmeters decode, with cuting the not supported field:
```cat rtl_wmbus_2021-12-30_v5_-13.bin.decode | cut -d";" -f2- | $wmbusmeters --format=json --logfile=rtl_wmbus_2021-12-30_v5_-13.bin.decode.json stdin:rtlwmbus MyNOKEY auto \* NOKEY```

it would  be nice to have the field supported.

weetmuts, 2022-05-25:
Interesting, but why do you want to capture the algorithm?

BIBOLV, 2023-01-01:
Closing due to inactivity. Please reopen issue if you want to continue this topic.

closed 2023-01-01 by BIBOLV