wmbusmeters/wmbusmeters #428: escaping is broken / add Support for ANYID / NOID as substitute for * meter_id, as * dont work with variables

MaxPeal, 2021-12-30:
can we please add a Support for ANYID / NOID as substitute for * meter_id, as * dont work in / with variables?

as no way for escaping works:
```
export METERS="MyWarmWater supercom587 12345678 NOKEY
      STERN       fhkvdataiv  \"*\" FCF41938F63432975B52505F547FCEDF
      Hetta       elf         01885619 NOKEY"
$wmbusmeters any $METERS
Not a valid id nor a valid meter match expression "\*"

export METERS="MyWarmWater supercom587 12345678 NOKEY
      STERN       fhkvdataiv  \* NOKEY
      Hetta       elf         01885619 NOKEY"
$wmbusmeters any $METERS
Not a valid id nor a valid meter match expression "\*"

export METERS="MyWarmWater supercom587 12345678 NOKEY
      STERN       fhkvdataiv  \'*\' NOKEY
      Hetta       elf         01885619 NOKEY"
$wmbusmeters any $METERS      
Not a valid id nor a valid meter match expression "\'*\'"

export METERS="MyWarmWater supercom587 12345678 NOKEY
      STERN       fhkvdataiv  \"*\" NOKEY
      Hetta       elf         01885619 NOKEY"
$wmbusmeters any $METERS      
Not a valid id nor a valid meter match expression ""*""

export METERS="MyWarmWater supercom587 12345678 NOKEY
      Octal       fhkvdataiv  \\042 NOKEY
      Hetta       elf         01885619 NOKEY"
$wmbusmeters any $METERS      
Not a valid id nor a valid meter match expression "\042"

export METERS="MyWarmWater supercom587 12345678 NOKEY
      Octal       fhkvdataiv  \x2A NOKEY
      Hetta       elf         01885619 NOKEY"
$wmbusmeters any $METERS      
Not a valid id nor a valid meter match expression "\x2A"
```

weetmuts, 2021-12-30:
Sure, we can add that. But this problem is a shell problem, not a wmbusmeters problem.
Also it is much more convenient to use a config files instead, then no escaping is needed.

weetmuts, 2021-12-30:
You can make a pull request. Add an if statement checking if id=="ANYID" if so then id="*"
adding this after line 635 in cmdline.cc

weetmuts, 2022-01-13:
Thanks for the pull request! It seems my proposed solution was not that good. Sorry. But I pushed the correct solution now. Please test!

closed 2022-04-14 by weetmuts