wmbusmeters/wmbusmeters #429: cant find kamstrup omnipower, is this correct setup to another frequency?

cicci44, 2021-12-31:
`wmbusmeters --debug auto:t1` makes cpu go in roof..  using instead
`wmbusmeters --debug rtlwmbus:t1` or `wmbusmeters --debug rtlwmbus:c1` showing output, but only watersensors on 868.625M.. So i guess the electric meter (kamstrup omnipower) is on another frequency.

identifing the kamstrups omni antenna as "Frekvensområden: 434/444/900/1800 mhz"

`wmbusmeters --debug rtlwmbus:433M` waiting 10min without nothing..
`wmbusmeters --debug rtlwmbus:434M` waiting 10min without nothing..
i scope into 444 using SDR# https://i.imgur.com/cAgF9A3.png and see something....
`wmbusmeters --debug rtlwmbus:444M` waiting 10min without nothing..but.. maybe something..

`(serial) received ascii "T1;0;0;2021-12-31 15:02:14.000;35;11;FFFFFFFF;0x0dffff98ffffffffffffffffffff<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2021-12-31 15:02:14.000;35;11;FFFFFFFF;0x0dffff98ffffffffffffffffffff<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
`

weetmuts, 2022-01-13:
Can you try rtl_433 ? It might explain better what kind of signal you have there.

Have you verified that your omnipower actually has a wmbus module transmitting? A lot of electricity meters are modular and can have different types of modules installed, gsm/wmbus/etc...

BIBOLV, 2023-01-01:
Closing due to inactivity. Please reopen issue to continue discussion if necessary.

closed 2023-01-01 by BIBOLV