wmbusmeters/wmbusmeters #430: (apator162) telegram contains a register (96) with unknown size

rafkan20, 2022-01-01:
(wmbusmeters) shutting down
(wmbusmeters) logging started 2021-12-26 09:07:28 using 1.5.0-8-g67e0072
[2021-12-26_09:07:28] Started config rtlwmbus[00000001] listening on none using fq 868.9M
(wmbusmeters) waiting for telegrams
[2021-12-26_09:07:30] (memory) rss 3309568 peak 3.15 MiB
(wmbusmeters) started meter 1 (zimna 00123456 apator162)
(wmbusmeters) started meter 2 (ciepla 00788234 apator162)
[2021-12-27_09:07:32] (memory) rss 3502080 peak 3.38 MiB
[2021-12-28_09:07:34] (memory) rss 3506176 peak 3.38 MiB
[2021-12-29_09:07:35] (memory) rss 3506176 peak 3.38 MiB
[2021-12-30_09:07:36] (memory) rss 3506176 peak 3.38 MiB
[2021-12-31_09:07:37] (memory) rss 3506176 peak 3.38 MiB
[2022-01-01_00:26:26] (apator162) telegram contains a register (96) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 6E4401063502890005077A1A0060852F2F0F0001C196180200438E078300061A62961A01B9340210127201007B0112720100DD6C0100EF680100C4630100AF5F01001C5B0100A5560100E5500100684B0100F24501007E400100493B0100A0D3111C02FFFFFFFFFFFF$
[2022-01-01_00:31:35] (apator162) telegram contains a register (96) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 6E4401063502890005077A1B0060852F2F0F0501C196180200438E0784F396B5764ABA0203000010127201007B0112720100DD6C0100EF680100C4630100AF5F01001C5B0100A5560100E5500100684B0100F24501007E400100493B0100A0D3111C02FFFFFFFFFFFF$
[2022-01-01_00:34:16] (apator162) telegram contains a register (96) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 6E4401064842830005077ABA0060852F2F0F0001C196980200438E07875E3D2C147B5718BB1C001017D903007B0117D903005BCB030013BE0300DCAD030069A00300F090030053820300C4700300476203000D5303007E45030082390300A0270F7D03FFFFFFFFFFFF$
[2022-01-01_00:35:39] (apator162) telegram contains a register (96) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 6E4401063502890005077A1C0060852F2F0F0A01C196180200438E078300061A62961A01B9340210127201007B0112720100DD6C0100EF680100C4630100AF5F01001C5B0100A5560100E5500100684B0100F24501007E400100493B0100A0D3111C02FFFFFFFFFFFF$
[2022-01-01_00:39:07] (apator162) telegram contains a register (96) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 6E4401064842830005077ABB0060852F2F0F0501C196980200438E078300081A2F751A01BDC8011017D903007B0117D903005BCB030013BE0300DCAD030069A00300F090030053820300C4700300476203000D5303007E45030082390300A0270F7D03FFFFFFFFFFFF$
[2022-01-01_00:40:39] (apator162) telegram contains a register (96) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 6E4401063502890005077A1D0060852F2F0F0F01C196180200438E0784F396B5764ABA0203000010127201007B0112720100DD6C0100EF680100C4630100AF5F01001C5B0100A5560100E5500100684B0100F24501007E400100493B0100A0D3111C02FFFFFFFFFFFF$
[2022-01-01_00:45:49] (apator162) telegram contains a register (96) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 6E4401064842830005077ABC0060852F2F0F0C01C196980200438E07847B03571EB85748C202001017D903007B0117D903005BCB030013BE0300DCAD030069A00300F090030053820300C4700300476203000D5303007E45030082390300A0270F7D03FFFFFFFFFFFF$
[2022-01-01_00:46:17] (apator162) telegram contains a register (96) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 6E4401063502890005077A1E0060852F2F0F1401C196180200438E078300061A62961A01B9340210127201007B0112720100DD6C0100EF680100C4630100AF5F01001C5B0100A5560100E5500100684B0100F24501007E400100493B0100A0D3111C02FFFFFFFFFFFF$
[2022-01-01_00:50:18] (apator162) telegram contains a register (96) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 6E4401064842830005077ABD0060852F2F0F1001C196980200438E07875E3D2C147B5718BB1C001017D903007B0117D903005BCB030013BE0300DCAD030069A00300F090030053820300C4700300476203000D5303007E45030082390300A0270F7D03FFFFFFFFFFFF$
[2022-01-01_00:50:38] (apator162) telegram contains a register (96) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 6E4401063502890005077A1F0060852F2F0F1901C196180200438E0784F396B5764ABA0203000010127201007B0112720100DD6C0100EF680100C4630100AF5F01001C5B0100A5560100E5500100684B0100F24501007E400100493B0100A0D3111C02FFFFFFFFFFFF$
[2022-01-01_00:54:02] (apator162) telegram contains a register (96) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 6E4401064842830005077ABE0060852F2F0F1401C196980200438E078300081A2F751A01BDC8011017D903007B0117D903005BCB030013BE0300DCAD030069A00300F090030053820300C4700300476203000D5303007E45030082390300A0270F7D03FFFFFFFFFFFF$
[2022-01-01_00:57:31] (apator162) telegram contains a register (96) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/

weetmuts, 2022-01-01:
Fix pushed. Please test!

rafkan20, 2022-01-01:
Is the version with the fix?
wmbusmeters version: 1.5.0-39-g2fc10a0

weetmuts, 2022-01-01:
Just do git log, and the latest commit should be today, with the comment Add another register size for apator162.
The version number is: 1.5.0-41-g5ccde88

rafkan20, 2022-01-01:
Thank you very much - it works as it should.

weetmuts, 2022-01-01:
Great!

closed 2022-01-01 by weetmuts