wmbusmeters/wmbusmeters #431: Apator162 water meter can not decode telegram after 31.12.2021

wajo666, 2022-01-01:
Hi,

in year 2021 apator162 worked correctly and return data, but in this year I got 0 value and I found in log this error

(apator162) telegram contains a register (96) with unknown size.
Please open an issue at https://github.com/weetmuts/wmbusmeters/
and report this telegram: 6E4401067906750305077AF00060852F2F0FE961C196080200437F0083000BA95879AB011CF60910570600007B01420600004801000054000000540000005400000000000000000000000000000000000000000000000000000000000000A07BDA1B04FFFFFFFFFFFFFFFFFFFF8F2E

weetmuts, 2022-01-01:
Thanks, it is actually just fixed. Please build the latest version.

wajo666, 2022-01-01:
can you release new docker version ?

closed 2022-01-01 by wajo666

weetmuts, 2022-01-01:
Yep, a new docker is built. See if it is updated for you...