wmbusmeters/wmbusmeters #432: (apator162) telegram contains a register (96) with unknown size

philips1923, 2022-01-01:
Hi, Im reporting an issue: 3E4401069250960205077A580030852F2F0F84A1C196980000432D0284B23E8A7152AC32170200105753030071018552030079330300A0126DEF03FFFFF02D

weetmuts, 2022-01-01:
Thanks, fix pushed and released in 1.6.0 today.

VoyteckPL, 2022-01-02:
I have the same. Running hass os and addon.

philips1923, 2022-01-02:
My situation as above. Could you please tell me how to update this addon with your newest fix? Auto-update doesnt show any updates in HA.

weetmuts, 2022-01-02:
Ah, yes, forgot to update the ha-addon/config.json version number.
I have now pushed an update where the version changed from 0.2.19 to 0.2.20

Hopefully this should trigger the ha-addon to rebuild. Please check!

weetmuts, 2022-01-02:
@to-masz Anything more that a hass user needs to do?

philips1923, 2022-01-03:
The only thing I had to was reinstall addon and now works great, BTW thanks a lot for this addon, finally I can control water usage :)

The only thing is quite strange that CPU usage is from few days ~17-18%, earlier was about 11-12%.
[image]

to-masz, 2022-01-05:
@weetmuts I checked some scenarios and I have found only one when the update option was not shown in Home Assistant Supervisor:
> It looks like ha supervisor keeps the static reference between repository and addon. If you re-added weetmuts repository to the add-on store it doesn't keep track of new versions of your addon. There were also some reports that this reference can be lost in specific ha updates. There were some major changes around December in Supervisor configuration.
>
> My recommendation is to backup your addon configuration, uninstall it, readd the repository to addon store, install it again and restore the configuration. Quite lengthy process, but proven to solve the issue. 

Besides that the version update in addon config is enough. 

However, if someone will have similar problems, please let me know. I can assist to figure out the detailed reasons.

to-masz, 2022-01-05:
@philips1923 if CPU is your concern, you can consider buying a dongle with hardware processing of rf wmbus protocol frames... some time ago there were few words about nanoCUL in the docs. I cannot see it right now.

I got the following one, flashed it with nanoCUL-MBUS.hex and it works perfectly fine (cpu usage <1% on rpi4).
https://www.smart-home-komponente.de/nano-cul/nano-cul-868-extern/ (cheap delivery to PL)
https://www.smart-home-komponente.de/support/firmware/
https://www.smart-home-komponente.de/support/programme/

closed 2022-01-11 by philips1923

philips1923, 2022-03-02:
My installation of HA was damaged (full disk space on proxmox) so I made bare-metal install of HAOS on my HP T630 and my RTL stick stands up together with system (after power-outage).