wmbusmeters/wmbusmeters #433: How to run wmbusmeters scheduled at a particular time

to-scho, 2022-01-02:
Hi, great project, I got readings for all of my 14 heat cost allocators now. The readings are quite frequent, something in the 15 minutes range for each unit. Nevertheless, changed values are only seen once during night.

Is it possible to run wmbusmeters not 24/7? Processor load on my raspberry pi is 100% all the time which does not allow to use it for other purpose. For me it would be sufficient to run it only once per day for e.g. 30 minutes.  

I tried around with system timers, exitafter (unfortunately not a config file option when using daemon) and finally an ugly RuntimeMaxSec in the system.service file. That gave me the appropriate behaviour but no readings any more. I only see wmbusmeters is started at 100% load but no more other processes rtf_wmbus/rtl_sdr. I had same when initially playing with some docker container versions of wmbusmeters which were not working for me.

Any hint for me?

BIBOLV, 2022-01-02:
Do you need to use daemon? Running from cron when needed with --exitafter or --oneshot is not what you are looking for?

to-scho, 2022-01-03:
Hi @BIBOLV, thanks for the advice. I finally got is as a systemd timed service, files posted for others below. Quite lengthy since everything including meter definition must be on the command line now but ok since this must not be changed too often.

BTW, The --oneshot was not working for me. Even when 14 meters are given the process exited after the first one was read.

```
$ cat /lib/systemd/system/wmbusmeters.timer 
[Unit]
Description=Run wmbusmeters

[Timer]
OnCalendar=*-*-* 4:20:00

[Install]
WantedBy=timers.target

```
```
$ cat /lib/systemd/system/wmbusmeters.service 
[Unit]
Description="wmbusmeters service"
Documentation=https://github.com/weetmuts/wmbusmeters
Documentation=man:wmbusmeters(1)

[Service]
Type=oneshot
PrivateTmp=yes
User=wmbusmeters
Group=wmbusmeters

# Run ExecStartPre with root-permissions
PermissionsStartOnly=true
ExecStartPre=-/bin/mkdir -p /var/log/wmbusmeters/meter_readings
ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters
ExecStartPre=-/bin/mkdir -p /run/wmbusmeters
ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbusmeters

ExecStart=/usr/bin/wmbusmeters --exitafter=15m --format=json --meterfiles='/var/log/wmbusmeters/meter_readings' --meterfilesaction=overwrite --logfile='/var/log/wmbusmeters/wmbusmeters.log' --shell='/usr/bin/mosquitto_pub -h 192.168.1.1 -t wmbusmeters/$METER_NAME -m "$METER_JSON"' auto:t1 arbeitszimmer_eg auto 12345678 NOKEY arbeitszimmer_og auto 12345678 NOKEY arbeitszimmer_ug auto 12345678 NOKEY bad_eg auto 12345678 NOKEY bad_og auto 12345678 NOKEY bad_ug auto 12345678 NOKEY esszimmer_eg auto 12345678 NOKEY gaestezimmer_ug auto 12345678 NOKEY kueche_og auto 12345678 NOKEY kueche_ug auto 12345678 NOKEY schlafzimmer_eg auto 12345678 NOKEY schlafzimmer_og auto 12345678 NOKEY wohnzimmer_eg auto 12345678 NOKEY wohnzimmer_og auto 12345678 NOKEY
ExecReload=/bin/kill -HUP wmbusmeters
```

`sudo systemctl disable wmbusmeters.service` to disable deamon service in case installed before, `sudo systemctl enable wmbusmeters.timer` to enable timer based service, `sudo reboot now` ... and `sudo systemctl list-timers --al` to see when next timer will execute.

weetmuts, 2022-02-26:
There was a bug in --oneshot, it is now fixed. https://github.com/weetmuts/wmbusmeters/issues/376

weetmuts, 2022-03-28:
Ping @to-scho ? Does it work now with the oneshot fix?

to-scho, 2022-04-16:
I can confirm that --oneshot is working now as expected.

closed 2022-04-16 by to-scho